avrdudes/avrdude #359: [bug #43471] avrdude 5.11 buffer overruns in stk500v2 code

avrdude-admin, 2021-12-10:
bob frazier <None>
Sat 25 Oct 2014 09:30:35 AM UTC

currently, several fixed-length 'on the stack' buffer sizes in the stk500v2 code cause severe failures if the page size for the device you are flashing is larger than 256 bytes.  In the case of the ATXMega devices, particularly those with more than 64kb of application area NVRAM, the flash memory page sizes must necessarily be assigned to 512 bytes, possibly more.  Applying the patch to stk500v2.c corrects the fixed-length buffer problem by using 'malloc/free' to allocate an arbitrary-sized buffer.  NOTE: this patch is NOT complete.  It only covers code that affects the use of 'wiring' protocol for uploading images larger than 64kb via the Arduino IDE to an appropriate ATXMega processor with a bootloader that is capable of using this protocol.

[file #32319: patch-stk500v2.c](https://savannah.nongnu.org/bugs/download.php?file_id=32319)

This issue was migrated from https://savannah.nongnu.org/bugs/?43471

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 19 Nov 2014 05:06:46 PM UTC

The problem is that this situation (Xmega but STK500v2 protocol
anyway) is uncommon.  The entire code had been tailored on the
non-Xmega devices as Xmegas use their own (different) protocols.

Also, the STK500v2 itself is limited to a maximal message
size of 275 bytes, see AVR068, page 2.  That used to cause
the current implementation.

I think the actual issue is that the following procedure
described in AVR068 is not implemented:

"For devices with page size bigger than what can be
transferred to STK500 in one command, several
CMD_PROGRAM_FLASH_ISP commands must be issued.  In such a
case, only the last command should have the Write Page mode
bit set."

I wonder whether your bootloader would be able to handle that.

Nevertheless, changing the fixed buffers into malloc'ed
buffers might make a lot of sense.  Other programmer
implementations (like those for the JTAGICEs) already do it
that way.

MCUdude, 2022-01-09:
It looks like Jörg provided a solution here. Issue can probably be closed

closed 2022-01-10 by dl8dtl