avrdudes/avrdude #360: [bug #43577] Extended address not sent with avr109

avrdude-admin, 2021-12-10:
Arne Rossius <arossius@ias.rwth-aachen.de>
Tue 11 Nov 2014 02:14:43 PM UTC

When programming the ATxmega256a3 with avrdude 6.1 (windows version), the extended address is not sent when jumping across a 128 KiB boundary. I have created a text hexfile which only contains data for flash pages 0 (word address 0x00000) and 511 (word address 0x1FF00). However as can be seen from the attached log, after programming the first page only a 16-bit address command ('A') is sent instead of the 24-bit address command ('H'), resulting in an address of 0x0FF00 instead of 0x1FF00. Verify doesn't fail because the same happens when reading, but does fail when programming an actual application (without a gap) larger than 128 KiB.

The programmer is the "xboot" bootloader <[https://github.com/alexforencich/xboot/wiki](https://savannah.nongnu.orghttps://github.com/alexforencich/xboot/wiki)>. The avrdude.conf is unchanged.

Programming the attached hex file does work with the older avrdude 5.11 because this version programs pages 1 to 510 with 0xFF, only sending the address once before the first page and then relying on the bootloader's auto-increment to reach the high addresses. As can be expected, it takes a very long time though.

Also, as can be seen from the xboot source, it resets the extended address byte to zero when receiving an 'A' command. I'm not sure if this behaviour is correct, but if it is, this means that avrdude must send an 'H' command for all flash pages above 128 KiB (unless no address is sent at all between adjacent pages).

[file #32418: napftest.hex](https://savannah.nongnu.org/bugs/download.php?file_id=32418)
[file #32419: avrdude-error.txt](https://savannah.nongnu.org/bugs/download.php?file_id=32419)

This issue was migrated from https://savannah.nongnu.org/bugs/?43577

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 11 Nov 2014 04:48:58 PM UTC

The "extended address" is not part of the original AVR109
appnote.

**[bug #29896](https://savannah.nongnu.org/bugs/?29896)** has already been filed to request adding this
protocol extension, but has not been handled so far. (Sorry
for that.)

avrdude-admin, 2021-12-10:
Arne Rossius <arossius>
Tue 11 Nov 2014 06:23:38 PM UTC

AFAICS, **[bug #29896](https://savannah.nongnu.org/bugs/?29896)** calls for additional memory types, while this bug is about accessing the area above 128 KiB of the regular flash memory (not bootloader area which is above 256 KiB for xmega256a3), so IMHO it's not a duplicate.

There is already a function in butterfly.c to send the 'H' command (butterfly_set_extaddr()), but it seems that the old butterfly_set_addr() is called instead. Surely this can't be right?

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 11 Nov 2014 07:41:51 PM UTC

OK, sorry, then I misunderstood it.  Reopening it, so I can
handle that (hopefully) alltogether.

mcuee, 2022-06-11:
Hopefully #454 patch can fix this issue.
* #454

mcuee, 2022-06-22:
Ref: xmega relared issues
* https://github.com/avrdudes/avrdude/issues/164
* https://github.com/avrdudes/avrdude/issues/190
* https://github.com/avrdudes/avrdude/issues/200
* https://github.com/avrdudes/avrdude/issues/226
* https://github.com/avrdudes/avrdude/issues/243
* https://github.com/avrdudes/avrdude/issues/289
* https://github.com/avrdudes/avrdude/issues/304
* https://github.com/avrdudes/avrdude/issues/311
* https://github.com/avrdudes/avrdude/issues/314
* https://github.com/avrdudes/avrdude/issues/337
* https://github.com/avrdudes/avrdude/issues/360
* https://github.com/avrdudes/avrdude/issues/365
* https://github.com/avrdudes/avrdude/issues/391 (but also affect other AVR chips).
* https://github.com/avrdudes/avrdude/issues/404
* https://github.com/avrdudes/avrdude/issues/435
* https://github.com/avrdudes/avrdude/issues/454

mcuee, 2022-08-07:
My test results in #454 show that the writing is good but not verification. But that is based on ATmega2560 with xboot, not ATxmega256a3. So ATxmega256a3 may get more issues (need to get some one to help testing as I do not have ATxmega256a3).

* https://github.com/avrdudes/avrdude/issues/454

mcuee, 2022-08-08:
https://savannah.nongnu.org/bugs/?29896 is now #200.

mcuee, 2022-09-24:
@MCUdude 

Now that you have access to the ATxmega256A3BU Xplained board, please take a look at this one and #404 when you got a chance. Thanks.

MCUdude, 2022-09-24:
I compiled xboot for ATxmega256A3BU, and I'm able to communicate with Avrdude using the bootloader:

```
$ ./avrdude -cavr109 -p atxmega256a3bu -b 115200 -P /dev/cu.usbserial-1410 

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)

avrdude done.  Thank you.
```

I can also read from flash:

```
$ ./avrdude -cavr109 -p atxmega256a3bu -b 115200 -P /dev/cu.usbserial-1410 -Uflash:r:-:I

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading flash memory ...

Reading | ################################################## | 100% 26.46s

avrdude: flash is empty, resulting file has no contents
avrdude: writing output file <stdout>
:00000001FF

avrdude done.  Thank you.
```

However, I'm not able to write:

```
$ ./avrdude -cavr109 -p atxmega256a3bu -b 115200 -P /dev/cu.usbserial-1410 -Uflash:w:/Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex:i

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex for flash
avrdude: writing 35062 bytes flash ...

Writing | #                                                  | 1% 0.00s ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  

...

 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
 ***failed;  
Writing | ################################################## | 100% 0.10s

avrdude: 35062 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex

Reading | ################################################## | 100% 3.46s

avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0xec
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

MCUdude, 2022-09-24:
It turns out that I forgot to add the `-e` flag. Now I can get the bootloader to work. I'll try with a bigger file soon.

```
$ ./avrdude -cavr109 -p atxmega256a3bu -b 115200 -P /dev/cu.usbserial-1410 -e -Uflash:w:/Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex:i

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: erasing chip
avrdude: reading input file /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex for flash
avrdude: writing 35062 bytes flash ...

Writing | ################################################## | 100% 4.29s

avrdude: 35062 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex

Reading | ################################################## | 100% 3.46s

avrdude: 35062 bytes of flash verified

avrdude done.  Thank you.
```

MCUdude, 2022-09-24:
@mcuee I can confirm that there's an issue writing to addresses above 128kiB.

```
$ ./avrdude -cavr109 -p atxmega256a3bu -b 115200 -P /dev/cu.usbserial-1410 -e -Uflash:w:0x55_256kib.hex 

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: erasing chip
avrdude: reading input file 0x55_256kib.hex for flash
avrdude: writing 262143 bytes flash ...

Writing | ################################################## | 100% 31.83s

avrdude: 262143 bytes of flash written
avrdude: verifying flash memory against 0x55_256kib.hex

Reading | ################################################## | 100% 25.66s

avrdude: verification error, first mismatch at byte 0x1ffff
         0xff != 0x55
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

commit 62aa4808 (2022-10-05): Fix avr109 extended address
Closes #360 #454
closed 2022-10-05 by stefanrueger