avrdudes/avrdude #363: [bug #43758] avrdude 6.1 does not program Attiny40

avrdude-admin, 2021-12-10:
apzax <None>
Fri 05 Dec 2014 11:36:58 AM UTC

When I program Attiny40, I get a verification error

C:avrdude>avrdude.exe -c serbb_prog -P com2 -p t40 -U flash:w:D:developattiny40-testDebugattiny40-test.hex:a

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.08s

avrdude.exe: Device signature = 0x1e920e
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "D:developattiny40-testDebugattiny40-test.hex"
avrdude.exe: input file D:developattiny40-testDebugattiny40-test.hex auto detected as Intel Hex
avrdude.exe: writing flash (70 bytes):

Writing | ################################################## | 100% 1.24s

avrdude.exe: 70 bytes of flash written
avrdude.exe: verifying flash memory against D:developattiny40-testDebugattiny40-test.hex:
avrdude.exe: load data flash data from input file D:developattiny40-testDebugattiny40-test.hex:
avrdude.exe: input file D:developattiny40-testDebugattiny40-test.hex auto detected as Intel Hex
avrdude.exe: input file D:developattiny40-testDebugattiny40-test.hex contains 70 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.93s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x0040
0xff != 0xfc
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

Flash dump differs from the hex file:

C:avrdude>avrdude.exe -c serbb_prog -P com2 -p t40 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.08s

avrdude.exe: Device signature = 0x1e920e
avrdude> dump flash 0x0 0x50

>>> dump flash 0x0 0x50

- - - - 11 c0 18 c0 17 c0 16 c0  15 c0 14 c0 13 c0 12 c0  |................|0010  11 c0 10 c0 0f c0 0e c0  0d c0 0c c0 0b c0 0a c0  |........ . . . .|
0020  09 c0 08 c0 11 27 1f bf  cf e3 d1 e0 de bf cd bf  | ....'..........|
0030  02 d0 07 c0 e5 df 28 9a  51 e0 46 b1 45 27 46 b9  |......(.Q.F.E'F.|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit

>>> quit

avrdude.exe done.  Thank you.

--
Best regards

[file #32598: attiny40-test.hex](https://savannah.nongnu.org/bugs/download.php?file_id=32598)

This issue was migrated from https://savannah.nongnu.org/bugs/?43758

mcuee, 2022-05-29:
Maybe someone with the Tiny40 can test this to see if this is still an issue or not.

MCUdude, 2022-05-29:
I have a bunch of AT42QT2120 touch chips at work that is just ATtiny40's with some secret sauce Qtouch firmware. I'll see if I can connect a programmer to the right pins and flash something to it.

MCUdude, 2022-05-30:
I can confirm that the attached hex file ([file #32598: attiny40-test.hex](https://savannah.nongnu.org/bugs/download.php?file_id=32598)) can be written and verified just fine:

```
$ ./avrdude -c usbasp -v -p attiny40 -Uflash:w:/Users/hans/Downloads/attiny40-test.hex:i

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/hans/Downloads/avrdude/src/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATtiny40
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3   16      0     0     0 0x00 0x00
           fuse                    0     0     4    0 no          1   16      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1   16      0     0     0 0x00 0x00
           lockbits                0     0     0    0 no          1   16      0     0     0 0x00 0x00
           flash                   0     0   128    0 no       4096   64      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e920e (probably t40)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/hans/Downloads/attiny40-test.hex"
avrdude: writing flash (70 bytes):

Writing | ################################################## | 100% 0.25s

avrdude: 70 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/attiny40-test.hex:

Reading | ################################################## | 100% 0.05s

avrdude: 70 bytes of flash verified

avrdude done.  Thank you.
```

closed 2022-05-30 by MCUdude