avrdudes/avrdude #364: [bug #43811] Writing a hex file using a ft2232H programmer fails when poll after a page write does not come back as expected.

avrdude-admin, 2021-12-10:
Jack Gassett <jack.gassett@gadgetfactory.net>
Sat 13 Dec 2014 08:13:32 AM UTC

I found this message that describes exactly the error that I am experiencing:
 [https://lists.nongnu.org/archive/html/avrdude-dev/2012-05/msg00013.html](https://savannah.nongnu.orghttps://lists.nongnu.org/archive/html/avrdude-dev/2012-05/msg00013.html)

I looked for a patch for this issue but was not able to find one. Has the patch been published anywhere?

[file #32658: avrdude-poll-failure.txt](https://savannah.nongnu.org/bugs/download.php?file_id=32658)

This issue was migrated from https://savannah.nongnu.org/bugs/?43811

mcuee, 2022-06-05:
Not so sure if this is still relevant or not.

List mail link:
https://lists.nongnu.org/archive/html/avrdude-dev/2012-05/msg00013.html

<html>
<body>

From: | Hannes Weisbach
-- | --
Subject: | Re: [avrdude-dev] FT2232 issues - ident yes, program no
Date: | Thu, 10 May 2012 13:48:30 +0200

<hr style="color: rgb(0, 0, 0); font-family: &quot;Times New Roman&quot;; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><pre style="color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Am 10.05.2012 um 11:49 schrieb Ing. Daniel Rozsnyó:

&gt;<i> Hello,</i>
&gt;<i>  I am trying my new board, which got integrated a FT2232 chip to support a </i>
&gt;<i> serial port (Port B) and act as a programmer (Port A). The wiring is </i>
&gt;<i> according to avrdude.conf with a LS244 tri-state buffer.</i>
&gt;<i> </i>
&gt;<i>  When I run a quick test, with chip identification, the result was ok - it </i>
&gt;<i> reads the chip ID correctly. But when it gets to actual programming, the </i>
&gt;<i> avrdude is stuck at start of programming and no other commands work (not even </i>
&gt;<i> ident), I must disconnect the USB cable.</i>
&gt;<i> </i>
&gt;<i>  The command which run is:</i>
&gt;<i> </i>
&gt;<i>       /usr/bin/avrdude -p atmega2560 -c 2232HIO -P usb -vvvv</i>
&gt;<i> </i>
&gt;<i>  The command which wont run and gets stuck is:</i>
&gt;<i> </i>
&gt;<i>       /usr/bin/avrdude -p atmega2560 -c 2232HIO -P usb -vvvv -U </i>
&gt;<i> flash:w:devel.hex</i>
&gt;<i> </i>
&gt;<i>  The resulting logs are in attachment.</i>
&gt;<i> </i>
&gt;<i>  Any idea where the issue is?</i>
I believe the issue is related to the polling following a page write - the 
software reads the last byte of the page, until it is read back correctly, thus 
the page is programmed and we can continue to program the next page (or finish 
up). There are multiple problems with this code:
a) a page may not be programmed up until the last byte
b) a value 0xff is indistinguishable from un-programmed flash

I have worked on the page write code for a while now and my changes include a 
fix for both problems and I tested the code for a while now. I actually wanted 
to rework the flash read routine too and submit all changes together. However, 
I could submit a patch just fixing the write routine, if necessary.
To check, if this really is your problem you can compile avrdude with debug 
symbols and check in gdb. If I'm correct, a loop located in avrftdi.c:917 will 
never be left.

Best regards,
Hannes Weisbach
</pre><br class="Apple-interchange-newline">
</body>
</html>

mcuee, 2022-06-05:
Maybe the patch in the following issue can fix this.
* https://github.com/avrdudes/avrdude/issues/303

mcuee, 2022-06-08:
I can not write to my ATmega2560 either with the FT2232H based Jtagkey-2. Maybe I will check if it is because of this or becauses of #474.

* #474

mcuee, 2022-06-08:
I can confirm that my issue is with #474. The patch there fixed my issues with both read and write of ATmega2560 using FT2232H based JTAGKey-2.
* https://github.com/avrdudes/avrdude/issues/474#issuecomment-1149967953

mcuee, 2022-06-25:
Most likely this issue has been fixed and no longer valid.

mcuee, 2022-07-10:
I will close this as `invalid` for now. Please re-open when the issue occurs.

closed 2022-07-10 by mcuee