avrdudes/avrdude #365: [bug #43839] unable to verify device signature or program boot memory on xmega16d4 from ELF file

avrdude-admin, 2021-12-10:
Galen Seitz <None>
Tue 16 Dec 2014 07:14:21 PM UTC

device: xmega16d4
programmer: jtagice mkII or avrisp mkII
host: CentOS 6.6 x86_64
avrdude version: 6.1 (locally built)

I am unable to verify the device signature and write the boot section of an xmega16d4 with an ELF file.

galens@lion:~$ avr-objdump -v
GNU objdump (GNU Binutils) 2.19
Copyright 2007 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
galens@lion:~$ avr-objdump -h boot.elf

boot.elf:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
0 .text         00004d14  00000000  00000000  000000b4  2**1
CONTENTS, ALLOC, LOAD, READONLY, CODE
1 .fuse         00000006  00820000  00820000  00004dc8  2**0
CONTENTS, ALLOC, LOAD, DATA
2 .lock         00000001  00830000  00830000  00004dce  2**0
CONTENTS, ALLOC, LOAD, DATA
3 .signature    00000003  00840000  00840000  00004dcf  2**0
CONTENTS, ALLOC, LOAD, READONLY, DATA

galens@lion:~$ sudo avrdude -p x16d4 -c jtag2pdi -P usb -U signature:v:boot.elf:e

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9442
avrdude: NOTE: Programmer supports page erase for Xmega devices.
Each page will be erased before programming it, but no chip erase is performed.
To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: verifying signature memory against boot.elf:
avrdude: load data signature data from input file boot.elf:
avrdude: ERROR: Cannot handle "signature" memory region from ELF file
avrdude: read from file 'boot.elf' failed

avrdude done.  Thank you.

galens@lion:~$ sudo avrdude -p x16d4 -c jtag2pdi -P usb -U boot:w:boot.elf:e

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9442
avrdude: NOTE: Programmer supports page erase for Xmega devices.
Each page will be erased before programming it, but no chip erase is performed.
To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file "boot.elf"
avrdude: read from file 'boot.elf' failed

avrdude done.  Thank you.

I can successfully write and verify fuse1, fuse2, fuse4, fuse5, and lock using this same file.

Note that this same file works when used to program the bootloader and fuses of an xmega16d4 using AVR Studio version 4.19 using the ELF Production File feature.

[file #32685: boot.elf](https://savannah.nongnu.org/bugs/download.php?file_id=32685)

This issue was migrated from https://savannah.nongnu.org/bugs/?43839

mcuee, 2022-06-06:
This is probably related to #310
* https://github.com/avrdudes/avrdude/issues/310

mcuee, 2022-07-12:
I have finally figured out the connection of my AVRISP mkii clone to the MCUZone ATxmega32D4 break-out board (they are not using a standard PDI connector).

```
PS C:\work\avr\avrdude\avrdude-7.0_bin64> .\avrdude.exe -p atxmega32d4 -c avrispmkii -D -U signature:r:-:h

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9542 (probably x32d4)
avrdude.exe: reading signature memory:

Reading | ################################################## | 100% 0.01s

avrdude.exe: writing output file "<stdout>"
0x1e,0x95,0x42

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude\avrdude-7.0_bin64> .\avrdude.exe -p atxmega32d4 -c avrispmkii -D -U fuse1:r:-:h -U fuse2:r:-:h -U fuse4:r:-:h -U fuse5:r:-:h

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9542 (probably x32d4)
avrdude.exe: reading fuse1 memory:

Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "<stdout>"
0x0
avrdude.exe: reading fuse2 memory:

Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "<stdout>"
0xff
avrdude.exe: reading fuse4 memory:

Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "<stdout>"
0xfe
avrdude.exe: reading fuse5 memory:

Reading | ################################################## | 100% 0.00s

avrdude.exe: writing output file "<stdout>"
0xff

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude\avrdude-7.0_bin64> .\avrdude.exe -p atxmega32d4 -c avrispmkii
 -U flash:r:atxmega32d4_readback_mcuzone.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9542 (probably x32d4)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: reading flash memory:

Reading | ################################################## | 100% 0.89s

avrdude.exe: writing output file "atxmega32d4_readback_mcuzone.hex"

avrdude.exe done.  Thank you.

```

mcuee, 2022-07-12:
I can reproduce the issue.

```
PS C:\work\avr\avrdude\avrdude-7.0_bin64> .\avrdude -p x32d4 -c avrispmkii -U signature:v:boot.elf:e

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9542 (probably x32d4)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: verifying signature memory against boot.elf:
avrdude.exe: ERROR: Cannot handle "signature" memory region from ELF file
avrdude.exe: read from file 'boot.elf' failed

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude\avrdude-7.0_bin64> .\avrdude -p x32d4 -c avrispmkii -U signature:w:boot.elf:e

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9542 (probably x32d4)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: reading input file "boot.elf"
avrdude.exe: ERROR: Cannot handle "signature" memory region from ELF file
avrdude.exe: read from file 'boot.elf' failed

avrdude.exe done.  Thank you.
```

stefanrueger, 2023-01-06:
This is likely solved by PR #1206 @mcuee could you test?

mcuee, 2023-01-07:
Programming of the flash seems to work.

```
PS C:\work\avr\avrdude_test\avrdude_bin> avr-objdump -h boot.elf

boot.elf:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00004d14  00000000  00000000  000000b4  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .fuse         00000006  00820000  00820000  00004dc8  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .lock         00000001  00830000  00830000  00004dce  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  3 .signature    00000003  00840000  00840000  00004dcf  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -p atxmega32a4u -c avrispmkii -U boot:w:boot.elf:e

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file boot.elf for boot
         with 0 bytes in 0 sections within [0, -1]
         using 0 pages and 0 pad bytes
avrdude: writing 0 bytes boot ...

Writing | ################################################## | 100% 0.00 s

avrdude: 0 bytes of boot written
avrdude: verifying boot memory against boot.elf

Reading | ################################################## | 100% 0.00 s avrdude: 0 bytes of boot verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -p atxmega32a4u -c avrispmkii -U application:w:boot.elf:e

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file boot.elf for application
         with 19732 bytes in 1 section within [0, 0x4d13]
         using 78 pages and 236 pad bytes
avrdude: writing 19732 bytes application ...

Writing | ################################################## | 100% 1.78 s

avrdude: 19732 bytes of application written
avrdude: verifying application memory against boot.elf

Reading | ################################################## | 100% 0.38 s

avrdude: 19732 bytes of application verified

avrdude done.  Thank you.

```

mcuee, 2023-01-07:
@stefanrueger 

Writing of signaure failed. This may be related to other issues.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -p atxmega32a4u -c avrispmkii -U signature:r:-:h

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading signature memory ...
avrdude: writing output file <stdout>
0x1e,0x95,0x41

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -p atxmega32a4u -c avrispmkii -U signature:w:boot.elf:e

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file boot.elf for signature
         with 3 bytes in 1 section within [0, 2]
avrdude: writing 3 bytes signature ...
avrdude error: unknown memory signature
 ***failed;
avrdude error: unknown memory signature
 ***failed;
avrdude error: unknown memory signature
 ***failed;
avrdude: 3 bytes of signature written
avrdude: verifying signature memory against boot.elf
avrdude warning: verification mismatch
        device 0x1e != input 0x42 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

```

mcuee, 2023-01-07:
As for the signature verify, I think it works. My chip is different, so it is normal the verify failed.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -p atxmega32a4u -c avrispmkii -U signature:v:boot.elf:e

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9541 (probably x32a4u)
avrdude: Note: programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: verifying signature memory against boot.elf
avrdude warning: verification mismatch
        device 0x1e != input 0x42 at addr 0x0000 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

```

mcuee, 2023-01-07:
Close this issue as it is fixed.

closed 2023-01-07 by mcuee

MCUdude, 2023-01-07:
I'm pretty sure the signature is a read-only memory. So it's expected to fail when you try to write to it

mcuee, 2023-01-07:
> I'm pretty sure the signature is a read-only memory. So it's expected to fail when you try to write to it

You are of course right. The original issue is the failure to verify the signature from the elf file.