avrdudes/avrdude #366: [bug #43898] atmega644p remains stopped after JTAG transaction

avrdude-admin, 2021-12-10:
leres <None>
Sun 28 Dec 2014 07:42:04 PM UTC

Back when I upgraded from 5.11 to 6.0.1 I noticed that my atmega644p did not resume after programming or reading. I'm using an AVR Dragon.

I've tracked this down to revisions 1070 and 1071 of jtagmkII.c.  From the commit messages there appears to be an issue with the xmega.  The old code issued the CMND_GO when device_descriptor_length was set. The current code only isses a CMND_GO when AVRPART_HAS_PDI is set.

Attached is a patch vs 1353 that changes jtagmkII_close() to also issue CMND_GO when PGM_FL_IS_JTAG is set which gives me 5.X behavior however is likely wrong for the xmega case. I suspect the correct fix is to just avoid issuing a GO when the target is an xmega.

[file #32725: patch-jtagmkII.c](https://savannah.nongnu.org/bugs/download.php?file_id=32725)

This issue was migrated from https://savannah.nongnu.org/bugs/?43898

mcuee, 2022-05-29:
The code has not been changed in this aspect.
https://github.com/avrdudes/avrdude/blob/main/src/jtagmkII.c#L1843

I have not encountered this issue but I think I only use the AVR Dragon in ISP mode and not JTAG mode.

MCUdude, 2022-05-29:
I have a really flakey Dragon lying around, and I can confirm that this is still an issue and that the applied patch resolved the issue. @mcuee I'm about to leave the house, so if you don't want to wait, feel free to submit a PR!

(The Dragon is the most delicate programmer I own. The traces on the PCB are incredibly thin, the via pads ridiculously small, but the drilling accuracy is not good at all. I can only get JTAG to work if I toch the 2x5 pin connector on the underside with a wet finger. If not, I'm getting this error:

```
$ ./avrdude -p atmega644p -c dragon_jtag -Uhfuse:r:-:h

avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_JTAG_ID
avrdude: retrying with external reset applied
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_JTAG_ID
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
```

Go figure...)

mcuee, 2022-05-29:
> @mcuee I'm about to leave the house, so if you don't want to wait, feel free to submit a PR!

Haha, I am not a programmer and do not expect me to come out with any more complicated PRs. But I can help testing if I have the right devices.

MCUdude, 2022-05-29:
> Haha, I am not a programmer and do not expect me to come out with any more complicated PRs. But I can help testing if I have the right devices.

Never mind, I had little extra time. It was just a matter of applying the patch and submitting a PR, nothing fancy.

mcuee, 2022-05-29:
> The Dragon is the most delicate programmer I own

So far it seems to work for me with my devices: a few Arduino Uno clones (with ATmega16U2 or with CH340), one official Arduino Leonardo, two cheap ATtiny88 boards). 

I am ordering more boards -- official Arduino Nano Every with ATmega4809, Microchip AVR128DB48 Curiosity Nano and Microchip ATMEGA328PB-XMINI.

Back in 2011, I contacted Atmel to help Travis and I with some boards, when Travis was developing libusbk project. Atmel was very helpful to send the AVR Dragon and two USB development boards (AT90USBKEY  and UC3-A3 Xplained board). 

In the end Travis did develop the Benchmark USB FW example for the two AVR boards. He was more familiar with PIC at that time. Interestingly now Atmel is part of Microchip.
Ref: https://github.com/mcuee/libusbk/tree/master/BmFW

MCUdude, 2022-05-29:
> The Dragon is the most delicate programmer I own
So far it seems to work for me with my devices: a few Arduino Uno clones (with ATmega16U2 or with CH340), one official Arduino Leonardo, two cheap ATtiny88 boards).

Because of all the Arduino cores I've developed, I've aquired a wide variety of AVR chips and programmers over the years. Some bought, some donated and some borrowed from the place I work. Its especially nice to have all these weird and wonderful devices now that I'm slightly involved with the development of this project.

> In the end Travis did develop the Benchmark USB FW example for the two AVR boards. He was more familiar with PIC at that time. Interestingly now Atmel is part of Microchip.
Ref: https://github.com/mcuee/libusbk/tree/master/BmFW

Interesting. I have very little experience with USB in general, but some say USB is here to stay!

PR #979 (2022-06-26): Dragon JTAG fix
This simple fix resolves the issue where a connected AVR JTAG target doesn't start its program after programming has occurred. Fix has been tested on real hardware.

Closes #366 and #377
closed 2022-06-26 by dl8dtl