avrdudes/avrdude #367: [bug #43912] PATCH avrdude direct SPI mode NULL pointer crash

avrdude-admin, 2021-12-10:
David Fries <None>
Fri 02 Jan 2015 02:34:56 AM UTC

The AVR serial programmer protocol is SPI, so it would be nice to use avrdude to send and receive data over SPI for debugging while the microcontroller is up and running, seems someone else agreed and implemented a direct SPI terminal mode, however use USBtiny and avrdude crashes when giving spi to enter direct SPI mode.  The man page lists "Only supported on parallel bitbang programmers.", however the requirement really is to be able to be able to set RESET high (or float), and the circuit must allow SS (slave select) to be pulled low, but unlike RESET can be left low the entire time, making it easy enough to get around in the circuit.

I'm including a patch for both 5.x and 6 avrdude to fix the crash for any programmer that doesn't support setpin, and implementing support for USBtiny, the limitation is SS held low and transferring multiples of 4 bytes at a time.

I would prefer to use a "spi_send", "spi_bytes" or some command different from the "send" because it would be really easy to accidentally be in programming mode without the normal checks.

[file #32745: spi_fix_6.patch](https://savannah.nongnu.org/bugs/download.php?file_id=32745)
[file #32744: spi_fix_5.11.1.patch](https://savannah.nongnu.org/bugs/download.php?file_id=32744)

This issue was migrated from https://savannah.nongnu.org/bugs/?43912

avrdude-admin, 2021-12-10:
David Fries <dfries>
Sun 11 Jan 2015 02:19:41 AM UTC

At least with the USBtiny 2.0 there is a one byte SPI command, that could be used instead of requiring 4 bytes at a time.  It also supports bit banging any pin on that port.

andrewshadura, 2022-01-03:
Hi @dl8dtl, could you please merge this?
cc @dfries

dl8dtl, 2022-01-08:
The "crash on missing setpin method" bug has been fixed a few days ago.
If anyone thinks generic SPI support is useful beyond those ancient bitbang programmers, please open a new issue and/or provide a pull request.

closed 2022-01-08 by dl8dtl

andrewshadura, 2022-01-09:
Well, in that case I hope @dfries won’t mind if I rebase and port his patches to the latest master. I would actually like to see generic SPI support (I added it in the first place).