avrdudes/avrdude #368: [bug #44001] fuses can't read and write with TPI and buspirate in command mode

avrdude-admin, 2021-12-10:
Andreas Brietzke <brietzke@germanynet.de>
Wed 14 Jan 2015 09:40:20 AM UTC

Conditions:
Linux Mint 17
avrdude V 6.1
Buspirate V3
Attiny 10 with 5V
MOSI of Buspirate via 1k to MISO

MISO of Buspirate to TPI data
CLK of Buspirate to TPI clk
CS of Buspirate to TPI reset

If I send the command:

> avrdude -p t10 -P /dev/ttyUSB0 -c buspirate_bb -n -U fuse:r:-:h

then I will become a empty answer:

Attempting to initiate BusPirate bitbang binary mode...
BusPirate binmode version: 1
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.18s

avrdude: Device signature = 0x1e9003
avrdude: reading fuse memory:

Reading | ################################################## | 100% 0.12s

avrdude: writing output file "<stdout>"

avrdude done.  Thank you.

Writing a fuse doesn't work, too.

If I go in the terminal mode of the avrdude then reading and writing of the fuses works well.

This issue was migrated from https://savannah.nongnu.org/bugs/?44001

mcuee, 2022-05-29:
I do not have a bus pirate to test with, but I am wondering if this is still relevant or not. If this is true, then Bus Pirate users will probbaly complained a lot. Or maybe there are very few avrdude users who are using Bus Pirate.

MCUdude, 2022-05-29:
I happen to have a Buspirate v4.0 and an ATtiny10 breakout bor for testing. Everything seems to be working just fine on my mac.

```
$ ./avrdude -p attiny10 -P /dev/cu.usbmodem000000011 -c buspirate_bb -n -U fuse:r:-:h

Attempting to initiate BusPirate bitbang binary mode...
BusPirate binmode version: 1
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 1.86s

avrdude: Device signature = 0x1e9003 (probably t10)
avrdude: reading fuse memory:

Reading | ################################################## | 100% 0.62s

avrdude: writing output file "<stdout>"
0xff

avrdude done.  Thank you.
```

closed 2022-05-29 by MCUdude