avrdudes/avrdude #369: [bug #44170] Update/Rectify/Clarify Documentation

avrdude-admin, 2021-12-10:
funker211 <None>
Tue 03 Feb 2015 11:26:01 AM UTC

In the AVRDUDE Documentation ([http://www.nongnu.org/avrdude/user-manual/avrdude_1.html](https://savannah.nongnu.orghttp://www.nongnu.org/avrdude/user-manual/avrdude_1.html)) it says in paragraph 6, sentence 4: "These all use the MPSSE mode, which has a specific pin mapping. Bit 1 (the lsb of the byte in the config file) is SCK. Bit 2 is MOSI, and Bit 3 is MISO. Bit 4 usually reset."

So, Bit number 1 would be SCK, Bit number 2 MOSI, and so forth.

Now, when I look into the "avrdude.conf.in" (the config file used to generate the default configuration file when building), for all "avrftdi" type programmers, SCK has bit 0 assigned, MOSI has bit 1 assigned, and so forth. This kind of numeration is the common one, and what a user normally expects. It maps ADBUS0 to ADBUS7 to bits 0 to 7.

This is in conflict with what the documentation says. In my view, the documentation should be updated and rectified (or clarified that bit 1 means "the first bit", which has <strong>BIT NUMBER</strong> 0).

Now, look at the point where "id = "UM232H";" is defined. It assigns sck = 1, mosi = 2, miso = 3 and reset  = 4, so it uses the bit numbers of the documentation, not a zero-based numbering. Yet, when I look at the cited web page for this programmer, the hardware assignment is SCK = ADBUS0, MOSI = ADBUS1, MISO = ADBUS2 and RST = ADBUS3. Again, here is a zero-based numbering and the assignment (as I think) is wrong!

I assume that the definition of all "avrftdi" type programmers is correct and indeed, a zero-based numbering is required here as well. In this case, the documentation is wrong (or not precise) or the definition of the "UM2232H" is wrong. Did no one test this?

This is documentation issue 1 of 2 I came across. I will post issue 2 in another thread, in order to not mix them.

This issue was migrated from https://savannah.nongnu.org/bugs/?44170

mcuee, 2022-06-04:
Documentation link fixed:
https://www.nongnu.org/avrdude/user-manual/avrdude.html

> The avrftdi adds support for the FT2232C/D, FT2232H, and FT4232H devices. These all use the MPSSE mode, which has a specific pin mapping. Bit 1 (the lsb of the byte in the config file) is SCK. Bit 2 is MOSI, and Bit 3 is MISO. Bit 4 usually reset. The 2232C/D parts are only supported on interface A, but the H parts can be either A or B (specified by the usbdev config parameter).

mcuee, 2022-06-04:
This issue seems to be valid.

mcuee, 2022-06-08:
UM232H

```
programmer
  id         = "ft232h";
  desc       = "FT232H in MPSSE mode";
  type       = "avrftdi";
  connection_type = usb;
  usbvid     = 0x0403;
  usbpid     = 0x6014;
  usbdev     = "A";
  usbvendor  = "";
  usbproduct = "";
  usbsn      = "";
#ISP-signals
  sck    = 0; # AD0 (TCK)
  mosi   = 1; # AD1 (TDI)
  miso   = 2; # AD2 (TDO)
  reset  = 3; # AD3 (TMS)
;

# Pin J2-7 (AD0) is SCK
# Pin J2-8 (AD1) is MOSI
# Pin J2-9 (AD2) is MISO
# Pin J2-10 (AD3) is RESET
# Pin J2-6 is GND
# Use the -b flag to set the SPI clock rate eg -b 3750000 is the fastest I could get
# a 16MHz Atmega1280 to program reliably.  The 232H is conveniently 5V tolerant.
programmer parent "ft232h"
  id    = "um232h";
  desc  = "UM232H module from FTDI";
;

# Orange (Pin 2) is SCK
# Yellow (Pin 3) is MOSI
# Green (Pin 4) is MISO
# Brown (Pin 5) is RESET
# Black (Pin 10) is GND
# Use the -b flag to set the SPI clock rate eg -b 3750000 is the fastest I could get
# a 16MHz Atmega1280 to program reliably.  The 232H is conveniently 5V tolerant.
programmer parent "ft232h"
  id    = "c232hm";
  desc  = "C232HM cable from FTDI";
;

```

mcuee, 2022-06-11:
The latest documentation is still having the problem.
https://avrdudes.github.io/avrdude/7.0/avrdude.html

> The avrftdi adds support for the FT2232C/D, FT2232H, and FT4232H devices. These all use the MPSSE mode, which has a specific pin mapping. Bit 1 (the lsb of the byte in the config file) is SCK. Bit 2 is MOSI, and Bit 3 is MISO. Bit 4 usually reset.

mcuee, 2023-01-09:
@mariusgreuel 

Maybe you can help to fix this one. Thanks.

mcuee, 2023-01-10:
@dl8dtl 

Since you are touching the document, probably you can fix this one as well. Thanks.

closed 2023-01-14 by stefanrueger