avrdudes/avrdude #370: [bug #44171] Clarify Documentation on signal polarity

avrdude-admin, 2021-12-10:
funker211 <None>
Tue 03 Feb 2015 11:33:13 AM UTC

In the avrdude.conf(.in) file, it says that pin polarity can be inverted with a preceeding tilde (~) character.

Yet, standard polarity is not defined (or not clearly documented). For example, an AVR usually requires an active low reset signal. All avrftdi type programmers simply assign "reset = 3;". Now, is that an active high or active low signal? Somewhere, it should be stated what signal polarity actually is. Does "reset = 3;" mean that the reset is high active? Or does it mean it is (usually) active low and should I need an active high reset, I need to invert this by writing "reset = ~ 3;"?

Secondly, the file states that a space is required after the tilde. Yet, in the file and in the documentation, there are examples where this is omitted. Is a space REQUIRED or not?

I think the documentation really needs an update here, both in the manual AND in the avrdude.conf.in file.

This issue was migrated from https://savannah.nongnu.org/bugs/?44171

mcuee, 2022-06-04:
Ref: https://github.com/avrdudes/avrdude/blob/main/src/avrdude.conf.in#L44

This seems to be a valid documentation issue.

mcuee, 2023-01-09:
@stefanrueger 

Do you think this is still an issue or not with the latest avrdude.conf.in file? If not we can close this one.

stefanrueger, 2023-01-10:
@mcuee Yes, above is a great question:

| Does "reset = 3;" mean that the reset is high active? Or does it mean it is (usually) active low and should I need an active high reset, I need to invert this by writing "reset = ~ 3;"?

This requires clarification indeed. The pins are all treated the same in the grammar, but I don't know whether reset receives a special treatment in the code. @dl8dtl?

I vaguely remember having removed the claim that ~ must be followed by a space. 

Also throws up the deep philosophical question what `vcc = ~7;` could mean... :grin:

mcuee, 2023-03-26:
The issue kind of came up during the discussion of #1262.
* https://github.com/avrdudes/avrdude/issues/1262#issuecomment-1455089570

@mariusgreuel mentions the following.
> Just to be on the same page, the code I proposed assumes a negative-logic USB to TTL serial adapter, high-level for RTS idle, and of course low-level to reset the AVR.
>
> With a direct connection from RTS to /RESET, the idea is to start out with a high-level, and then briefly apply a low-level pulse.

WIthout knowing the above background, it was initially difficult for me and @stefanrueger to understand the code.

```
  // The code assumes a negative-logic of the USB to TTL serial adapter, idle-high, low to reset
  // Set RTS/DTR high to discharge the series-capacitor, if present
  serial_set_dtr_rts(&pgm->fd, 0);
  usleep(50*1000);
  // Pull the RTS/DTR line low
  serial_set_dtr_rts(&pgm->fd, 1);
  usleep(50*1000);
  // Set the RTS/DTR line back to high
  serial_set_dtr_rts(&pgm->fd, 0);
  usleep(50*1000);
```

mcuee, 2023-03-26:
On the other hand, when it comes to FT232H/2232H/4232H based programmers, the `~` sign means logic NOT gate.

Related discussion:
* https://github.com/avrdudes/avrdude/pull/1324

Example from KT-Link configuration in avrdude.conf.in.

```
#------------------------------------------------------------
# ktlink
#------------------------------------------------------------

# Kristech KT-LINK FT2232H interface with IO switching and voltage buffers.
# Created on 20220410 by CeDeROM Tomasz CEDRO (www.cederom.io).
# Interface DataSheet: https://kristech.pl/files/KT-LINK-UM-ENG.pdf
# AVRDUDE FT2232H PIN NUMBER DECODE:
#  | 0      | 1      | .. | 7      | 8      | 9      | .. | 15     |
#  | ADBUS0 | ADBUS1 | .. | ADBUS7 | ACBUS0 | ACBUS1 | .. | ACBUS7 |
# KT-LINK JTAG CONN:
#  1=Vsense(->EXT13), 19=5V(EXT1->EXT3), 20=GND, 3=TPIRST, 9=TPICLK, 7=TPIDATA.
# INTERNALS CONFIGURATION ("~" MEANS ACTIVE LOW):
#  ~TRST_EN=10(ACBUS2), ~CLK_EN=14(ACBUS6), ~SDO_EN=13(ACBUS5),
#  TMS_SEL=5(ADBUS5), ~TMS_EN=12(ACBUS4), LED=~15(ACBUS7).
# CONNECTION NOTES:
#  * Connect EXT connector pin 1 with 3 to get 5V on JTAG connector pin 19.
#  * Connect JTAG connector pin 1 to 5V (i.e. EXT pin 13 or JTAG pin 19).
#  * For TPI connection use resistors: TDO --[470R]-- TPIDATA --[470R]-- TDI.
#  * Powering target from JTAG pin 19 allows KT-LINK current measurement.

programmer
    id                     = "ktlink";
    desc                   = "KT-LINK FT2232H interface with IO switching and voltage buffers.";
    type                   = "avrftdi";
    prog_modes             = PM_TPI | PM_ISP;
    connection_type        = usb;
    usbvid                 = 0x0403;
    usbpid                 = 0xbbe2;
    usbdev                 = "A";
    buff                   = 5, ~10, ~13, ~14;
    reset                  = 8;
    sck                    = 0;
    sdo                    = 1;
    sdi                    = 2;
    rdyled                 = ~15;
;

```

[image]

cederom, 2023-03-26:
* KT-LINK interface is quite old interface designed and manufactured in Poland by Krzysztof Kajstura (KrisTech company).. but its probably not manufactured anymore, it was rocking solid when first steps of SWD were made on ARM Cortex... and I still use them for rapid prototyping and they work for over 10 years now :-)
* KT-LINK is FT2232H based which can be operated directly with libftdi in MPSSE or bitbang mode. 
* The picure in the comment above is a logical block diagram of the KT-LINK circuity. Aside from having FTDI chip it also has some buffers, multiplexers, and voltage translators that enabled safe operation on devices powered with 1,8 .. 5V power supply - there is a dedicated voltage sense pin to provide target reference voltage to the KT-LINK buffers and voltage shifters.
* This buffer map is specific only to the KT-LINK interface, other FTDI interfaces will probably have other buffer configuration, even no buffers/multiplexers at all. 
* Thanks to complex buffers it can act as various serial protocol interface (i.e. JTAG, SWD, I2C, SPI, UART, etc), but user/driver needs to take care of the buffer configuration and switching depending on the procotol used. 
* In normal operation user sets buffers to match electrical connections for a given interface only once on start. Then flow of bits may flow using mpsse. I guess this is the case in avrftdi. In other cases with other protocols and software it is possible to switch buffers and multiplexers in the middle of read-write operations to match the desired bus.
* Regarding the question on polarity - this may not involve physical pin state but a logical pin state - that is if signal is active high (no `~`) then setting it active will result in high voltage on a pin, while signal that is active low (`~` present) will result in no voltage when active (so called inverted logic).
* Regarding the configuration syntax, I guess this is simple `,` ` ` and `~` distinction to make life easier for the parser.
* Here are some examples for MPSSE: https://ftdichip.com/software-examples/mpsse-projects/.
* Here are some examples for LibFTDI: https://github.com/lipro/libftdi/tree/master/examples.

Luckily I left a note on the numbers decode for the `avrftdi` driver:
```
# AVRDUDE FT2232H PIN NUMBER DECODE:
#  | 0      | 1      | .. | 7      | 8      | 9      | .. | 15     |
#  | ADBUS0 | ADBUS1 | .. | ADBUS7 | ACBUS0 | ACBUS1 | .. | ACBUS7 |
```

Here is an example led activity configuration:
```
rdyled = ~15;
```
This signal marks "ready" operation, by asserting signal active when ready, and because LED is connected to the circuit with Cathode that will emit light when `GPIOH7`/`ACBUS7` line goes low, then we need to set logic active signal as low, thus use of `~` so line will be set to `0` when active and `1` when inactive (inverted logic).

Here is an example buffer configuration:
```
buff = 5, ~10, ~13, ~14;
```
* In order to enable `TCK` output buffer, the `GPIOH6`/`ACBUS6` pin needs to be put low. `GPIOH6`/`ACBUS6` is number `14` (see note for mapping above). We want the inverted logic (active low) so we provide `~14`. Now `TCK` will act as the clock and will flow out of the FTDI to the target.
* In order to enable `TDI` output buffer, the `GPIOH5`/`ACBUS5` pin needs to be put low. `GPIOH5`/`ACBUS5` is number `13`. We want the inverted logic (active low) so we provide `~13`. Now `TDI` will provide bitstream out of the FTDI to the target.
* In oder to enable `TRST` output buffer, the `GPIOH2`/`ACBUS2` pin needs to be put low. `GPIOH2`/`ACBUS2` is number `10`. We want inverted logic (active low) so we provide `~10`. Now `TRST` reset signal will flow out of the FTDI to the target reset pin.
* Signal number `5` is standard noninverted logic and sets `TDI`/`TMS` (output) and `TDO`/`TMS` (input) multiplexers. This provides standard operation. The multiplexers are there in order to enable read-write operations of several signals multiplexed into one wire (like SWD) but that would also require special driving between read-write (or other operations) by the driver. This requires also special (onboard / onchip) circuity on the target. That is not the case here, so we set those pins to default configuration.

I hope that helps :-)

cederom, 2023-03-26:
> @mcuee Yes, above is a great question:
>
> This requires clarification indeed. The pins are all treated the same in the grammar, but I don't know whether reset receives a special treatment in the code. [...]

I do not the code that well, but `reset = 3;` means active high, while `reset = ~3;` would mean active low.

> I vaguely remember having removed the claim that ~ must be followed by a space.

Looking at my configuration there is no space after `~`. Maybe a special case is constructed for a list of signals that will be all active low after the `~` with space?

> Also throws up the deep philosophical question what `vcc = ~7;` could mean... grin

That would mean there is a signal named `vcc` that is attached to `ADBUS7` and is active low. Probably some buffer driver enable signal specific to a  particular interface hardware?

mcuee, 2023-03-27:
@cederom 

Thanks a lot for the insight.

So the pin number in the `buf` pin has a sequence for KT-Link. How does the code know about the buf pin for other adapters?
1) 1st pin is for `tdi/tms` or `tdo/tms` multiplexer
2) 2nd pin is for `trst` output buffer
3) 3rd pin is for `tdi` output buffer
4) 4th pin is for `tck` output buffer

mcuee, 2023-03-27:
[image]

This is the example for JTAGkey (including JTAGkey 2).

```

programmer parent "ft2232h_jtag"
    id                     = "jtagkey";
    desc                   = "Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2 (JTAG)";
    usbpid                 = 0xcff8;
    buff                   = ~4, 10, ~11;
    reset                  = 10;
;
```

cederom, 2023-03-27:
> So the pin number in the `buf` pin has a sequence for KT-Link. How does the code know about the buf pin for other adapters?

Each adapter needs its own configuration definition, unless its buffer is the same as for the existing interface configuration. The provided example and explanation is valid only for the KT-LINK. Different interfaces with different buffers configuration will need their own configuration.

However, there are common pins to all FT2232 interfaces: TCK, TMS, TDI, TDO, RST.

cederom, 2023-03-27:
> [image]
> 
> This is the example for JTAGkey (including JTAGkey 2).
> 
> The following proposal does not seem to be right then.
> 
> ```
> programmer parent "ft2232h_jtag"
>     id                     = "jtagkey";
>     desc                   = "Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2 (JTAG)";
>     usbpid                 = 0xcff8;
>     buff                   = ~4, 10, ~11;
>     reset                  = 10;
> ;
> ```

This is what I would do (I should have this jtagkey still somewhere maybe older release):

```
programmer
    id                     = "jtagkey";
    desc                   = "Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2 (JTAG).";
    type                   = "avrftdi";
    prog_modes             = PM_TPI | PM_ISP; <--- PM_JTAG ???
    connection_type        = usb;
    usbvid                 = 0xXXYY; <-- please provide VID
    usbpid                 = 0xcff8;
    usbdev                 = "A";
    buff                   = ~4, ~11;
    reset                  = 9;
    sck                    = 0;
    sdo                    = 1;
    sdi                    = 2;
    rdyled                 = ~14;
;
```

* With the above configuration you will use `SRST_IN` as the system reset (number `9` because its `ACBUS1`/`GPIOH1`).
* `buff = ~4, ~11;` because you are using `ADBUS4`/`GPIOL0` (`~4` active low pin) and `ACBUS3`/`GPIOH3` (`~11` active low pin) to enable output buffers for the data signals.
* You need to use `reset=9` because you want to use `ACBUS1`/`GPIOH1` as the reset signal.
*  `reset=10` will give you `ACBUS2` as the reset signal, that is `TRST_N` buffer enable pin, which is invalid and would not produce valid reset signal to your target, it will only enable output of the `ACBUS0` and whatever state it is in.
* You may also use `ACBUS0` as the system reset pin (`TRST_N`), not to interfere with "ready led", then you would set `reset = 8` and `buff = ~4, ~10`.

Let me know if that works :-)

mcuee, 2023-03-27:
@cederom 

Thanks a lot for the explanation.

Basically I am having some difficulties to the avrdude codes handling the different FTDI adapter configurations. OpenOCD is pretty flexible.

avrdude as of now is quite rigid and it is okay now since it only supports ISP and it only needs to deal with 4 pins.

The following code is from `avrftdi.c`.
```
enum { FTDI_SCK = 0, FTDI_SDO, FTDI_SDI, FTDI_RESET };
```

PR #1342 have to touch many files, including `pindef.c` to work around. It also removes the above line.
https://github.com/avrdudes/avrdude/pull/1324/files

```
/**
 * Returns the name of the pin as string.
 *
 * @param pinname the pinname which we want as string.
 * @returns a lowercase string with the pinname, or <unknown> if pinname is invalid.
 */
const char * avr_pin_lcname(int pinname) {
  switch(pinname) {
  case PPI_AVR_VCC   : return "vcc";
  case PPI_AVR_BUFF  : return "buff";
  case PIN_AVR_RESET : return "reset";
  case PIN_AVR_SCK   : return "sck";
  case PIN_AVR_SDO   : return "sdo";
  case PIN_AVR_SDI   : return "sdi";
  case PIN_JTAG_TCK  : return "tck";
  case PIN_JTAG_TDI  : return "tdi";
  case PIN_JTAG_TDO  : return "tdo";
  case PIN_JTAG_TMS  : return "tms";
  case PIN_LED_ERR   : return "errled";
  case PIN_LED_RDY   : return "rdyled";
  case PIN_LED_PGM   : return "pgmled";
  case PIN_LED_VFY   : return "vfyled";
  default : return "<unknown>";
  }
}
```

My confusion is more with the dealing of the `buff` pin, especially if there are multiple `buff` pins.

cederom, 2023-03-27:
* No worries, this `buff` stuff is really simple, it just sets provided pins high (low if `~` is in use) in order to enable output buffers so the signal can flow from/to the FTDI chip and the target.
* I am not really familiar with existing code, but it should provide JTAG connectivity out of the box, I would not touch it, remove anything, etc, because there is a high risk that existing code will stop working.
* In the code example above you can see the `clear_pin()` function that sets logic level to inactive, and there should be somewhere `set_pin()` function that would set logic level to active, thus straight or inverted logic is possible.

mcuee, 2023-03-27:
@cederom 

Please help to test and review PR #1324 when you get the AVR JTAG target. Thanks a lot in advance.
* https://github.com/avrdudes/avrdude/pull/1324

As for JTAGkey, I have no issues with ISP.

For JTAG, as per @jkent, target reset is not needed. So I am fine with the following simple config, But I am having trouble with PR #1324 with strange readback failure. I will try your mod as well. But I am not positive your mod will help.

```
#------------------------------------------------------------
# jtagkey_jtag
#------------------------------------------------------------

programmer parent "ft2232h_jtag"
    id                     = "jtagkey_jtag";
    desc                   = "Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2 (JTAG)";
    usbpid                 = 0xcff8;
    buff                   = ~4;
;
```

That simply means the following.

```
    buff                   = ~4;
    tck                    = 0; # AD0 (TCK)
    tdi                    = 1; # AD1 (TDI)
    tdo                    = 2; # AD2 (TDO)
    tms                    = 3; # AD3 (TMS)

```

stefanrueger, 2023-06-01:
Is this still an issue, and if so, which piece of documentation should be changed? If not, perhaps best closing this?

mcuee, 2023-06-02:
I think we can close this now.

closed 2023-06-02 by mcuee