avrdudes/avrdude #372: [bug #44218] linuxgpio not working with raspberryPi 2. "Can't open gpioX/direction" : Permission denied. Fixed by little delay

avrdude-admin, 2021-12-10:
Rizlik <rizlik@inventati.org>
Mon 09 Feb 2015 11:31:25 PM UTC

Hi,

When trying to use avrdude with my raspberryPi2 using linuxgpio programmer I got:

"Can't open gpioX/direction" : Permission denied

When stepping the code in gdb the problem is not reproduced, so I thought that the problem is that avrdude try to change direction too fast after he exported the gpio pin.

Indeed adding a little delay before trying to switch direction fixes the problem. I attach a little patch that solve the problem for me.

Same problem is shown by

$ echo 2 > /sys/class/gpio/export; echo in > /sys/class/gpio2/direction

So isn't a problem of avrdude.

Hope it helps,
rizlik

[file #33041: gpio_fix.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33041)

This issue was migrated from https://savannah.nongnu.org/bugs/?44218

mcuee, 2022-06-06:
Yes #917 seems to be the same as #917 and #472.
* #472
* #917

mcuee, 2022-07-10:
I will mark this as duplicate of #386 and the fix is in #917.

closed 2022-07-10 by mcuee