avrdudes/avrdude #373: [bug #44248] trunk version doesn't work with ch340g in linux

avrdude-admin, 2021-12-10:
neeo <neeo@xl.wp.pl>
Sat 14 Feb 2015 04:56:30 PM UTC

There're lots of arduino clones using ch340g usb converter - which uses ch341 linux driver. However latest (newer than 6.0.1) avrdude doesn't get serial settings from these devices (with error "Cannot get serial port settings").

The function ser_setspeed tries to get settings using ioctl(fd->ifd, TIOCGSERIAL, &ss) - unfortnatelly the ch341 (nor the ch34x from the chipset vendor) doesn't implement any ioctls.

avrdude 6.0.1 (and older) doesn't do all the ifdef magic for different OS, and it works as expected.

This issue was migrated from https://savannah.nongnu.org/bugs/?44248

avrdude-admin, 2021-12-10:
Jon Escombe <jescombe>
Mon 13 Jul 2015 09:20:42 AM UTC

This behaviour was introduced by **[patch #8380](https://savannah.nongnu.org/patch/?8380)**, in SVN revision 1351.

I've run into the same issue trying to use an STK500 with a Linux laptop, as none of my USB-serial converters support these ioctls either.

MCUdude, 2022-03-19:
I'm closing this issue because patch 8380 was later reverted because it caused lots of issues like this.

closed 2022-03-19 by MCUdude