avrdudes/avrdude #376: [bug #44539] Avoid USB reconnect when running in VM

avrdude-admin, 2021-12-10:
Michael Lipp <mnl@mnl.de>
Sat 14 Mar 2015 01:51:51 PM UTC

This is rather an enhancement request than a bug report.

When executing a sign off command, jtagmkII and dragion do a hard reset by default. This also implies re-discovery of the usb device. While this isn't a big problem when attaching those devices to your computer directly, it becomes a big problem when using avrdude in a Virtual Machine.

There, on every sign-off, the hosts detects the removal of a USB device, signals it to the guest, then detects a new USB device, and signals it to the guest. This can take up to 5 seconds.

Both devices offer the possibility to modifiy the behaviour of sign-off to a "soft reset", thus avoiding that lengthy cycle. The appended patch forces this by passing "-x soft-reset" on the command line.

[file #33340: avrdude-soft-reset.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33340)

This issue was migrated from https://savannah.nongnu.org/bugs/?44539

mcuee, 2022-05-29:
Interesting question. With avrdude being cross-platform, I guess this is probably a niche use case.

stefanrueger, 2023-06-12:
Time has moved on, and we don't know whether this is still a desirable feature. Maybe close the issue for the time being?

mcuee, 2023-06-12:
Yes I agree.

closed 2023-06-12 by mcuee