avrdudes/avrdude #377: [bug #44540] Broken reset mode in jtagmkII_program_enable

avrdude-admin, 2021-12-10:
Michael Lipp <mnl@mnl.de>
Sat 14 Mar 2015 01:54:50 PM UTC

I think the intended behaviour of trying to enter programming mode first without external reset and then with external reset is broken. The attached patch should fix this.

[file #33341: avrdude-progmode.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33341)

This issue was migrated from https://savannah.nongnu.org/bugs/?44540

MCUdude, 2022-03-28:
I'm not able to reproduce the issue. It may be because my Dragon programmer is a bit flakey at times. I'm able to program the device over JTAG, but I'm always getting the following error at the end, and the program starts immediately.

```
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_FAILED
```

mcuee, 2022-05-28:
I have not tried jtag node, but dragon_isp mode works fine with my AVR Dragon (tested with ATtiny2313, ATmega328P and ATmega32U4)
Ref: https://github.com/avrdudes/avrdude/issues/955#issuecomment-1123676668

mcuee, 2022-06-20:
The patch is copied here for easier checking.
```
*** avrdude-6.1/jtagmkII.c.orig	2015-03-14 11:28:05.991037948 +0100
--- avrdude-6.1/jtagmkII.c	2015-03-14 12:46:03.227272372 +0100
***************
*** 1194,1204 ****
--- 1194,1208 ----
    int status;
    unsigned char buf[1], *resp, c;
    int use_ext_reset;
+   unsigned char parm[1];
  
    if (PDATA(pgm)->prog_enabled)
      return 0;
  
    for (use_ext_reset = 0; use_ext_reset <= 1; use_ext_reset++) {
+     parm[0] = use_ext_reset;
+     (void)jtagmkII_setparm(pgm, PAR_EXTERNAL_RESET, parm);
+ 
      buf[0] = CMND_ENTER_PROGMODE;
      if (verbose >= 2)
        fprintf(stderr, "%s: jtagmkII_program_enable(): "
***************
*** 1223,1249 ****
        fprintf(stderr, "0x%02x (%d bytes msg)\n", resp[0], status);
      c = resp[0];
      free(resp);
!     if (c != RSP_OK) {
!       fprintf(stderr,
! 	      "%s: jtagmkII_program_enable(): "
! 	      "bad response to enter progmode command: %s\n",
! 	      progname, jtagmkII_get_rc(c));
!       if (c == RSP_ILLEGAL_JTAG_ID) {
! 	if (use_ext_reset == 0) {
! 	  unsigned char parm[] = { 1};
! 	  if (verbose > 0)
! 	    fprintf(stderr,
! 		    "%s: retrying with external reset applied\n",
! 		    progname);
! 
! 	  (void)jtagmkII_setparm(pgm, PAR_EXTERNAL_RESET, parm);
! 	  continue;
! 	}
  
! 	fprintf(stderr, "%s: JTAGEN fuse disabled?\n", progname);
! 	return -1;
        }
      }
    }
  
    PDATA(pgm)->prog_enabled = 1;
--- 1227,1258 ----
        fprintf(stderr, "0x%02x (%d bytes msg)\n", resp[0], status);
      c = resp[0];
      free(resp);
!     if (c == RSP_OK) {
!       break;
!     }
!     fprintf(stderr,
!             "%s: jtagmkII_program_enable(): "
! 	    "bad response to enter progmode command: %s\n",
! 	    progname, jtagmkII_get_rc(c));
!     if (c == RSP_ILLEGAL_JTAG_ID) {
!       if (use_ext_reset == 0) {
! 	if (verbose > 0)
! 	  fprintf(stderr,
! 		  "%s: retrying with external reset applied\n",
! 		  progname);
  
! 	(void)jtagmkII_setparm(pgm, PAR_EXTERNAL_RESET, parm);
! 	continue;
        }
+ 
+       fprintf(stderr, "%s: JTAGEN fuse disabled?\n", progname);
      }
+     return -1;
+   }
+ 
+   if (use_ext_reset) {
+     parm[0] = 0;
+     (void)jtagmkII_setparm(pgm, PAR_EXTERNAL_RESET, parm);
    }
  
    PDATA(pgm)->prog_enabled = 1;

```

MCUdude, 2022-06-20:
My [jtagmkii stick](https://www.aliexpress.com/item/32961233532.html) is a total mess and doesn't work properly. However, the AVR dragon uses the jtagmkii protocol, so I guess we can test this patch using the AVR Dragon.

For reference, PR #979 resolves an issue where the program isn't started after programming using JTAG. It may even be that this issue isn't relevant anymore if/when #979 gets merged.

MCUdude, 2022-06-20:
@mcuee I haven't tried the applied patch, but I can confirm that with #979 applied, the program starts automatically after flashing, even with the OCDEN fuse set. So from what I can see, this issue can be closed as soon as #979 gets merged.

But when I look at the patch, I find _very_ similar code in the jtagmkii.c file. Perhaps this has been added in the past, and later re-formatted to replace `fprintf` with `avrdude_message`, among otehr things? 

```
$ ./avrdude -cdragon_jtag -patmega128 -Ulfuse:r:-:h -Uhfuse:r:-:h -Uefuse:r:-:h

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9702 (probably m128)
avrdude: reading lfuse memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0xbf
avrdude: reading hfuse memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0x7
avrdude: reading efuse memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0xff

$ ./avrdude -cdragon_jtag -patmega128 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_54198/Blink.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9702 (probably m128)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_54198/Blink.ino.hex"
avrdude: writing flash (1202 bytes):

Writing | ################################################## | 100% 0.08s

avrdude: 1202 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_54198/Blink.ino.hex:

Reading | ################################################## | 100% 0.12s

avrdude: 1202 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-06-21:
@MCUdude 
> But when I look at the patch, I find very similar code in the jtagmkii.c file. Perhaps this has been added in the past, and later re-formatted to replace fprintf with avrdude_message, among otehr things?

You are exactly right. This can be closes.

closed 2022-06-21 by mcuee