avrdudes/avrdude #38: [bug #14374] Feature Request : terminal mode : improving EEPROM writes

avrdude-admin, 2021-12-10:
Vincent Trouilliez <vincent.trouilliez@modulonet.fr>
Fri 02 Sep 2005 04:08:14 PM UTC

I love to use the terminal mode to look at the content of the EEPROM, the data is very niceley presented.
It is so convenient to modify the EEPROM quickly/directly without having to make a hex file and download it !

So, in order to extend the usefulness of this command, it would be great to make its parser more clever/flexible.

For example we could type in one single command :

write eeprom 0x0000 "A string",0,"another string",0,2.456,-456.76,16036,-113

A bit like when declaring constants with the DB DW etc, in an assembler. The more data formats the parser could understand, the more marvelous/useful/practical/magic it would be ! :o)

This issue was migrated from https://savannah.nongnu.org/bugs/?14374

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sat 17 Sep 2005 07:45:36 PM UTC

Without a patch, that's going to be rather low-piority, as
the requested feature while being useful, might require a
bit of thought before.

MCUdude, 2022-02-28:
Implemented in #880

closed 2022-02-28 by MCUdude