avrdudes/avrdude #382: [bug #44848] OS X and avrmkII, again

avrdude-admin, 2021-12-10:
jesid <None>
Wed 15 Apr 2015 03:18:01 PM UTC

Hi

this bug is related to my previous bugfix #41357.

I recently started to reuse my old avrmkII to flash ATXmegas.
Every second flashing attempt reports exactly 2 errors:

avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_recv_mk2: error in USB receive

After that one execution works fine before everything starts over again...

Everything works fine. It just takes a lot of time to complete the task.

I recommend to undo my bugfix #41357. usb_reset() removes this errors and speeds up the deployment. Since my last bugfix everything got updated. Now OS X does not seem to be sensible to usb_reset() anymore. Well, at least on my system... Honestly, I have no clue what has changed.

BTW: I am using: avrdude 6.1, latest OS X, and latest avrmkII firmware.

A quick note about the avrmkII->atxmega programming performance:
In the error-less case it takes about 6 sec for the devices to get in sync. After that it works fast as hell. I don't know whether the processor, the avrmkII, or avrdude is to blame for. Just wanted to mention this. Here is a log including some timing notes:

avrdude -px128a1 -cavrispmkII -Uflash:w:../00_EclipseWorkspaces/AVR/ATxmegaTest/Release/ATxmegaTest.hex:a -vvvv

avrdude: Version 6.1, compiled on Apr 15 2015 at 01:36:32
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/usr/local/etc/avrdude.conf"
User configuration file is "/Users/sid/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : avrispmkII
avrdude: usbdev_open(): Found AVRISP mkII, serno: 000200100981

{{{{{{{{ waits for about 1sec }}}}}}}}}}

avrdude: Sent: . [01]
avrdude: Recv: . [01] . [00] . [0a] A [41] V [56] R [52] I [49] S [53] P [50] _ [5f] M [4d] K [4b] 2 [32]
avrdude: stk500v2_getsync(): found AVRISP mkII programmer

{{{{{{{{ waits for about 5sec }}}}}}}}}}

AVR Part                      : ATxmega128A1
Chip Erase delay              : 0 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 0
StabDelay                     : 0
CmdexeDelay                   : 0
SyncLoops                     : 0
ByteDelay                     : 0
PollIndex                     : 0
PollValue                     : 0x00
Memory Detail                 :

{{{{{{{{ from here on everything is fast as usual }}}}}}}}}}

Thanks, awesome work!
Juergen

This issue was migrated from https://savannah.nongnu.org/bugs/?44848

avrdude-admin, 2021-12-10:
Hisashi ITO <orangkucing>
Wed 05 Aug 2015 02:47:17 PM UTC

Maybe this patch will work for your mac, too.

[https://trac.macports.org/ticket/41654](https://savannah.nongnu.orghttps://trac.macports.org/ticket/41654)

avrdude-admin, 2021-12-10:
Richard Falk <richfalk>
Thu 15 Oct 2015 07:26:40 AM UTC

I have found that the problem occurs due to some missing cleanup (closing?) at the end of execution because the first time avrdude 6.1 is run the problem does not occur, but then it occurs on every subsequent execution.  If I then run avrdude 6.0.1 the problem still occurs only for that execution while every subsequent 6.0.1 execution is fine.  If I then run 6.1 again it is OK the first time, but every subsequent 6.1 execution shows the problem.

Perhaps the following comment on another forum may be helpful:

[http://www.avrfreaks.net/comment/1404011#comment-1404011](https://savannah.nongnu.orghttp://www.avrfreaks.net/comment/1404011#comment-1404011)

since it describes a possible missing handleClose.

avrdude-admin, 2021-12-10:
Richard Falk <richfalk>
Thu 15 Oct 2015 07:40:14 AM UTC

I just tested the patch mentioned in **[comment #1](https://savannah.nongnu.org#comment1)** (from Hisashi ITO) and it works to fix the problem for me so that seems to be the source of the problem so ignore my **[comment #2](https://savannah.nongnu.org#comment2)** and instead apply the simple patch to the file stk500v2.c.

avrdude-admin, 2021-12-10:
Richard Falk <richfalk>
Thu 15 Oct 2015 07:55:52 AM UTC

Sorry, ignore my **[comment #3](https://savannah.nongnu.org#comment3)**.  I got fooled by the ping-pong nature of this bug.  The patch referred to in **[comment #1](https://savannah.nongnu.org#comment1)** did not fix the problem.  My **[comment #2](https://savannah.nongnu.org#comment2)** is still relevant.

MCUdude, 2022-03-29:
I'm running MacOS 10.14 and connected an AVRISPmkII programmer to an ATxmega128A3 target, and is not able to reproduce the issue with the latest upstream Avrdude version. Please open a new issue if this still is a problem for you.

closed 2022-03-29 by MCUdude