avrdudes/avrdude #387: [bug #45561] Error writing to the address type xxx00

avrdude-admin, 2021-12-10:
Kasatky <kasatky@rambler.ru>
Thu 16 Jul 2015 11:58:21 AM UTC

System: WinXP SP3, Celeron 2.14 GHz, i865PE
AVRDUDE version: 6.1
Target: Atmega 1284P
Programmer: STK200 (4 resistors)

Reading is always without errors.
Error writing only at the beginning of page, in the addresses xxx00, that are multiples of 256 or 512.
Delay (-i) does not give the desired effect at any values, but significantly reduces the write speed.
If I use other programs with the programmer, writing passes without errors.

d:\avrdude\avrdude.exe -c stk200 -p m1284p -P lpt1 -i 10 -U flash:w:"D:\true.bin":r

The files in the attachment example.zip:
true.bin - recorded file
false.bin - read file

Thanks for attention!

[file #34452: example.zip](https://savannah.nongnu.org/bugs/download.php?file_id=34452)

This issue was migrated from https://savannah.nongnu.org/bugs/?45561

mcuee, 2022-05-29:
This is probably not relevant any more as Windows XP is no longer supported, giveio driver has been removed, lpt based programmer will no longer works.

MCUdude, 2022-05-29:
Thanks @mcuee !

closed 2022-05-29 by MCUdude