avrdudes/avrdude #391: [bug #45826] ATMEL-ICE content mismatch on Linux and OS X

avrdude-admin, 2021-12-10:
Daniel Kotschate <None>
Tue 25 Aug 2015 06:10:08 PM UTC

Unfortunately unable to upload any hex file  into several different MCUs and different protocols (JTAG, PDI and ISP). Device (ATMEL-ICE) works on Windows 7 Pro with Atmel Studio 6.2 perfectly.

Reading seems to works on OS X and Linux (no error message.) Write back to device results in content mismatch message.

Tested with atmega32u4, atxmega128a1 (rev. h) and atxmega256a3bu.

[file #34727: log_pdi.txt](https://savannah.nongnu.org/bugs/download.php?file_id=34727)
[file #34726: log_jtag.txt](https://savannah.nongnu.org/bugs/download.php?file_id=34726)

This issue was migrated from https://savannah.nongnu.org/bugs/?45826

mcuee, 2022-06-06:
Not so sure this is still relevant or not. It involved non-xmega parts -- the ATmega32U4 and two xMega parts.

MCUdude, 2022-08-19:
Sorry for taking so long to look at this. It's probably not an issue anymore. At least, I can't reproduce it on my mac:

```
$ srec_cat -generate 0x00 0x1ffff -repeat_data 0x55 -o 0x55_128kib.hex -Intel

$ ./avrdude -catmelice_pdi -patxmega128a3u -Uflash:w:0x55_128kib.hex:i -v

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/hans/Downloads/avrdude-terminal/src/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : atmelice_pdi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATxmega128A3U
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         50   50      0     0     0 0x00 0x00
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no       2048   32      0     0     0 0x00 0x00
           application             0     0     0    0 no     131072  512      0     0     0 0x00 0x00
           apptable                0     0     0    0 no       8192  512      0     0     0 0x00 0x00
           boot                    0     0     0    0 no       8192  512      0     0     0 0x00 0x00
           flash                   0     0     0    0 no     139264  512      0     0     0 0x00 0x00
           usersig                 0     0     0    0 no        512  512      0     0     0 0x00 0x00
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_PDI
         Description     : Atmel-ICE (ARM/AVR) in PDI mode
         ICE HW version  : 0
         ICE FW version  : 1.42 (rel. 161)
         Serial number   : [xxxxxxxx]
         Vtarget         : 3.30 V
         JTAG clock megaAVR/program   : 1500 kHz
         JTAG clock megaAVR/debug     : 200 kHz
         JTAG clock Xmega             : 1000 kHz
         PDI/UPDI clock Xmega/megaAVR : 500 kHz

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9742 (probably x128a3u)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file 0x55_128kib.hex for flash
         with 131071 bytes in 1 section within [0, 0x1fffe]
         using 256 pages and 1 pad byte
avrdude: writing 131071 bytes flash ...

Writing | ################################################## | 100% 7.47s

avrdude: 131071 bytes of flash written
avrdude: verifying flash memory against 0x55_128kib.hex

Reading | ################################################## | 100% 4.47s

avrdude: 131071 bytes of flash verified

avrdude done.  Thank you.
```

MCUdude, 2022-08-19:
I'm closing this issue as it's probably no longer relevant. Please open a new issue if this is still a problem.

closed 2022-08-27 by mcuee