avrdudes/avrdude #392: [bug #46005] arduino auto-reset invocation is outside of retry loop

avrdude-admin, 2021-12-10:
Bill Wesfield <westfw@gmail.com>
Sun 20 Sep 2015 08:49:24 AM UTC

The new avrdude will attempt the upload several times, but apparently the arduino-specific code to toggle DTR (and perform auto-reset) is outside of the retry loop, which means that the board is not reset before each try. On Optiboot-based boards like Uno, which have a one-second bootloader timeout, this will mean that the board has long-since run the application.
On the other hand, sometimes the retries are quick (wrong response, rather than no response?), perhaps giving the host time to flush buffered data or noise, which might be effective...

[file #36907: invert_dtr.patch](https://savannah.nongnu.org/bugs/download.php?file_id=36907)

This issue was migrated from https://savannah.nongnu.org/bugs/?46005

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sat 31 Oct 2015 08:27:52 PM UTC

Sorry, it is not quite clear to me which part of the code
(nor even which programmer implementation) you are talking
about.  In particular, I have no idea what might trigger the
DTR toggling on Arduino.

Can you please elaborate a little more?  (I don't have an
Arduino at hand to tinker with.)

avrdude-admin, 2021-12-10:
Michal Suchanek <hramrach1>
Tue 12 Apr 2016 01:05:33 PM UTC

Hello,

I tried to fix the DTR handling for Arduino Pro Mini which requires that the DTR is pulled and released before doing the programming. For UNO holding the DTR whole time works but not for Pro Mini.

Did not actually retry with UNO to test if holding the DTR is required or there is a circuit that just cuts it after a moment.

However, the patch which tries both ways in turn shows how the DTR handling is not part of the loop.

(**[file #36907](https://savannah.nongnu.org/support/download.php?file_id=36907)**)

MCUdude, 2022-03-25:
Fixed in #855

closed 2022-03-25 by MCUdude