avrdudes/avrdude #395: [bug #46092] xmega128d4 flash page size wrong

avrdude-admin, 2021-12-10:
jstutch <None>
Thu 01 Oct 2015 01:24:32 PM UTC

avrdude.conf lists the flash page size for atxmega128d4 incorrectly. This leads to corruption when writing.

Datasheet shows 256b flash page size, avrdude.conf currently inherits 512b.

Patch attached.

[file #35021: avrdude.conf.patch](https://savannah.nongnu.org/bugs/download.php?file_id=35021)

This issue was migrated from https://savannah.nongnu.org/bugs/?46092

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 30 Oct 2015 11:34:13 PM UTC

I agree that the ATxmega128D4 has a different pagesize than
the ATxmega128C3 the config entry is based on.  However, are
you sure about the patch you submitted, in particular, did
you really test it?  To me, it appears the patch re-configures
just the same 0x200 page_size.

MCUdude, 2022-03-25:
Fixed in #899

closed 2022-03-25 by MCUdude