avrdudes/avrdude #396: [bug #46461] Problems with lock bits with parallel programmer

avrdude-admin, 2021-12-10:
Mcudude <None>
Mon 16 Nov 2015 11:09:24 PM UTC

Hi!
I got an STK500 dev kit lying around, and wanted to add it as an ISP programmer and a high voltage programmer option for the Arduino IDE. So far, so good. The STK500 as ISP works without any problems what so ever. (I started testing in Arduino IDE, but moved on by typing the commands directly into my terminal window)

The problem is when I use the stk500pp setting. I want to keep the original Arduino lock and unlock bits (0x0F, 0x3F), and be able to change fuses and upload a bootloader. When I erase the attached microcontroller and set the lock bit with this command:
avrdude -v -patmega32 -cstk500pp -P /dev/cu.UC-232AC -e -Ulock:w:0x3f:m -Uhfuse:w:0xde:m -Ulfuse:w:0xa1:m

I get this error message at the bottom:
avrdude: verification error, first mismatch at byte 0x0000
0xff != 0x3f
avrdude: verification error; content mismatch

The exact same avrdude command, using "-cstk500" returns no error.

As far as I've understood, there is a problem with the 6th and 7th lock bit. The HVPP tries to set the 7th and the 6th bit, but avrdude ignores these two bits, and the written fuse bits are actually 0xFF. This gives a content mismatch error.

Please have a look at this Arduino post, where the problem is discussed:
 [http://forum.arduino.cc/index.php?topic=359814.0](https://savannah.nongnu.orghttp://forum.arduino.cc/index.php?topic=359814.0)

This issue was migrated from https://savannah.nongnu.org/bugs/?46461

avrdude-admin, 2021-12-10:
mcudude <mcudude>
Sun 22 Nov 2015 11:43:29 PM UTC

Anything? An explaination, a workaround? Fixed in 6.2?

MCUdude, 2022-03-28:
This is future-me explaining stuff to past-me.

The avrdude.conf file shipped with Arduino IDE has a minor flaw when it cones to the two upper unused lock fuse bits:

Shipped with Arduino IDE
```
    memory "lock"
        size            = 1;
        min_write_delay = 2000;
        max_write_delay = 2000;
        read            = "0 1 0 1  1 0 0 0   0 0 0 0  0 0 0 0",
                          "x x x x  x x x x   x x o o  o o o o";

        write           = "1 0 1 0  1 1 0 0   1 1 1 x  x x x x",
                          "x x x x  x x x x   1 1 i i  i i i i";
      ;
```

Upstream version
```
    memory "lock"
        size            = 1;
        min_write_delay = 2000;
        max_write_delay = 2000;
        read            = "0 1 0 1  1 0 0 0   0 0 0 0  0 0 0 0",
                          "x x x x  x x x x   o o o o  o o o o";

        write           = "1 0 1 0  1 1 0 0   1 1 1 x  x x x x",
                          "x x x x  x x x x   1 1 i i  i i i i";
      ;
```

As you can see, the upper two bits in `read` is set to `o o` in the upstream version, and `x x` in the Arduino version. This means that the upstream version will read these bits as 1's regardless, because that's the correct way of doing it. For some reason, the hvpp programmer "got this right" back in 2015.

Bottom line: Avrdude 6.3 and upwards threw a warning about unused bits and moved on, but unused bits should _always_ be set to 1 instead of 0.

closed 2022-03-28 by MCUdude