avrdudes/avrdude #397: [bug #46483] version 6.2. ser_open(): can't set attributes for device

avrdude-admin, 2021-12-10:
Yuri <yuri@schaeffer.tk>
Fri 20 Nov 2015 03:52:02 PM UTC

After upgrading avrdude from 6.1 to 6.2. I have been unable to program my avr.

avrdude output:

Using Port                    : /dev/ttyACM0
Using Programmer              : arduino
Overriding Baud Rate          : 115200
avrdude: Cannot set serial port speed to 115200. ioctl returned -1
avrdude: ser_open(): can't set attributes for device "/dev/ttyACM0": Operation
not permitted

User is added to the dialout group. Downgrading the package makes it work
again. Furthermore other software has no problem setting attributes of the
serial device.

Installed from the Debian repository.

debian bug: [https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=805642](https://savannah.nongnu.orghttps://bugs.debian.org/cgi-bin/bugreport.cgi?bug=805642)

[file #35509: strace.tar.gz](https://savannah.nongnu.org/bugs/download.php?file_id=35509)
[file #35534: avrdude-r1531-custom-baudrate.patch](https://savannah.nongnu.org/bugs/download.php?file_id=35534)

This issue was migrated from https://savannah.nongnu.org/bugs/?46483

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 20 Nov 2015 04:46:53 PM UTC

As I don't have an Arduino around, please re-run the command
prepending

strace -o trace.txt avrdude …

and attach the file trace.txt to this bug report.

Preferrably, also do it for version 6.1, for comparison.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 20 Nov 2015 09:40:06 PM UTC

strace archive added on behalf of Yuri.

(**[file #35509](https://savannah.nongnu.org/support/download.php?file_id=35509)**)

avrdude-admin, 2021-12-10:
Martin Furter <sabor>
Tue 24 Nov 2015 04:00:14 AM UTC

I had the same problem but with a stk500 connected with a PL2303 USB-serial cable.

I assume not all of these USB-serial devices support custom baud rates. So I wrote a patch (avrdude-r1531-custom-baudrate.patch) which sets the baud rate the standard way for standard baud rates. For custom baud rates it uses the linux way on linux and errors on other OS.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 30 Nov 2015 07:59:49 PM UTC

Sigh, I wish the driver API were the same regardless of which
baudrate the driver supports.

But OK, thanks for the hint, I got the idea.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 04 Dec 2015 07:09:33 AM UTC

The problem came in with **[patch #8380](https://savannah.nongnu.org/patch/?8380)**.  I'm adding the author
of that patch to the Cc list.

avrdude-admin, 2021-12-10:
Anatol <arzedos>
Mon 07 Dec 2015 09:02:27 PM UTC

This bug breaks Linux Arch package as well. It is a severe problem and commit that introduced it should be reverted.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 09 Dec 2015 10:46:47 PM UTC

The offending change has been backed out in r1365.

MCUdude, 2022-01-09:
I'm guessing this issue is not present in Avrdude 6.3 and onwards?

closed 2022-03-19 by MCUdude