avrdudes/avrdude #398: [bug #46573] avrdude doesn't detach from the terminal

avrdude-admin, 2021-12-10:
Laszlo Monda <None>
Mon 30 Nov 2015 04:45:49 PM UTC

I can upload the firmware successfully but after the upload avrdude doesn't detach from the terminal.

I've made a video of this bug: [https://www.youtube.com/watch?v=7Mv5EnGqgmE](https://savannah.nongnu.orghttps://www.youtube.com/watch?v=7Mv5EnGqgmE)

I'm using avrdude 6.0.1

Feel free to ask me any further questions.

Thanks!

This issue was migrated from https://savannah.nongnu.org/bugs/?46573

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 30 Nov 2015 07:57:12 PM UTC

First of all, AVRDUDE version 6.2 is the current one.

Second, some more information about the environment is needed:
which programmer, which AVR, what commandline arguments are
used.

Third, please run the same AVRDUDE command with -vvvv added,
collect the standard error output in a file:

```
avrdude [other commandline arguments] -vvvv 2> logfile.txt

```
and attach that logfile to this bugreport.

mcuee, 2022-05-27:
Youtube video is here:
https://www.youtube.com/watch?v=7Mv5EnGqgmE

Command line is:
```
avrdude -p atmega32u4 -P /dev/uhk-right-bootloader -c avr109 -U flash:w:uhk-right.hex
```

mcuee, 2022-05-27:
But still this kind of issue should just be closed as the reporter does not come back for a long time.

MCUdude, 2022-05-27:
I agree. I'm sure we would get lots of reported issues if this turned out to be a problem with Avrdude 7.0

closed 2022-05-27 by MCUdude