avrdudes/avrdude #399: [bug #46587] patch: transparent SPI (relay) mode

avrdude-admin, 2021-12-10:
Tibor 'Igor2' Palinkas <avrdude@igor2.repo.hu>
Wed 02 Dec 2015 07:30:22 AM UTC

(This is not a bug report but a patch for a new feature. This was the
only tracker I could add it to without registration.)

In my next open source hardware project I program an attiny to be an SPI
slave. I figured the easiest way to test the firmware would be to use my
parallel port avr programmer device to emulate an SPI bus master. For this
I need avrdude to be able to:

- avoid resetting the target device
- omit any SPI commands related to AVR programming during initialization
- initiate arbitrary long SPI transactions

The patch attached implements this in the following steps.

1. a new device flag AVRPART_NO_RESET can disable the reset signal
normally generated for no-tpi devices

2. a new device configuration called "spi" or "transparent_spi" defines
a device that omits the reset and all SPI communication during initialization

3. a new command line switch -S puts avrdude in transparent SPI mode

4. in transparent SPI mode binary data is relayed between stdio and SPI until
EOF on stdin.

I've tested it with a parallel port programmer and a scope, with
MISO-MOSI looped back with a wire, left unconnected or an attiny2313
connected. Both the transparent SPI mode and programming the attiny worked.
I have no idea if it works with non-bitbang programmers.

Command line for the transparent SPI mode: avrdude -c <whatever> -p spi -S

[file #35611: SPI.patch](https://savannah.nongnu.org/bugs/download.php?file_id=35611)

This issue was migrated from https://savannah.nongnu.org/bugs/?46587

mcuee, 2022-06-03:
I tend to think this is out of scope for avrdude and can be closed.

mcuee, 2022-06-04:
@dl8dtl or @MCUdude Please check. Thanks.

MCUdude, 2022-06-04:
If this were ever to be implemented, it would need extensive testing and proper documentation. I'm closing this one as "Not planned", but we might have a look at the patch file if we would consider adding something like this in a later stage.

closed 2022-06-04 by MCUdude