avrdudes/avrdude #400: [bug #46610] Floating point exception (core dumped) arch linux rpi2

avrdude-admin, 2021-12-10:
Mircea Dan <None>
Sat 05 Dec 2015 09:23:23 AM UTC

this was working just fine few days back, i believe there was a recent update and now it's crashing, same command same hex file, i have attached the dumps
i'm programming an arduino basically

[root@alarmpi ~]# /usr/bin/avrdude         -v                   -p atmega2560        -c wiring            -P /dev/ttyACM0      -b 115200            -D                   -U flash:w:/srv/http/upload/Marlin_RUMBA_i3v_GLCDPercent.ino.mega.hex:i

avrdude: Version 6.2, compiled on Dec  3 2015 at 10:47:27
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/etc/avrdude.conf"
User configuration file is "/root/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyACM0
Using Programmer              : wiring
Overriding Baud Rate          : 115200
Floating point exception (core dumped)

uname -a
Linux alarmpi 4.1.13-1-ARCH #1 SMP PREEMPT Fri Nov 13 20:11:44 MST 2015 armv7l GNU/Linux

avrdude -v

avrdude: Version 6.2, compiled on Dec  3 2015 at 10:47:27
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/etc/avrdude.conf"
User configuration file is "/root/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

avrdude: no programmer has been specified on the command line or the config file
Specify a programmer using the -c option and try again

[file #35647: Marlin_RUMBA_i3v_GLCDPercent.ino.mega.hex](https://savannah.nongnu.org/bugs/download.php?file_id=35647)
[file #35646: coredump.zip](https://savannah.nongnu.org/bugs/download.php?file_id=35646)

This issue was migrated from https://savannah.nongnu.org/bugs/?46610

avrdude-admin, 2021-12-10:
Sergey Burobin <burser>
Mon 07 Dec 2015 11:41:18 AM UTC

Almost the same thing.
I've updated the OS yesterday (December, the 6th; Arch Linux 64bit). Among the other updated packages there was AVRDUDE. Now, when I'm downloading the hex file, it writes:
--------------------
[burser@Arch sketch_dec07a]$ avrdude -v -v -C/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -Uflash:w:sketch_dec07a.cpp.standard.hex:i

avrdude: Version 6.2, compiled on Dec  2 2015 at 15:23:56
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf"
User configuration file is "/home/burser/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyACM0
Using Programmer              : arduino
Overriding Baud Rate          : 115200
Исключение в операции с плавающей точкой (core dumped)
--------------------
(The last line is translated as "Exception in operation with floating-point number").

The version of AVRDUDE is the same, 6.2.

IMHO it's somehow connected to configuring the serial port (setting the baud rate), because when I try to set a different baud rate, the results are different:
--------------------
[burser@Arch sketch_dec07a]$ avrdude -v -v -C/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf -patmega328p -carduino -P/dev/ttyACM0 -b9600 -D -Uflash:w:sketch_dec07a.cpp.standard.hex:i

avrdude: Version 6.2, compiled on Dec  2 2015 at 15:23:56
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf"
User configuration file is "/home/burser/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyACM0
Using Programmer              : arduino
Overriding Baud Rate          : 9600
avrdude: Cannot set serial port speed to 9600. ioctl returned -1
avrdude: ser_open(): can't set attributes for device "/dev/ttyACM0": Operation not supported

avrdude done.  Thank you.
--------------------
--------------------
[burser@Arch sketch_dec07a]$ avrdude -v -v -C/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf -patmega328p -carduino -P/dev/ttyACM0 -b19200 -D -Uflash:w:sketch_dec07a.cpp.standard.hex:i

avrdude: Version 6.2, compiled on Dec  2 2015 at 15:23:56
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/usr/share/arduino/hardware/tools/avr/etc/avrdude.conf"
User configuration file is "/home/burser/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyACM0
Using Programmer              : arduino
Overriding Baud Rate          : 19200
avrdude: Cannot set serial port speed to 19200. Closest possible is 9600
avrdude: ser_open(): can't set attributes for device "/dev/ttyACM0": No such file or directory

avrdude done.  Thank you.
--------------------

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 09 Dec 2015 10:37:57 PM UTC

**[Bug #46618](https://savannah.nongnu.org/bugs/?46618)** is a duplicate of this, but contains a good
analysis.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 09 Dec 2015 10:40:21 PM UTC

Another failure related to **[patch #8380](https://savannah.nongnu.org/patch/?8380)**, adding the author of
that patch to Cc list.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 09 Dec 2015 10:47:24 PM UTC

The offending change has been backed out in r1365.

mcuee, 2022-05-29:
> Joerg Wunsch <joerg_wunsch> Wed 09 Dec 2015 10:47:24 PM UTC
> 
> The offending change has been backed out in r1365.

@dl8dtl 
In that case, this issue can be closed, right?

closed 2022-05-29 by MCUdude