avrdudes/avrdude #402: [bug #46684] AVR JtagIce3

avrdude-admin, 2021-12-10:
Pete Bux <None>
Tue 15 Dec 2015 08:44:13 AM UTC

My Jtagice3 system is not recognised. I've used tyhe following command to test:

avrdude p- x128d4 c- jtag3pdi

The produced error is:

avrdude : jtag3_open_common(): Did not find any device matcing VID 0x03eb and PID 0x2110, 0x2410
avrdude done. Thank you

Avrdude has the version 6.2.

The ice3 box is present, has firmware code 3.25 and has VID 0x03EB and PID 0x2410. No problems in using it with ATstudio 6.2.

Please advise.

This issue was migrated from https://savannah.nongnu.org/bugs/?46684

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sat 05 Mar 2016 09:04:07 PM UTC

Are you able to test the current SVN version?

I implemented libhidapi support a couple of weeks ago, which
presumably fixes your issue.

MCUdude, 2022-01-07:
> The ice3 box is present, has firmware code 3.25 and has VID 0x03EB and PID 0x2410. No problems in using it with ATstudio 6.2.

Works with Atmel studio and not with Avrudude? Very likely that the WinUSB driver is used instead of libusb. BTW will the next Avrdude release support the WinUSB driver?

The issue can probably be closed though.

dl8dtl, 2022-01-08:
> will the next Avrdude release support the WinUSB driver?

AFAICT, that's Marius' preference.

closed 2022-01-08 by dl8dtl