avrdudes/avrdude #404: [bug #46843] avrdude=>6 fails to write from 128k-mark onwards, loops back to 0x0000

avrdude-admin, 2021-12-10:
Maarten van Eeuwijk <maarten.van.eeuwijk@vpinstruments.com>
Thu 07 Jan 2016 01:34:16 PM UTC

avrdude version 6 series fails to write programs bigger than 131.072 bytes to flash on my atxmega256a3bu. Instead of continuing onwards from the 131.072 byte (128kb) mark, the write procedure loops back to 0x0000, corrupting data from there on.

Or to put it an another way: When I flash a program 130kb in size, I end up with 128k of data in my flash, whereof the first 2kb is corrupted.

This problem was introduced in the version 6 series of avrdude (I tried 6.0.1, 6.1, 6.1-svn-20131205 and 6.2). Version 5.11 works fine for me.

No errors show up during writing, the bug manifests itself during verification and of course execution of the program, as the mangled program crashes the MCU.

Curiously this 128k mark is exactly on the second time an offset is defined in my intel .hex file. To rule out problems with the file reader code for that format I also tried .srec and .elf; same result for all.

This issue was migrated from https://savannah.nongnu.org/bugs/?46843

mcuee, 2022-06-05:
The above is saying that this is a regression from 5.11. Not so sure if this has been fixed or not. 

There are quite some xmega related issues list below but I do not have an xmega to carry out the test.
* https://github.com/avrdudes/avrdude/issues/435#issuecomment-1140370771

MCUdude, 2022-09-23:
I'm not able to reproduce the issue. I've created two 256kiB hex files using `srec_cat`. The first one contains only 0x55s, and the other one contains 0x00's. I can flash and verify both files, and I can confirm that the correct content is there.

As a test, I've also created a with with 0xAA's that's written from the 128kiB mark and onwards. (My Dragon is a bit flakey, but it gets the job done). **I'm pretty sure we can close this issue.**

```
$ ./avrdude -cdragon_jtag -p atxmega256a3bu -e -Uflash:w:0x55_256kib.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: erasing chip
avrdude: reading input file 0x55_256kib.hex for flash
avrdude: writing 262143 bytes flash ...

Writing | ######################                             | 44% 28.21savrdude: jtagmkII_paged_write(): timeout/error communicating with programmer (status -1)
Writing | ################################################## | 100% 73.80s

avrdude: 262143 bytes of flash written
avrdude: verifying flash memory against 0x55_256kib.hex

Reading | ################################################## | 100% 52.63s

avrdude: 262143 bytes of flash verified

avrdude done.  Thank you.

$ ./avrdude -cdragon_jtag -p atxmega256a3bu -e -Uflash:w:0x00_256kib.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: erasing chip
avrdude: reading input file 0x00_256kib.hex for flash
avrdude: writing 262143 bytes flash ...

Writing | #############################################      | 89% 56.31savrdude: jtagmkII_paged_write(): timeout/error communicating with programmer (status -1)
Writing | ################################################## | 100% 73.26s

avrdude: 262143 bytes of flash written
avrdude: verifying flash memory against 0x00_256kib.hex

Reading | ################################################## | 100% 52.44s

avrdude: 262143 bytes of flash verified

avrdude done.  Thank you.

$ ./avrdude -cdragon_jtag -p atxmega256a3bu -e -Uflash:w:0xAA_128kib_256kib.hex 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: erasing chip
avrdude: reading input file 0xAA_128kib_256kib.hex for flash
         with 131072 bytes in 1 section within [0x1ffff, 0x3fffe]
         using 257 pages and 512 pad bytes
avrdude: writing 131072 bytes flash ...

Writing | ################################################## | 100% 0.25s

avrdude: 131072 bytes of flash written
avrdude: verifying flash memory against 0xAA_128kib_256kib.hex

Reading | ################################################## | 100% 0.21s

avrdude: 131072 bytes of flash verified

avrdude done.  Thank you.
```

@stefanrueger for some reason, the read and write progress bar for the 0xAA_128kib_256kib.hex immediately jumped up to 100%, even though it took about 2x30 seconds for a read and write operation. I've attached the file for reference. 

[0xAA_128kib_256kib.hex.zip](https://github.com/avrdudes/avrdude/files/9637201/0xAA_128kib_256kib.hex.zip)

closed 2022-09-23 by MCUdude