avrdudes/avrdude #406: [bug #46992] AVR dragon does not recognize target

avrdude-admin, 2021-12-10:
Daniel <nusse+savannahavrdude@teamidiot.de>
Wed 27 Jan 2016 01:16:27 PM UTC

I am trying to program a ATtiny85 using avrdude 6.2
running on Debian Testing under Linux 4.3.0.

I am launching avrdude with the following parameters:
-p t85 -c usbasp -e -V -B 10

The AVR dragon recognizes the target VCC voltage but
is unable to read the target device id.

Attached is a avrdude -vvvv log and a pulseview screenshot
that was captured on the target device. Using this setup
I'm able to program the target using a USBasp clone,
so the wiring should be correct.

[file #36175: screenshot.png](https://savannah.nongnu.org/bugs/download.php?file_id=36175)
[file #36174: log.txt](https://savannah.nongnu.org/bugs/download.php?file_id=36174)

This issue was migrated from https://savannah.nongnu.org/bugs/?46992

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 15 Feb 2016 08:45:30 PM UTC

While the original issue turned out to be a hardware problem,
in the course of analyzing this, I think the current Dragon
code ignores the ISP clock setting initially.

mcuee, 2022-05-27:
Does the problem that `current Dragon
code ignores the ISP clock setting initially' still apply to the current AVR Dragon code?

Ref:
https://github.com/avrdudes/avrdude/blob/main/src/jtagmkII.c

MCUdude, 2022-05-27:
I have an ATtiny13 that's running off its internal 128kHz oscillator. That's a maximum ISP clock speed of about 32 kHz.
The -B flag seems to work fine here...

```
$ ./avrdude -C avrdude.conf -c dragon_isp -p attiny13 -v

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : dragon_isp
avrdude: usbdev_open(): Found AVRDRAGON, serno: 00A200000F1A
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:0f:1a
Device ID:                       AVRDRAGON
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

         Programmer Type : DRAGON_ISP
         Description     : Atmel AVR Dragon in ISP mode
         Vtarget         : 5.0 V
         SCK period      : 1.00 us

avrdude: stk500v2_command(): command failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

$ ./avrdude -C avrdude.conf -c dragon_isp -p attiny13 -B32 -v

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
[... 34 lines cut ...]
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash                  65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

         Programmer Type : DRAGON_ISP
         Description     : Atmel AVR Dragon in ISP mode
         Vtarget         : 5.0 V
         SCK period      : 32.12 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9007 (probably t13)

avrdude done.  Thank you.

mcuee, 2022-05-30:
In this case, I think the issue can be closed.

closed 2022-05-30 by MCUdude