avrdudes/avrdude #407: [bug #47043] Support for TinCan Tools Flyswatter 2

avrdude-admin, 2021-12-10:
Russ Dill <russ.dill@gmail.com>
Mon 01 Feb 2016 10:09:13 PM UTC

The Flyswatter2 is a FT2232H based JTAG programmer. It can be used to program AVRs, but it needs a buff pin set. The attached programmer fragment makes that possible.

[https://www.tincantools.com/wiki/Flyswatter2](https://savannah.nongnu.orghttps://www.tincantools.com/wiki/Flyswatter2)

[file #36230: .avrduderc](https://savannah.nongnu.org/bugs/download.php?file_id=36230)

This issue was migrated from https://savannah.nongnu.org/bugs/?47043

mcuee, 2022-06-02:
This seems to be a pretty popular FTDI based debugger/programmer, not so sure if any user can test this out.

mcuee, 2023-01-14:
@russdill

Just wondering if you are still interested in this issue. If yes, please create a new PR for that. Take note that there are some changes in the latest avrdude.conf.in.
```
mosi --> sdo
miso --> sdi
```

stefanrueger, 2023-01-14:
@russdill @mcuee Always best to add a canonical form of the entry; you get this by 
 - Adding the entry provisionally to `avrdude.conf.in` and install it
 - Then using the output of `avrdude -cflyswatter2/s` to replace the provisional entry in `avrdude.conf.in`

closed 2023-04-06 by stefanrueger