avrdudes/avrdude #411: [bug #47450] "lfuse reads as" not displayed in verbose mode.

avrdude-admin, 2021-12-10:
Elwin Beall <frotzen@gmail.com>
Fri 18 Mar 2016 02:17:39 AM UTC

"lfuse reads as" is not displayed in verbose mode.

tail output:

Programmer Type : USBtiny
Description     : USBtiny simple USB programmer, [http://www.ladyada.net/make/usbtinyisp/](https://savannah.nongnu.orghttp://www.ladyada.net/make/usbtinyisp/)
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as DE
avrdude: safemode: efuse reads as FD

avrdude: safemode: hfuse reads as DE
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)

avrdude done.  Thank you.

I didn't see this in the list, so thought I'd submit it.
Thanks.

This issue was migrated from https://savannah.nongnu.org/bugs/?47450

MCUdude, 2022-01-09:
I'm not able to reproduce this ussie with the current upstream Avrdude git version. Issue can probably be closed

```
$ ./avrdude -C avrdude.conf -patmega328p -c usbtiny -v

avrdude: Version 6.99-20211218
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:020
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    1      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably lgt8fx328p)
avrdude: safemode: lfuse reads as F7
avrdude: safemode: hfuse reads as D6
avrdude: safemode: efuse reads as FD

avrdude: safemode: lfuse reads as F7
avrdude: safemode: hfuse reads as D6
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D6, L:F7)

avrdude done.  Thank you.
```

mariusgreuel, 2022-01-09:
USBtiny works well for me, too.

closed 2022-01-09 by mariusgreuel