avrdudes/avrdude #417: [bug #47890] ft232r serial dongle connected to ponyser interface

avrdude-admin, 2021-12-10:
Christian Mueller <None>
Tue 10 May 2016 06:07:34 PM UTC

I tried to get a regular USB/RS232 adapter with an FT232R chip working on a board/programmer using the "ponyser" interface using type "ftdi_syncbb". However, the [oscillo]scope showed that both clk and mosi, which are inverted in my configuration, jumped back to high levels directly after each command.

It turned out that ft245r_cmd(), line 518, finishes by simply sending a 0-byte, ignoring pins that might be inverted:

```
for (i=0; i<4; i++) {
buf_pos += set_data(pgm, buf+buf_pos, cmd[i]);
}
buf[buf_pos] = 0;
buf_pos++;

```
The following should work better:

```
for (i=0; i<4; i++) {
buf_pos += set_data(pgm, buf+buf_pos, cmd[i]);
}
ft245r_out = SET_BITS_0(ft245r_out,pgm,PIN_AVR_SCK,0);
ft245r_out = SET_BITS_0(ft245r_out,pgm,PIN_AVR_MOSI,0);
buf[buf_pos] = ft245r_out;
buf_pos++;

```
NOTE: The RS232 dongle still didn't work in the end but the scope showed correct signal levels. It also showed, however, that my dongle apparently doesn't provide enough current to switch the signals fast enough so that the peak-to-peak voltage at the default baud rate for CLK was only around 2.5V... I gave up on this for the time being - I still do have a regular RS232 port in my computer - but I thought it would be good to keep track of the issue, hence this bug report.

For what it's worth at this point, here's the configuration for this combination (haven't checked whether MISO is inverted or not because I've never seen anything on that pin):

```
# ponyser serial design with regular FT232R USB/serial adapter
programmer
id    = "ponyser-ft232r";
desc  = "ponyser serial design, using FT232R Synchronous BitBang";
type  = "ftdi_syncbb";
connection_type = usb;
miso  = ~3;  # CTS
sck   = ~2;  # RTS
mosi  = ~4;  # DTR
reset = ~0;  # TXD
;

```

This issue was migrated from https://savannah.nongnu.org/bugs/?47890

avrdude-admin, 2021-12-10:
Christian Mueller <cm1>
Thu 26 May 2016 06:52:33 AM UTC

Just got myself a small logic analyzer so I could watch the signals in relation to each other and it turned out I had the reset pin wrong. The following definition now works fine with rates up to 4000 bits/s (-b 4000):

```
# ponyser serial design with FT232R-based USB/RS232 adapter
programmer
id    = "ponyser-ft232r";
desc  = "ponyser serial design, using FT232R Synchronous BitBang";
type  = "ftdi_syncbb";
connection_type = usb;
miso  = ~3;  # CTS
sck   = ~2;  # RTS
mosi  = ~4;  # DTR
reset = 0;   # TXD
;

```
4000 bits/s is not exactly fast but it makes using an FT232R-based USB/RS232 dongle a usable option - without that, it would take some 40 minutes to write 8K of flash. The evaluation board I'm using also adds to this because it has multiple sockets wired in parallel for various chips which adds to the load applied to the RS232 level boost chip. A more direct connection might be a lot faster.

avrdude-admin, 2021-12-10:
Christian Mueller <cm1>
Fri 27 May 2016 01:24:17 PM UTC

Small update: I had to reduce the bit rate to "-b 2000" to get it 100% stable but it's still fast enough to be usable.

mcuee, 2022-05-29:
The code has not been changed in this case. 
https://github.com/avrdudes/avrdude/blob/main/src/ft245r.c#L686

I think the OP wants to have the following changes.
From:
```
    for (i=0; i<4; i++) {
        buf_pos += set_data(pgm, buf+buf_pos, cmd[i]);
    }
    buf[buf_pos] = 0;
    buf_pos++;
```

To:
```
        for (i=0; i<4; i++) {
            buf_pos += set_data(pgm, buf+buf_pos, cmd[i]);
        }
        ft245r_out = SET_BITS_0(ft245r_out,pgm,PIN_AVR_SCK,0);
        ft245r_out = SET_BITS_0(ft245r_out,pgm,PIN_AVR_MOSI,0);
        buf[buf_pos] = ft245r_out;
        buf_pos++;
```

mcuee, 2022-06-12:
On the other hand, normal FT232R USB to serial converter works fine in my tests.

* https://github.com/avrdudes/avrdude/issues/464

mcuee, 2022-06-12:
@dl8dtl and @mariusgreuel 
Please take a look when you got the time. Thanks.

mcuee, 2022-07-29:
It seems to me a simple FT232R based programmer is faster than this one. So I doubt its usefulness. I will close this one for now unless the OP comes back to revive the post.

closed 2022-07-29 by mcuee