avrdudes/avrdude #419: [bug #48004] Dead link for linuxgpio in avrdude description

avrdude-admin, 2021-12-10:
Nick <none>
Wed 25 May 2016 01:25:19 AM UTC

Paragraph 4 of the avrdude description in Linux "man avrdude" gives a link to '[http://kolev.info/avrdude-linuxgpio](https://savannah.nongnu.orghttp://kolev.info/avrdude-linuxgpio)'.  That link leads to a 404 error page.  Searching the site '[http://kolev.info](https://savannah.nongnu.orghttp://kolev.info)' for "avrdude" returns an incomplete page.

This issue was migrated from https://savannah.nongnu.org/bugs/?48004

avrdude-admin, 2021-12-10:
John <johnbentcope>
Sat 25 Apr 2020 07:40:47 PM UTC

This is still an issue with the documentation. It looks as though kolev has never had a page cached with the Wayback Machine that is ever populated with information on linuxgpio

MCUdude, 2022-01-09:
This is the working URL: http://kolev.info/blog/2013/01/06/avrdude-linuxgpio/

commit c7f7fcda (2022-01-10): Fix URL for linuxgpio programmer

Closes #419
closed 2022-01-10 by dl8dtl