avrdudes/avrdude #421: [bug #48131] Problem with EEprom address of avrdude for different processors!

avrdude-admin, 2021-12-10:
kh_kuebbeler <None>
Sat 04 Jun 2016 06:06:37 PM UTC

I have found out, that the address of EEprom load with the stk500v1 protocol is send as word address for the target ATmega328 and as a byte address for the target ATmega8.
I have build a bootloader from the optiboot source in assembly language with additional EEprom support, which match in a 512 byte page.
If I handle the transmitted address different for the ATmega328 and
the ATmega8, all works well. For the ATmega328 I must multiply the
transmitted address by 2.
The problem is already reported at [https://www.mikrocontroller.net/topic/398978](https://savannah.nongnu.orghttps://www.mikrocontroller.net/topic/398978) in german language. Joerg Wunsch has this information already.
I can handle the different address type in my bootloader version,
if I know, which processors use the byte address (mega8, mega16, mega32, ...). But what is with other transfer programs as avrdude?
The AVR061 from Atmel does not specify the address type of the
Cmnd_STK_LOAD_ADDRESS command.

This issue was migrated from https://savannah.nongnu.org/bugs/?48131

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sat 04 Jun 2016 06:29:52 PM UTC

The problem is apparently that piece of code in stk500.c:

```
if ((m->op[AVR_OP_LOADPAGE_LO]) || (m->op[AVR_OP_READ_LO]))
a_div = 2;
else
a_div = 1;

```
Later on, a_div is used to determine whether the address in
the load command needs to be divided by 2:

```
stk500_loadaddr(pgm, m, addr/a_div);

```
(Similar code exists in both, stk500_paged_write() as well as
stk500_paged_load()).

I think the intention was to divide the load address by 2 when
accessing flash.  Now, the ATmega328 entry (as well as others)
has an AVR_OP_LOADPAGE_LO instruction even in the EEPROM memory
section.  Thus, load addresses are halved for it when accessing
EEPROM.

If the only decision needed is whether flash or EEPROM is going
to be accessed, my suggestion is to combine that with the 'F' and
'E' decision a few lines before:

```
if (strcmp(m->desc, "flash") == 0) {
memtype = 'F';
a_div = 2;
}
else if (strcmp(m->desc, "eeprom") == 0) {
memtype = 'E';
a_div = 1;
}
else {
return -2;
}

```

avrdude-admin, 2021-12-10:
Bill Westfield <westfw>
Tue 24 Jul 2018 05:39:34 AM UTC

Hmm.  Also breaks the ability to read or write EEPROM to ATMEGA8 using the Arduino as ISP sketch, which is expecting the addresses in the STK500 "Load Address" command to be word addresses (as it is for the m328.)

stefanrueger, 2022-05-22:
There is more discussion on this in Issue #967 

Basically STK500v1 is prone to misinterpretation and suffered from different implementations. AVRDUDE's `stk500.c` (v1) does the right thing and handles undocumented quirks in the STK500 kit well (EEPROM byte address for some. EEPROM word address for most devices, go figure). Others (optiboot, old AVRISP STK500 v1 implementations that I have seen) use EEPROM word addresses throughout.

Suggestions for solutions in Issue #967

stefanrueger, 2022-07-24:
Discussion in Issue #967 has concluded with PR #1046

MCUdude, 2022-08-05:
I'm pretty sure #1046 resolved this issue.

closed 2022-08-05 by MCUdude