avrdudes/avrdude #422: [bug #48134] ft245r.c : dispatch/dispatch.h: No such file or directory

avrdude-admin, 2021-12-10:
Phillip Stevens <phillip.stevens@gmail.com>
Sun 05 Jun 2016 08:17:51 AM UTC

When compiling 6.3 on Mac OSX 10.4.11 with MacPorts the build fails because the dispatch.h file does not exist.

dispatch/dispatch.h was added with Grand Central in 10.6, I understand.

This bug was originally reported here
 [https://trac.macports.org/ticket/44717](https://savannah.nongnu.orghttps://trac.macports.org/ticket/44717)

And I reported the same thing here
 [https://trac.macports.org/ticket/51566](https://savannah.nongnu.orghttps://trac.macports.org/ticket/51566)

Tested that the fix suggested works on
avrdude 6.3
MacPorts 2.3.4
OSX10.4.11

Proposed resolution, in ft245r.c :

< #include <sys/semaphore.h>
---

> /* Mac OS X defines sem_init but actually does not implement them */
> #include <dispatch/dispatch.h>
>
> typedef dispatch_semaphore_t  sem_t;
>
> #define sem_init(psem,x,val)  *psem = dispatch_semaphore_create(val)
> #define sem_post(psem)                dispatch_semaphore_signal(*psem)
> #define sem_wait(psem)                dispatch_semaphore_wait(*psem, DISPATCH_TIME_FOREVER)

[file #37395: ancient_os_x.patch](https://savannah.nongnu.org/bugs/download.php?file_id=37395)
[file #37438: ft245r.c.diff](https://savannah.nongnu.org/bugs/download.php?file_id=37438)

This issue was migrated from https://savannah.nongnu.org/bugs/?48134

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 05 Jun 2016 02:41:30 PM UTC

I'd welcome if you provide the diff output as a so-called
"context diff" (diff -u).  That makes it a lot easier to
follow it.  Make sure to specify the original file first,
and the modified file second.

Can you provide a pointer to OSX documentation about these
functions?  I'm not familar with OSX.

avrdude-admin, 2021-12-10:
Phillip Stevens <feilipu>
Thu 09 Jun 2016 08:40:20 AM UTC

Joerg,

I've prepared the diff as requested, and attached it to this bug.

Confirmed to work on OSX 10.4 Tiger.
avrdude 6.3
MacPorts 2.3.4
OSX10.4.11

I can't confirm it on later >10.6 OSX versions.

Pasted below, because it is short and helps traceability.

Cheers, Phillip

--- ft245r_original.c   2016-06-09 18:12:42.000000000 +1000
+++ ft245r.c    2016-06-09 18:27:39.000000000 +1000
@@ -123,14 +123,21 @@
#include <pthread.h>

#ifdef **_APPLE_**
-/* Mac OS X defines sem_init but actually does not implement them */
-#include <dispatch/dispatch.h>
+#include <AvailabilityMacros.h> /* since OS X 10.2 */
+#if(MAC_OS_X_VERSION_MAX_ALLOWED >= 1060)

+/* Mac OS X defines sem_init but actually does not implement them */
+#include <dispatch/dispatch.h> /* since OSX 10.6 */
typedef dispatch_semaphore_t   sem_t;

#define sem_init(psem,x,val)   *psem = dispatch_semaphore_create(val)
#define sem_post(psem)         dispatch_semaphore_signal(*psem)
#define sem_wait(psem)         dispatch_semaphore_wait(*psem, DISPATCH_TIME_FOREVER)
+
+#else
+#include <sys/semaphore.h> /* since OSX 10.2 through OSX 10.5 */
+#endif
+
#else
#include <semaphore.h>
#endif

(**[file #37438](https://savannah.nongnu.org/support/download.php?file_id=37438)**)

mariusgreuel, 2022-01-18:
I think this has been fixed, the current CMake project builds on all known variants.

closed 2022-01-18 by mariusgreuel