avrdudes/avrdude #423: [bug #48188] Avrdude Permission denied on Atmega32u4

avrdude-admin, 2021-12-10:
Paul Hein <dehein@gmail.com>
Thu 09 Jun 2016 07:38:13 PM UTC

I am using Archlinux with Avrdude version 6.3.
I have an Atmega32u4 Sparkfun Pro Micro Clone.
My udev rules look like this

SUBSYSTEM=="usb", ATTRS{idVendor}=="1b4f", ATTRS{idProduct}=="9205", ATTRS{serial}=="/dev/ttyACM3", GROUP="users", MODE="0660"

also here:
crw-rw---- 1 root uucp 166, 3  9. Jun 21:30 /dev/ttyACM3

Everytime i try to upload i get this error
avrdude: ser_open(): can't open device "/dev/ttyACM3": Permission denied

unless i start my ide with root rights.

What could here possibly be wrong. Because it seems avrdude seems to still not work although i am in the uucp group.

This issue was migrated from https://savannah.nongnu.org/bugs/?48188

MCUdude, 2022-05-29:
I'm closing this issue because this problem isn't really Avrdude's fault. Please correct me if I'm wrong!

closed 2022-05-29 by MCUdude

mcuee, 2022-05-30:
Yes I agree this is not related to avrdude but rather wrong udev rule. 
`uucp` may not be the right group depending on the distro, sometimes it is `dialout`.