avrdudes/avrdude #425: [bug #48261] ATmega32A, AVR DRAGON ISP programming, verification error

avrdude-admin, 2021-12-10:
Johnny Quest <scotty264b@gmx.com>
Sat 18 Jun 2016 04:51:15 AM UTC

Hello:
First off, I have done my homework on this matter and would not have posted seeking assistance unless it was a last resort.  smiley

This message was posted on AVR FREAKS. [http://www.avrfreaks.net/forum/atmega32a-avr-dragon-programming-flaky-ness](https://savannah.nongnu.orghttp://www.avrfreaks.net/forum/atmega32a-avr-dragon-programming-flaky-ness)

Background:  Ex-disti-FAE for ATMEL as one of my support lines.  Very familiar with ATmega48/88/168/328, ATtiny25/45/85, ATmega32U4, AT90USB1286 (and somewhat on the ATmega2560).

Development environment:  Using Linux for development running AVR STUDIO in a "Virtual WINDOWS" environment.   AVRDUDE V6.3 for Linux.  ATmega32A, 16MHz external crystal, VCC = 5V (PwrSup is 1A capable).  Fuses set for external high-freq crystal, bootloader at 0x7E00, bootloader reset enabled (lfuse = 0x8F   hfuse = 0x96).

Quandary: developing code for ATmega32A; when using AVRDUDE to program FLASH using ISP with "DRAGON_ISP" programmer, AVRDUDE fails with:

avrdude: verification error, first mismatch at byte 0x7e00
0xff != 0x0f

-------------
Command line invokation is:

avrdude -u -p m32 -c dragon_isp -B 1MHz -U flash:w:AVR_Specific_Builds/m32/ATTOBASICV234_m32-16MHZ-uart_btldr.hex

-------------
Partial dump from original HEX file around address 0x7E00:
:107D00000000000000000000000000000000000073
:107D10000000000000000000000000000000000063
:107D20000000000000000000000000000000000053
:107D30000000000000000000000000000000000043
:107D40000000000000000000000000000000000033
:107D50000000000000000000000000000000000023
:107D60000000000000000000000000000000000013
:107D70000000000000000000000000000000000003
:107D8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF03
:107D9000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3
:107DA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE3
:107DB000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD3
:107DC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC3
:107DD000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB3
:107DE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA3
:107DF000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF93
:107E00000F92CDB7DEB711248FE598E09EBF8DBFEE
:107E100084B714BE81FFD6D085E08EBD82E08BB9D9
:107E200088E18AB986E880BD80E189B98EE0B5D065
:107E3000BD9A26E080E39CEF54E040E29DBD8CBDFE
:107E400058BF08B602FEFDCF38B3342738BBA8951B
:107E50002150A1F788249924CC24C394F5E0DF2E87
:107E6000E1E1EE2E73E0F72E91D0813469F48ED0EB
:107E7000898398D08981823811F1813811F485E0A5
:107E800001C083E07FD07BC0823411F484E103C061
:107E9000853419F485E08ED072C0853561F476D0D2
:107EA000082F10E073D090E0982E8824802A912A21
:107EB000880C991C63C0863521F484E07BD080E077
:107EC000E1CF843609F03FC061D060D0B82E5ED0DB
:107ED00080E0881680E7980618F4F401F7BEE8956C
:107EE00000E610E053D0F80181938F01BA94D1F7E6
:107EF000F0E08F16F0E79F0618F0F401F7BEE89562

Addresses 0x7D00 to 0x7DFF is waveform data for DDS function.

-------------
After failure, partial dump of reback of FLASH memory around address 0x7E00.  Even the data starting at 0x7D00 is incorrect.
:107D0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF83
:107D1000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF73
:107D2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF63
:107D3000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF53
:107D4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF43
:107D5000FFFF000000000000000000000000000025
:107D60000000000000000000000000000000000013
:107D70000000000000000000000000000000000003
:107D8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF03
:107D9000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3
:107DA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE3
:107DB000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD3
:107DC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC3
:107DD000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB3
:107DE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA3
:107DF000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF93
:107E0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF82
:107E1000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF72
:107E2000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF62
:107E3000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF52
:107E4000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF42
:107E5000FFFFA1F788249924CC24C394F5E0DF2EFA
:107E6000E1E1EE2E73E0F72E91D0813469F48ED0EB
:107E7000898398D08981823811F1813811F485E0A5
:107E800001C083E07FD07BC0823411F484E103C061
:107E9000853419F485E08ED072C0853561F476D0D2
:107EA000082F10E073D090E0982E8824802A912A21
:107EB000880C991C63C0863521F484E07BD080E077
:107EC000E1CF843609F03FC061D060D0B82E5ED0DB
:107ED00080E0881680E7980618F4F401F7BEE8956C
:107EE00000E610E053D0F80181938F01BA94D1F7E6
:107EF000F0E08F16F0E79F0618F0F401F7BEE89562

-------------
All other ATmega devices program properly using ISP and JTAG, without incident, so I do not believe it is the DRAGON.  The ATmega32A programs properly using JTAG with AVRDUDE and DRAGON, only ISP fails using AVRDUDE.  Using DRAGON (1MHz sck) under AVR STUDIO, ISP programming is successful.

I have tried ISP with USBASP and USBtinyISP as well and they function correctly.

I have tried with V6.3, V6.2 V6.0 of AVRDUDE.  V5.x versions fail to communicate with DRAGON due to "parallel port connection error" (?????DRAGON is USB!!)  I have looked at the avrdude.conf for other "compatible devices" and found ATmega328p and ATmega329 are accurate.  Using AVRDUDE "-F" switch produces the same verification failure with these two part definitions.

I have tried various "-B" settings for AVRDUDE to no avail.
I have tried using external 8MHz crystal to no avail.
I have tried using internal 8MHz oscillator to no avail.

I have four (4) NEW ATmega32A's, none are blown as I can successfully program all of them using JTAG, USBtinyISP and USBasp but using DRAGON_ISP, all fail in the same manner.

Chinese fakes?  I do not see how as these parts do not look like it. They have ATMEL logo and date code of "1508".

Is this an AVRDUDE issue, an ATmega32A issue or combination of both?

Does anyone have any answers, ideas or clues to offer?

Thank you in advance.

Peace and blessings,
Scott

[file #37526: ATTOBASICV234_m32-16MHZ-uart_btldr.hex](https://savannah.nongnu.org/bugs/download.php?file_id=37526)
[file #37550: ATTOBASICV234_m16-16MHZ-uart_btldr.hex](https://savannah.nongnu.org/bugs/download.php?file_id=37550)

This issue was migrated from https://savannah.nongnu.org/bugs/?48261

avrdude-admin, 2021-12-10:
Scott Vitale <jquest>
Wed 22 Jun 2016 06:53:34 PM UTC

Hello:
As a further test, I recently ordered and received two (2) ATmega16A's.  Using the same test conditions, but uploading a 16KB image to FLASH, I receive the same "verification error" as I did with the ATmega32A's.  Again, the failure is complained about at the bootloader's selected address.  The clock speed is 16MHz, external crystal.  Here is the command line invocation and subsequent error message.  0x3e00 is the selected start address of the bootloader via the fuse settings.

avrdude -u -p m16 -c dragon_isp -B 500kHz -U flash:w:ATTOBASICV234_m16-16MHZ-uart_btldr.hex

"avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x3e00
0xff != 0x0f
avrdude: verification error; content mismatch
"

I will upload the original file for this one as well.

Peace and blessings,
Johnny Quest

avrdude-admin, 2021-12-10:
Scott Vitale <jquest>
Sat 25 Jun 2016 04:35:40 PM UTC

Thought this might be helpful.

My AVR DRAGON: Hardware Revision: 0x107, Firmware Version: 0x610610

BTW:  The original HEX file was generated by AVRASM.EXE (from AS 4.19).

I also recently generated and tested with with two 32KB files. one filled with all 0x00's and the other with random values.  Both programmed properly on my setup.

Thinking that missing data between unused address ranges was a contributing factor,  I used srec_cat to fill in the missing address ranges of my original test file with 0x00's and again with 0xFF's to generate two separate full 32KB address files.  Once again, avrdude + DRAGON_ISP failed.

Discussion is here: [http://www.avrfreaks.net/comment/1916896#comment-1916896](https://savannah.nongnu.orghttp://www.avrfreaks.net/comment/1916896#comment-1916896)

Also, This error was confirmed by some one else for me.  The specifics for his setup are AVRDUDE 6.1 with ATmega32 (non "A").

Discussion is here: [http://www.avrfreaks.net/comment/1917241#comment-1917241](https://savannah.nongnu.orghttp://www.avrfreaks.net/comment/1917241#comment-1917241)

mcuee, 2022-05-30:
Not so sure if this is still relevant or not. I have the AVR Dragon but not the M32 or M32A for testing. Interestingly the avrfreaks thread also says Dragon is flaky.

MCUdude, 2022-05-30:
I'll give it a try later today with an ATmega32.

The dragon is indeed very flakey. Some information here:
http://www.aplomb.nl/TechStuff/Dragon/Dragon.html

> Three main reasons:
>1. The Dragon is USB-powered, and has a step-up converter on board to generate 12V for the HV-programming.
However, when touching the area where this circuit located (down-left corener on the picture), causes the voltage to go up, and that's fatal. Dragon DEAD.
>2. Powering the Dragon from a USB-port that is not "prepared" to deliver the 850 mA rush-in current, causes the step-up coverter to die, because of the sudden drop of the USB-supply voltage.
>3. If the Dragon is connected to a faulty board (sh*t happens ...), it's interface isn't rugged enough to withstand that abuse.
> 
> Since we know this, .... time for some action.
>1. Solution is simple: don't touch the Dragon in that area. And how to accomplish that ? Give it a proper housing.
>2. Not all PC-USB-ports can deliver the rush-in current, and failing to do so means ..... ah, we know that now.
Solution: Power the Dragon from an USB-hub with an external, beefy power-supply.
>3. Add a buffer that can withstand that abuse.

MCUdude, 2022-05-30:
I can confirm that there's something going on with the dragon.

Here are the ATmega32's fuse settings:
<details>

```
$ ./avrdude -p atmega32 -c usbasp -Ulfuse:r:-:h -Uhfuse:r:-:h

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: reading lfuse memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0x8f
avrdude: reading hfuse memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0x96

avrdude done.  Thank you.
```

</details>

Here is OPs hex file written with an AVR dragon (fails):
<details>

```
$ ./avrdude -p atmega32 -c dragon_isp -Uflash:w:/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex"
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 12.89s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex:
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex

Reading | ################################################## | 100% 12.40s

avrdude: verification error, first mismatch at byte 0x7e00
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

</details>

Here's OPs hex file written with a USBasp (succeeds):
<details>

```
$ ./avrdude -p atmega32 -c usbasp -Uflash:w:/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex"
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 4.08s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex:
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex

Reading | ################################################## | 100% 2.51s

avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.
```

</details>

To my surprise, the AVRISPmkII also fails:
<details>

```
$ ./avrdude -p atmega32 -c avrispmkii -Uflash:w:/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex"
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 18.16s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex:
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex

Reading | ################################################## | 100% 17.64s

avrdude: verification error, first mismatch at byte 0x7e00
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

</details>

@dl8dtl do you have any idea why the Dragon and the AVRISPmkII fail?

MCUdude, 2022-05-30:
Interesting!

If I remove the following part of the hex file, located _between_ the user program and the bootloader section, it works with both the Dragon and the AVRISPmkII...

```patch
 :10385000666F756E642120000000FBFD61206269C7
 :103860006E617279206E756D62657200FB496E76CD
 :10387000616C6964206E756D626572206F6620618F
 :103880007267756D656E74732E000000FB204576BF
 :10389000656E2061646472657373206F6E6C79214C
 :0438A0002000189557
 :107D00000000000000000000000000000000000073
 :107D10000000000000000000000000000000000063
 :107D20000000000000000000000000000000000053
 :107D30000000000000000000000000000000000043
 :107D40000000000000000000000000000000000033
 :107D50000000000000000000000000000000000023
 :107D60000000000000000000000000000000000013
 :107D70000000000000000000000000000000000003
-:107D8000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF03
-:107D9000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3
-:107DA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE3
-:107DB000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD3
-:107DC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC3
-:107DD000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB3
-:107DE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA3
-:107DF000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF93
 :107E00000F92CDB7DEB711248FE598E09EBF8DBFEE
 :107E100084B714BE81FFD6D085E08EBD82E08BB9D9
 :107E200088E18AB986E880BD80E189B98EE0B5D065
 :107E3000BD9A26E080E39CEF54E040E29DBD8CBDFE
 :107E400058BF08B602FEFDCF38B3342738BBA8951B

```

mcuee, 2022-06-03:
Interesting I got verfication error with my newly bough Arduino Mega2560 clone (with CH340) and initially I also got verification error with an hex file (exported from Arduino blinky example). And Arduino itself also has the verification error message after followed by stk500v2 timeout error.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c dragon_isp -U .\Blink.ino.with_bootloader.mega.hex -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : dragon_isp
avrdude.exe: usbdev_open(): Found AVRDRAGON, serno: 00A200001487
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:14:87
Device ID:                       AVRDRAGON
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : DRAGON_ISP
             Description     : Atmel AVR Dragon in ISP mode
             Vtarget         : 4.9 V
             SCK period      : 1.00 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\Blink.ino.with_bootloader.mega.hex"
avrdude.exe: input file .\Blink.ino.with_bootloader.mega.hex auto detected as Intel Hex
avrdude.exe: writing flash (261406 bytes):
[... 159 lines cut ...]

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\Blink.ino.with_bootloader.mega.hex"
avrdude.exe: input file .\Blink.ino.with_bootloader.mega.hex auto detected as Intel Hex
avrdude.exe: writing flash (261406 bytes):

Writing | ################################################## | 100% 0.71s

avrdude.exe: 261406 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.with_bootloader.mega.hex:
avrdude.exe: input file .\Blink.ino.with_bootloader.mega.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.72s

avrdude.exe: 261406 bytes of flash verified

avrdude.exe done.  Thank you.

```
[Blink.ino.with_bootloader.mega.hex.txt](https://github.com/avrdudes/avrdude/files/8832130/Blink.ino.with_bootloader.mega.hex.txt)

mcuee, 2022-06-03:
> Interesting!
> 
> If I remove the following part of the [hex file](https://savannah.nongnu.org/bugs/download.php?file_id=37526), located _between_ the user program and the bootloader section, it works with both the Dragon and the AVRISPmkII...
> 

Old info which is for Mega2560 for reference. Not so sure if this is related.
https://forum.arduino.cc/t/error-with-arduino-mega-2560/935534

MCUdude, 2022-06-08:
@dl8dtl you're probably the one that knows the source code the best. How can one type of programmer (USBasp) work with OPs hex files, while others (AVRISPmkII, Dragon ISP) don't? Is this related to the programmer hardware/firmware, or does Avrdude treat these programmers differently in terms of what's "legal" to write?

dl8dtl, 2022-06-08:
Sounds to me like a bug in the stk500v2 code (which is used for Dragon, STK500, AVRISPmkII and so on).

mcuee, 2022-06-15:
Not so sure if it is to do with the `boot_start` handling in stk500v2.c, but it is supposedly to be used only for xmega.

mcuee, 2022-06-16:
This is what I get.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c usbtiny -p m2560 -v -D -U flash:w:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbtiny
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\?\usb#vid_1781&pid_0c9f#6&8694159&0&3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}--libusb0
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude.exe: programmer operation not supported

avrdude.exe: Using SCK period of 10 usec
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verification error, first mismatch at byte 0x11000
             0xff != 0x54
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

Then I verified with AVRDragon and it still shows error.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c dragon_isp -p m2560 -U flash:v:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verification error, first mismatch at byte 0x3fc00
             0x00 != 0x80
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.
```

BTW, AVRDragon itself is okay.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64>
                                                      .\avrdude.exe -c dragon_isp -p m2560 -U flash:w:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\atmega2560_0x11000_offset.hex"
avrdude.exe: writing flash (261414 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash written
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -c dragon_isp -p m2560 -U flash:v:.\atmega2560_0x11000_offset.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: verifying flash memory against .\atmega2560_0x11000_offset.hex:

Reading | ################################################## | 100% 0.01s

avrdude.exe: 261414 bytes of flash verified

avrdude.exe done.  Thank you.
```

mcuee, 2022-06-16:
I do not have the ATmega32A, not so sure if I can test with ATmega328P or not.

mcuee, 2022-07-11:
I just got an ATmega32A board running MightyCore.

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude -p atmega32a -c stk500v2 -P COM6 -Ulfuse:r:-:h -Uhfuse:r:-:h

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.09s

avrdude.exe: Device signature = 0x1e9502 (probably m32a)
avrdude.exe: reading lfuse memory:

Reading | ################################################## | 100% 0.03s

avrdude.exe: writing output file "<stdout>"
0x8f
avrdude.exe: reading hfuse memory:

Reading | ################################################## | 100% 0.03s

avrdude.exe: writing output file "<stdout>"
0x96

avrdude.exe done.  Thank you.

```

mcuee, 2022-07-11:
> Interesting!
> 
> If I remove the following part of the [hex file](https://savannah.nongnu.org/bugs/download.php?file_id=37526), located _between_ the user program and the bootloader section, it works with both the Dragon and the AVRISPmkII...
> 

@MCUdude 
And yes I can reproduce the same thing with my AVRISP mkii clone (this looks like a good clone which works under Atmel Studio 7).

```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude -p atmega32a -c avrispmkii -U flash:w:ATTOBASICV234_m32-16MHZ-uart_btldr_ok.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9502 (probably m32a)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "ATTOBASICV234_m32-16MHZ-uart_btldr_ok.hex"
avrdude.exe: writing flash (32768 bytes):

Writing | ################################################## | 100% 3.41s

avrdude.exe: 32768 bytes of flash written
avrdude.exe: verifying flash memory against ATTOBASICV234_m32-16MHZ-uart_btldr_ok.hex:

Reading | ################################################## | 100% 2.84s

avrdude.exe: 32768 bytes of flash verified

avrdude.exe done.  Thank you.
```

mcuee, 2022-07-11:
Then I have another STK500v2 clone using a 8051 CPU (STC STC15W1K16S) which also claims to support HVSP/HVPP, and it does not have this issue. Looks like this one and the previous STK500v2 clone may not follow exactly the official STK500 FW. And then the AVeRISP mkii clone and the STK500v2 clone using ATmega8535 (which also claims to support HVSP/HVPP) seem to follow the official FW.

```
mcuee@UbuntuSwift3:~/build/avr$ avrdude -c stk500v2 -P /dev/ttyUSB1 -B 1 -p atmega32a  -U flash:w:./ATTOBASICV234_m32-16MHZ-uart_btldr.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9502 (probably m32a)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./ATTOBASICV234_m32-16MHZ-uart_btldr.hex"
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 3.66s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against ./ATTOBASICV234_m32-16MHZ-uart_btldr.hex:

Reading | ################################################## | 100% 2.90s

avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-11:
BTW, I can reproduce the issue with the default demo FW on my board.
[atmega32a_demo_muzi.hex.txt](https://github.com/avrdudes/avrdude/files/9084323/atmega32a_demo_muzi.hex.txt)

mcuee, 2022-07-31:
> Then I have another STK500v2 clone using a 8051 CPU (STC STC15W1K16S) which also claims to support HVSP/HVPP, and it does not have this issue. Looks like this one and the previous STK500v2 clone may [...]

So this somehow points for FW issues? Or there needs to be a workaround in avrdude?

MCUdude, 2022-08-03:
Maybe @stefanrueger is able to figure out what's wrong here?

@mcuee I beleve there is an issue with the Avrdude implementation, and not necessary the official Atmel programmers. I can confirm that the `ATTOBASICV234_m32-16MHZ-uart_btldr.hex` file can sucessfully be written to an ATmega32 using an Atmel AVRISPmkII programmer though Microchip Studio 7 or using their CLI tool, `atprogram.exe`.

stefanrueger, 2022-08-04:
It's a bootloader section write issue, so I wonder whether it's worthwhile playing with the lock bytes and the fuses: change the fuses for a larger, say 1024 bytes, boot section and see whether the verification error happens at 31 KiB rather than 31.5 KiB; that requires a non-0xff input test file up to the brim (maybe a test file with 16 K `nop`; or for variety, maybe some different effective nops such as `mov rx,rx` or even `mov rx,ry` to generate non-trivial bit patterns; just avoid a random file that risks port I/O). I know the BL lock bits normally are about allowing LPM/SPM in the selected bootloader area, but who knows whether that's the same for the `ATmega32A`? So, I'd ensure lock bits are all-permissive (though this could just be unnecessary superstition).

I also had a quick look at the write delays:
```
$ avrdude -p m32a/w
.wd_chip_erase 9.000 ms ATmega32A
.wd_eeprom 9.000 ms ATmega32A
.wd_flash 4.500 ms ATmega32A
.wd_lfuse 2.000 ms ATmega32A
.wd_hfuse 2.000 ms ATmega32A
.wd_lock 2.000 ms ATmega32A
```
That kinda looks suspicious. Normally, wd for fuses is the same as for flash.

Are the lock/fuse r/w commands in order? Perhaps check with `avrdude -p m32a/s` against data sheet. If that fails, I'd look at the write lock/fuse routines in `stk500v2` (they may well not use the `avrdude.conf` values, so a careful check of the code might be in order). 

If `stk500v2.c` reasons about where the bootloader sits that might well be a good place to check (I think @mcee mentioned that already, that's a great shout).

I realise I am a bit vague here, but I neither have the programmers in question nor the part.

MCUdude, 2022-08-04:
> It's a bootloader section write issue, so I wonder whether it's worthwhile playing with the lock bytes and the fuses: change the fuses for a larger, say 1024 bytes, boot section and see whether the verification error happens at 31 KiB rather than 31.5 KiB

For reference, here is the same file written but the bootloader section size is set to 1024 bytes instead of 512:

```
$ ./avrdude -patmega32 -cavrispmkii -Ulfuse:w:0xbf:m -Uhfuse:w:0xc4:m -Uflash:w:/Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file 0xbf for lfuse
avrdude: writing 1 byte lfuse ...

Writing | ################################################## | 100% 0.01s

avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xbf

Reading | ################################################## | 100% 0.00s

avrdude: 1 byte of lfuse verified
avrdude: reading input file 0xc4 for hfuse
avrdude: writing 1 byte hfuse ...

Writing | ################################################## | 100% 0.01s

avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xc4

Reading | ################################################## | 100% 0.00s

avrdude: 1 byte of hfuse verified
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex auto detected as Intel Hex
avrdude: reading input file /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex for flash
         with 15206 bytes in 4 sections within [0, 0x7fff]
         using 120 pages and 154 pad bytes
avrdude: writing 15206 bytes flash ...

Writing | ################################################## | 100% 1.31s

avrdude: 15206 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/ATTOBASICV234_m32-16MHZ-uart_btldr.hex

Reading | ################################################## | 100% 0.82s

avrdude: verification error, first mismatch at byte 0x7e00
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude done.  Thank you.

```

I also generated a 16kiB intel hex file with all zeros, and a 32kiB as well. Both of these files can be verified just fine...

```
$ srec_cat -generate 0x0000 0x3FFF -repeat_data 0x00 -o zeros16kib.hex -Intel
$ srec_cat -generate 0x0000 0x7FFF -repeat_data 0x00 -o zeros32kib.hex -Intel
$ ./avrdude -patmega32 -cavrispmkii -Ulfuse:w:0xbf:m -Uhfuse:w:0xc6:m

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: reading input file 0xbf for lfuse
avrdude: writing 1 byte lfuse ...

Writing | ################################################## | 100% 0.01s

avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xbf

Reading | ################################################## | 100% 0.00s

avrdude: 1 byte of lfuse verified
avrdude: reading input file 0xc6 for hfuse
avrdude: writing 1 byte hfuse ...

Writing | ################################################## | 100% 0.01s

avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xc6

Reading | ################################################## | 100% 0.00s

avrdude: 1 byte of hfuse verified

avrdude done.  Thank you.

$ ./avrdude -patmega32 -cavrispmkii -Uflash:w:zeros16kib.hex 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: input file zeros16kib.hex auto detected as Intel Hex
avrdude: reading input file zeros16kib.hex for flash
avrdude: writing 16383 bytes flash ...

Writing | ################################################## | 100% 1.49s

avrdude: 16383 bytes of flash written
avrdude: verifying flash memory against zeros16kib.hex

Reading | ################################################## | 100% 0.94s

avrdude: 16383 bytes of flash verified

avrdude done.  Thank you.

$ ./avrdude -patmega32 -cavrispmkii -Uflash:w:zeros32kib.hex 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: input file zeros32kib.hex auto detected as Intel Hex
avrdude: reading input file zeros32kib.hex for flash
avrdude: writing 32767 bytes flash ...

Writing | ################################################## | 100% 2.93s

avrdude: 32767 bytes of flash written
avrdude: verifying flash memory against zeros32kib.hex

Reading | ################################################## | 100% 1.85s

avrdude: 32767 bytes of flash verified

avrdude done.  Thank you.
```

> That kinda looks suspicious. Normally, wd for fuses is the same as for flash.

This is the table from the [ATmega32 datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/doc2503.pdf). Chaning the timing in avrdude.conf to match the values specified in the datasheet didn't make any difference.
[image]

stefanrueger, 2022-08-04:
Looks like a really hard nut. @MCUdude's test put paid to my theory about the boot section. Certain input files can be written certain input files cannot be written. The input file that does not work has 4 sections. Then @mcuee has found an input file with one section that also does not work. The OP seems to mention m328p can be written, the m32a cannot. 

There were clear-cut errors of the `m32` entry in `avrdude.conf`, but correcting these made no difference. I checked `stk500v2.c` and indeed these wd values are not used (only wd_chip_erase). Instead there are a couple of `usleep(10000)` with comments indicating the author thought that a hack (so hacking these up any further might make a change *but* that's not a satisfying solution even if it worked). Could it be that other vital `m32` `avrdude.conf` entries used in `stk500v2.c` are wrong? I have no idea where to look up the correct values needed, and which ones are actually used, so I am out of my depth here. But I think giving the `avrdude.conf` entry a long hard check will at least have the side-effect of getting that correct and removing this as potential error source.

Also, what is the flash readout with :I of the correctly written flash and what's the one of the incorrectly written flash? The OP seemed to say errors started earlier at `0x7d00` (which might be in a "hole" that's not part of the verification), and could be where the write attempt for `0x7e00` has diffracted to. A diff of the two read-backs might give some insight?

mcuee, 2022-08-05:
Indeed @MCUdude's testing results make this issue more difficult.

Two things -- maybe not relevant at all but just my guess from going through all the issues in recent months.
1) Can it be similar to the strange issue of #https://github.com/avrdudes/avrdude/issues/455? Apparent the proposed fix does not seem to be right there but it did fix the issue.
* https://github.com/avrdudes/avrdude/issues/455#issuecomment-991388243

2) Could it be related to https://github.com/avrdudes/avrdude/issues/1050?

mcuee, 2022-08-05:
I can repeat the problem as well. Writing has the problem, reading is good.
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c usbasp -U flash:v:m32a_muzi_avrisp2_readback.hex && echo OK

OK
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii -U lfuse:w:0xbf:m -U hfuse:w:0xc6:m && echo OK
avrdude.exe: stk500v2_command(): command failed
avrdude.exe: stk500v2_program_enable(): bad AVRISPmkII connection status: Target reverse inserted
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii 
-U lfuse:w:0xbf:m -U hfuse:w:0xc6:m && echo OK
OK

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii 
-U .\atmega32a_demo_muzi.hex && echo OK
avrdude.exe: verification error, first mismatch at byte 0x7e00
             0xff != 0x0f
avrdude.exe: verification error; content mismatch

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii 
-U flash:r:m32a_muzi_avrisp2_readback.hex && echo OK
OK

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c usbasp 
-U flash:v:m32a_muzi_avrisp2_readback.hex && echo OK
```

mcuee, 2022-08-05:
And as per @MCUdude, I need to remove all the empty lines before it will work. Even single empty lines will cause problems.

```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii 
-U .\atmega32a_demo_muzi_mod.hex && echo OK
avrdude.exe: verification error, first mismatch at byte 0x7e00
             0xff != 0x0f
avrdude.exe: verification error; content mismatch

C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii 
-U .\atmega32a_demo_muzi_mod1.hex && echo OK
OK

```

mcuee, 2022-08-05:
Actually I can reproduce the issue with very simple hex file like the following.

```
:020000040000FA
:20000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:20008000000000000000000000000000000000000000000000000000000000000000000060
:2000A000000000000000000000000000000000000000000000000000000000000000000040
:2000C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF40
:2000E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF20
:20010000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
:00000001FF

```

MCUdude, 2022-08-05:
> Actually I can reproduce the issue with very simple hex file like the following. 26bytes will be written wrongly as 0xFF.

The hex file you fail to be written to an ATmega8, ATmega16, or an ATmega32, but I can confirm that I can write it to an ATmega324P, ATmega328P, ATmega64, ATmega128, and ATmega1281.

ATmega32:
<details>

```
$ cat test-error.hex 
:020000040000FA
:20000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:20008000000000000000000000000000000000000000000000000000000000000000000060
:2000A000000000000000000000000000000000000000000000000000000000000000000040
:2000C000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF40
:2000E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF20
:20010000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
:00000001FF
$ ./avrdude -patmega32 -cavrispmkii -Uflash:w:test-error.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file test-error.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 128 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | #########################                          | 50% 0.00savrdude: stk500v2_command(): warning: Command timed out
avrdude: stk500v2_paged_write: write command failed
Writing | ################################################## | 100% 0.02s

avrdude: 128 bytes of flash written
avrdude: verifying flash memory against test-error.hex

Reading | ################################################## | 100% 0.02s

avrdude: 224 bytes of flash verified

avrdude done.  Thank you.

$ echo "read flash 0 0x100" | ./avrdude -patmega32 -cavrispmkii -t

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
>>> read flash 0 0x100 

Reading | ################################################## | 100% 0.01s

0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
[... 185 lines cut ...]
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9403 (probably m16)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: input file /Users/hans/Downloads/ATTOBASICV234_m16-16MHZ-uart_btldr.hex auto detected as Intel Hex
avrdude: reading input file /Users/hans/Downloads/ATTOBASICV234_m16-16MHZ-uart_btldr.hex for flash
         with 15204 bytes in 4 sections within [0, 0x3fff]
         using 120 pages and 156 pad bytes
avrdude: writing 15204 bytes flash ...

Writing | ################################################## | 100% 1.28s

avrdude: 15204 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/ATTOBASICV234_m16-16MHZ-uart_btldr.hex

Reading | ################################################## | 100% 0.82s

avrdude: verification error, first mismatch at byte 0x3e00
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

MCUdude, 2022-08-05:
Just to complete the test:

ATmega8535 (fails):
<details>

```
$ ./avrdude -patmega8535 -cavrispmkii -Uflash:w:test-error.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9308 (probably m8535)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file test-error.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 0 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.00s

avrdude: stk500v2_command(): warning: Command timed out
avrdude: stk500v2_paged_write: write command failed
avrdude: 128 bytes of flash written
avrdude: verifying flash memory against test-error.hex

Reading | ################################################## | 100% 0.02s

avrdude: 224 bytes of flash verified

avrdude done.  Thank you.
```

</details>

ATmega8515 (fails):
<details>

```
$ ./avrdude -patmega8515 -cavrispmkii -Uflash:w:test-error.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9306 (probably m8515)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file test-error.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 0 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.00s

avrdude: stk500v2_command(): warning: Command timed out
avrdude: stk500v2_paged_write: write command failed
avrdude: 128 bytes of flash written
avrdude: verifying flash memory against test-error.hex

Reading | ################################################## | 100% 0.01s

avrdude: 224 bytes of flash verified

avrdude done.  Thank you.
```

</details>

ATmega162 (succeeds):
<details>

```
$ ./avrdude -patmega162 -cavrispmkii -Uflash:w:test-error.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9404 (probably m162)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file test-error.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 128 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.01s

avrdude: 128 bytes of flash written
avrdude: verifying flash memory against test-error.hex

Reading | ################################################## | 100% 0.03s

avrdude: 224 bytes of flash verified

avrdude done.  Thank you.
```

</details>

mcuee, 2022-08-05:
@MCUdude 

I see some error in your run log.
```
avrdude: stk500v2_command(): warning: Command timed out
avrdude: stk500v2_paged_write: write command failed
```

Interestingly I do not see such warnings communication issue warnings myself with ATmega32A.
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p m32a -c avrispmkii  -U .\simpletest1.hex -v

avrdude.exe: Version 7.0-20220804 (5f5002e)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : avrispmkii
avrdude.exe: usbdev_open(): Found AVRISP mkII, serno: 001D2C990079
             AVR Part                      : ATmega32A
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                  4    10    64    0 no       1024    4      0  9000  9000 0xff 0xff
               flash                  33     6    64    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

             Programmer Type : STK500V2
             Description     : Atmel AVR ISP mkII
             Programmer Model: AVRISP mkII
             Hardware Version: 1
             Firmware Version Master : 1.24
             Vtarget         : 4.9 V
             SCK period      : 4.00 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9502 (probably m32a)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: input file .\simpletest1.hex auto detected as Intel Hex
avrdude.exe: reading input file .\simpletest1.hex for flash
             with 128 bytes in 2 sections within [0, 0x11f]
             using 2 pages and 128 pad bytes, cutting off 96 trailing 0xff bytes
avrdude.exe: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.02s

avrdude.exe: 128 bytes of flash written
avrdude.exe: verifying flash memory against .\simpletest1.hex

Reading | ################################################## | 100% 0.08s

avrdude.exe: verification error, first mismatch at byte 0x0080
             0xff != 0x00
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

```

mcuee, 2022-08-05:
Looks like the issue has been there for long time. I switched the driver from Microchip official WinUSB based driver to libusbk.sys (you can also use libusb0.sys) to use old release version of avrdude.

```
C:\work\avr\avrdude_test\release\avrdude-6.0.1-mingw32> .\avrdude -p m32 -c avrispmkii 
 -P usb -U .\simpletest1.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9502
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\simpletest1.hex"
avrdude.exe: input file .\simpletest1.hex auto detected as Intel Hex
avrdude.exe: writing flash (192 bytes):

Writing | ################################################## | 100% 0.03s

avrdude.exe: 192 bytes of flash written
avrdude.exe: verifying flash memory against .\simpletest1.hex:
avrdude.exe: load data flash data from input file .\simpletest1.hex:
avrdude.exe: input file .\simpletest1.hex auto detected as Intel Hex
avrdude.exe: input file .\simpletest1.hex contains 192 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.08s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x0080
             0xff != 0x00
avrdude.exe: verification error; content mismatch

avrdude.exe: safemode: Fuses OK (H:FF, E:C6, L:BF)

avrdude.exe done.  Thank you.
```

mcuee, 2022-08-05:
But the older version seems to be okay.
avrdude-5.11-Patch7610-win32 was released on 31-Aug-2011.

```
C:\work\avr\avrdude_test\release\avrdude-5.11-Patch7610-win32> .\avrdude -p m32 -c avrispmkii  -P usb -U .\simpletest1.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9502
avrdude.exe: NOTE: FLASH memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\simpletest1.hex"
avrdude.exe: input file .\simpletest1.hex auto detected as Intel Hex
avrdude.exe: writing flash (192 bytes):

Writing | ################################################## | 100% 0.02s

avrdude.exe: 192 bytes of flash written
avrdude.exe: verifying flash memory against .\simpletest1.hex:
avrdude.exe: load data flash data from input file .\simpletest1.hex:
avrdude.exe: input file .\simpletest1.hex auto detected as Intel Hex
avrdude.exe: input file .\simpletest1.hex contains 192 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude.exe: verifying ...
avrdude.exe: 192 bytes of flash verified

avrdude.exe: safemode: Fuses OK

avrdude.exe done.  Thank you.

```

MCUdude, 2022-08-05:
> But the older version seems to be okay.

It would be interesting if we could trace it down to a single commit. However, I have no experience with SVN, and I'm having a hard time even finding the old source code that is (was?) hosted over at [Savannah](https://savannah.nongnu.org/projects/avrdude/).

mcuee, 2022-08-05:
> > But the older version seems to be okay.
> 
> It would be interesting if we could trace it down to a single commit. However, I have no experience with SVN, and I'm having a hard time even finding the old source code that is (was?) hosted over at [Savannah](https://savannah.nongnu.org/projects/avrdude/).

@MCUdude 
No need to go to SVN. This repo has all the histories.

Example for stk500v2.c.
https://github.com/avrdudes/avrdude/commits/5633a6d88ab199b68ba68d53717efc21c0cc5868/stk500v2.c?browsing_rename_history=true&new_path=src/stk500v2.c&original_branch=main

MCUdude, 2022-08-05:
@mcuee I did have a look at the changes done to stk500v2.c from 09/2011 to 09/2013 (changes from 5.11 to 6.0.1) but I couldn't really find anything that could cause this. Either I didn't search well enough, or the issue may be somewhere else. I don't know really.

mcuee, 2022-08-05:
> @mcuee I did have a look at the changes done to stk500v2.c from 09/2011 to 09/2013 (changes from 5.11 to 6.0.1) but I couldn't really find anything that could cause this. Either I didn't search well enough, or the issue may be somewhere else. I don't know really.

Indeed it could be other files as well but that can be strange.
Example: the following issue is somewhat related to fileio.c but it could be incorrect fix.
* https://github.com/avrdudes/avrdude/issues/455

fileio.c change history.
https://github.com/avrdudes/avrdude/commits/5633a6d88ab199b68ba68d53717efc21c0cc5868/fileio.c?browsing_rename_history=true&new_path=src/fileio.c&original_branch=main

mcuee, 2022-08-05:
I have futher narrow down that this commit is already bad (20-Nov-2012)
https://github.com/avrdudes/avrdude/tree/147137a218a41e672cc301659cc5782afd1a615a

mcuee, 2022-08-05:
Just want to document the process I used to zoom down to the paticular commits. I am not good at git so I do not know how to use `git bisect` but I kind of follow the principal.

```
 2025  git clone https://github.com/avrdudes/avrdude.git
 2026  cd avrdude
 2027  ls
 2028  git checkout 147137a
 2029  git log
 2030  ls
 2031  ./bootstrap 
 2032  ./configure 
 2033  make
 2034  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../simpletest1.hex
 2035  ls
 2036  rm INSTALL 
 2037  git checkout main
 2038  git pull
 2039  git diff
 2040  git checkout f4db29d
 2041  ./bootstrap 
 2042  ./configure 
 2043  make
 2044  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../simpletest1.hex
 2045  rm INSTALL 
 2046  git checkout main
 2047  git checkout c3d58bc
 2048  ./bootstrap 
 2049  ./configure 
 2050  make
 2051  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../simpletest1.hex
 2052  cd ..
 2053  ls
 2054  mv ~/Downloads/avrdude-5.11.1.tar.gz .
 2055  tar zxvf avrdude-5.11.1.tar.gz 
 2056  cd avrdude-5.11.1/
 2057  ls
 2058  cd avrdude/
 2059  ls
 2060  ./bootstrap 
 2061  ./configure 
 2062  make
 2063  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../simpletest1.hex
 2064  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../../simpletest1.hex
 2065  cd ..
 2066  cd avrdude
 2067  ls
 2068  rm INSTALL 
 2069  git checkout main
 2070  git checkout 0926a2c
 2071  ./bootstrap 
 2072  ./configure 
 2073  make
 2074  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../simpletest1.hex
 2075  rm INSTALL 
 2076  git checkout main
 2077  git checkout d742827
 2078  ./bootstrap 
 2079  ./configure 
 2080  make
 2081  sudo ./avrdude -C ./avrdude.conf -p m32 -c avrispmkii -P usb -U ../simpletest1.hex
 2082  git log
 2083  ls
 2084  rm INSTALL 
 2085  git checkout main

```

mcuee, 2022-08-05:
@dl8dtl and @stefanrueger and @MCUdude 
Hopefuly now you can look at the paticual mega commit to see where is the issue. Thanks.
https://github.com/avrdudes/avrdude/commit/d742827da15a8e6165d956735ec48f1acff575e3

mcuee, 2022-08-06:
> Great work @mcuee! Sadly, it's a huge commit, and I don't really know what to look for. But now tracking it down to a single commit is still _very_ helpful clap

Ineed it is a **mega** commit. Hopefully @dl8dtl or @stefanrueger can look at the commit and see where is the issue.

mcuee, 2022-08-06:
BTW, the mega commit touched quite some programmers and USBASP is one of them. It also got the same issue at tthat time. But later it was fixed.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude$ ./avrdude -C ./avrdude.conf -p m32 -c usbasp -P usb -U ../simpletest1.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502
avrdude: NOTE: FLASH memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "../simpletest1.hex"
avrdude: input file ../simpletest1.hex auto detected as Intel Hex
avrdude: writing flash (192 bytes):

Writing | ################################################## | 100% 0.06s

avrdude: 192 bytes of flash written
avrdude: verifying flash memory against ../simpletest1.hex:
avrdude: load data flash data from input file ../simpletest1.hex:
avrdude: input file ../simpletest1.hex auto detected as Intel Hex
avrdude: input file ../simpletest1.hex contains 192 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.06s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0080
         0xff != 0x00
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK

avrdude done.  Thank you.

mcuee@UbuntuSwift3:~/build/avr/avrdude_release/avrdude$ avrdude -p m32 -c usbasp -P usb -U ../simpletest1.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: input file ../simpletest1.hex auto detected as Intel Hex
avrdude: reading input file ../simpletest1.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 128 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.08s

avrdude: 128 bytes of flash written
avrdude: verifying flash memory against ../simpletest1.hex

Reading | ################################################## | 100% 0.06s

avrdude: 224 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-08-06:
For usbasp, it was fixed by the following commit.
https://github.com/avrdudes/avrdude/commit/0b3feb71b81e9f12a21994526897f6d8e4446db6

mcuee, 2022-08-06:
For USBtinyISP, the mega commit also got the same issue (along with some other issues). But then it is fixed by similar commits as the above-mentioned usbasp fix.
https://github.com/avrdudes/avrdude/commit/bde3c841e5c9563ab696e9df9c82f5a64ac36a42

mcuee, 2022-08-06:
For the original STK500 v1 FW, there is no issue with the mega commit and latest git main.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/bin_archive_git$ ./avrdude_bde3c84 
-C ./avrdude_bde3c84.conf -qqp m32 -c stk500v1 -P /dev/ttyUSB0 -U ../simpletest1.hex && echo OK
OK
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/bin_archive_git$ avrdude -qqp m32 
-c stk500v1 -P /dev/ttyUSB0 -U ../simpletest1.hex && echo OKOK

```

mcuee, 2022-08-06:
For JTAg ICE mki, there was issues to run avrdude with mage commit and release 6.0.1/6.1/6.2, release 6.3 onwards works fine. And there is no issues with latets git main.

mcuee, 2022-08-06:
The mega commit also affects avr109 bootloader, avr910, avrftdi and jtagmkii. I will see if I can test avr109 and avrftdi with my ATmega32 later.

@dl8dtl and @MCUdude 
Not so sure if you can help testing the avr910 and jtagmkii. Most likely jtagmkii will have the same issue as it is similar to AVR Dragon. I am still waiting for my AVR910 programmer to come.

mcuee, 2022-08-06:
I can confirm that avr109 is okay for the git main.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/bin_archive_git$ avrdude -qqp m32a -c avr109 -P /dev/ttyUSB0 -b 115200 -D -U ../simpletest1.hex && echo OK
Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.5; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x7f

OK

```

mcuee, 2022-08-06:
For avrftdii, git main is having strange thing with the test hex file.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/bin_archive_git$ avrdude -qqp m32 -c jtagkey -U ../simpletest1.hex -v

avrdude: Version 7.0-20220804 (5f5002e)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/usr/local/bin/../etc/avrdude.conf"
         User configuration file is "/home/mcuee/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : jtagkey
         AVR Part                      : ATmega32
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    10    64    0 no       1024    4      0  9000  9000 0xff 0xff
           flash                  33     6    64    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  2000  2000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : avrftdi
         Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2
W avrftdi_flash_write(1092): Skipping empty page (containing only 0xff bytes)

```

mcuee, 2022-08-06:
Looking at the avrftdi.c history, there are not many changes after the 6.2 release on 17 Feb 2016.
https://github.com/avrdudes/avrdude/commits/5633a6d88ab199b68ba68d53717efc21c0cc5868/avrftdi.c?browsing_rename_history=true&new_path=src/avrftdi.c&original_branch=mainl

Most likely the fix in this commit helps the things but may not sort all the issues.
https://github.com/avrdudes/avrdude/commit/eb461f706f35611d8d968bc21197dae3a8f64c78

The above commit somehow does not work for me though.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_release/bin_archive_git$ ./avrdude_eb461f7 
-C ./avrdude_eb461f7.conf -c jtagkey -qqp m32 -U ../simpletest1.hex 
avrdude_eb461f7: error at ./avrdude_eb461f7.conf:4422: can't find parent part
avrdude_eb461f7: error reading system wide configuration file "./avrdude_eb461f7.conf"

```

mcuee, 2022-08-06:
Just a summary of this issue.
1) the root cause of the issue is the mega commit https://github.com/avrdudes/avrdude/commit/d742827da15a8e6165d956735ec48f1acff575e3

The mega commit affects multiple programmers but some of them (like jtagmki, usbasp and usbtiny) have since fixed the issue. It does not seem to affect avr109 bootloader. 

1) To be tested: avr910 programmer 
2) Confirmed and need to be fixed: stk500v2, jtagmkii (AVR Dragon and AVR JTAGICE mkii)
3) For avrftdi, the paticular issue may have been fixed but there is still issues.

mcuee, 2022-08-07:
@dl8dtl and @stefanrueger 

Hopefully you now have the info to identidy the problem and the hopefully a simple fix can be done for stk500v2 and jtagemkii. Thanks.

mcuee, 2022-08-21:
avr910 does not seem to be affected (probably similar to avr109)

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p m32a -c avr910 -P COM8 -U .\simpletest1.hex

Found programmer: Id = "AVR ISP"; type = S
    Software Version = 3.8; Hardware Version = 1.2
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize = 64 bytes.

Programmer supports the following devices:
    Device code: 0x13 = AT90S1200
    Device code: 0x20 = AT90S2313
    Device code: 0x28 = AT90S4414
    Device code: 0x30 = AT90S4433
    Device code: 0x34 = AT90S2333
    Device code: 0x38 = AT90S8515
    Device code: 0x48 = (unknown)
    Device code: 0x4c = AT90S2343
    Device code: 0x51 = (unknown)
    Device code: 0x55 = ATtiny12
    Device code: 0x56 = ATtiny15
    Device code: 0x68 = AT90S8535
    Device code: 0x6c = AT90S4434
    Device code: 0x86 = (unknown)
    Device code: 0x87 = (unknown)
    Device code: 0x01 = (unknown)
    Device code: 0x04 = (unknown)
    Device code: 0x05 = (unknown)
    Device code: 0x06 = (unknown)
    Device code: 0x07 = (unknown)
    Device code: 0x08 = (unknown)
    Device code: 0x09 = (unknown)
    Device code: 0x0e = (unknown)
    Device code: 0x1a = (unknown)
    Device code: 0x0f = (unknown)
    Device code: 0x1b = (unknown)
    Device code: 0x14 = (unknown)
    Device code: 0x1c = (unknown)
    Device code: 0x1d = (unknown)
    Device code: 0x23 = (unknown)
    Device code: 0x31 = (unknown)
    Device code: 0x33 = (unknown)
    Device code: 0x35 = (unknown)
    Device code: 0x37 = (unknown)
    Device code: 0x39 = (unknown)
    Device code: 0x3d = (unknown)
    Device code: 0x3e = (unknown)
    Device code: 0x57 = (unknown)
    Device code: 0x3a = ATmega8515
    Device code: 0x3b = (unknown)
    Device code: 0x41 = ATmega103
    Device code: 0x43 = ATmega128
    Device code: 0x44 = (unknown)
    Device code: 0x45 = ATmega64
    Device code: 0x46 = (unknown)
    Device code: 0x5e = ATtiny26
    Device code: 0x60 = ATmega161
    Device code: 0x61 = (unknown)
    Device code: 0x62 = (unknown)
    Device code: 0x63 = ATmega162
    Device code: 0x64 = ATmega163
    Device code: 0x66 = (unknown)
    Device code: 0x69 = ATmega8535
    Device code: 0x6a = (unknown)
    Device code: 0x72 = ATmega32
    Device code: 0x73 = (unknown)
    Device code: 0x74 = ATmega16
    Device code: 0x75 = ATmega329
    Device code: 0x76 = ATmega8
[... 100 lines cut ...]
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9502 (probably m32a)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file .\ATTOBASICV234_m32-16MHZ-uart_btldr.hex for flash
             with 15206 bytes in 4 sections within [0, 0x7fff]
             using 120 pages and 154 pad bytes
avrdude.exe: writing 15206 bytes flash ...

Writing | ################################################## | 100% 15.81s

avrdude.exe: 15206 bytes of flash written
avrdude.exe: verifying flash memory against .\ATTOBASICV234_m32-16MHZ-uart_btldr.hex

Reading | ################################################## | 100% 9.11s

avrdude.exe: 15206 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-08-21:
Updated summary:
**The root cause of the issue is the mega commit** https://github.com/avrdudes/avrdude/commit/d742827da15a8e6165d956735ec48f1acff575e3 on 15 Seot 2011.

The mega commit affects multiple programmers but some of them (like jtagmki, avrftdi, usbasp and usbtiny) have since fixed the issue. It does not seem to affect avr109/avr910 and STK500v1. 

**Confirmed and need to be fixed: stk500v2, jtagmkii (AVR Dragon and AVR JTAGICE mkii) in ISP mode**

mcuee, 2022-09-07:
@dl8dtl 
Maybe you can take a look at this one as you are looking at the stk500v2 codes. Thanks.

mcuee, 2022-11-05:
I want to revisit the issue to see if there is a simple fix or not.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii -U lfuse:w:0xbf:m
 -U hfuse:w:0xc6:m && echo OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qqp m32a -c avrispmkii -U .\simpletest1.hex 
&& echo OK
avrdude error: verification mismatch, first encountered at addr 0x0080
        device 0xff != input 0x00
avrdude error: verification mismatch
```

mcuee, 2022-11-05:
Different clones seem to have quite different performances.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c stk500v2 -P COM4 -B 1 -p m2560 -U .\zeros256KB.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9801 (probably m2560)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\zeros256KB.hex for flash
         with 262143 bytes in 1 section within [0, 0x3fffe]
         using 1024 pages and 1 pad byte
avrdude: writing 262143 bytes flash ...

Writing | ################################################## | 100% 64.75 s

avrdude: 262143 bytes of flash written
avrdude: verifying flash memory against .\zeros256KB.hex

Reading | ################################################## | 100% 53.60 s

avrdude: 262143 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-11-05:
The more exact STK500v2 clone (using ATmega8535) will show the issue.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p m32a -c stk500v2 -P COM5 -B1 -U .\simpletest1.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\simpletest1.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 128 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.05 s

avrdude: 128 bytes of flash written
avrdude: verifying flash memory against .\simpletest1.hex

Reading | ################################################## | 100% 0.06 s

avrdude error: verification mismatch, first encountered at addr 0x0080
        device 0xff != input 0x00
avrdude error: verification mismatch

avrdude done.  Thank you.

```

mcuee, 2022-11-06:
> Sounds to me like a bug in the stk500v2 code (which is used for Dragon, STK500, AVRISPmkII and so on).

@dl8dtl 
Does this apply to PICKit 4 too? I can reproduce the issue with pickit4_isp.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -p m32a -c pickit4_isp -U .\simpletest1.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9502 (probably m32a)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
erasing chip
avrdude: reading input file .\simpletest1.hex for flash
         with 128 bytes in 2 sections within [0, 0x11f]
         using 2 pages and 128 pad bytes, cutting off 96 trailing 0xff bytes
avrdude: writing 128 bytes flash ...

Writing | ################################################## | 100% 0.05 s

avrdude: 128 bytes of flash written
avrdude: verifying flash memory against .\simpletest1.hex

Reading | ################################################## | 100% 0.16 s

avrdude error: verification mismatch, first encountered at addr 0x0080
        device 0xff != input 0x00
avrdude error: verification mismatch

avrdude done.  Thank you.
```

mcuee, 2022-11-06:
I think @MCUdude's test results may ring a bell as well. For the simple hex file, he got the following errors and then the flash was not written correctly for ATmega8/16/32, no matter the fuse settings.
```
avrdude: stk500v2_command(): warning: Command timed out
avrdude: stk500v2_paged_write: write command failed
```

With the more exact STK500 v2 clone (using ATmega8535) and Linux, I can reproduce the issue.

Here is one example using ATmega8A. I can also further remove the trailing 96 bytes of 0xFF and I can still reproduce the issue.

The bad thing is that it actually verifies okay but in reality did not write anything. So if you ignore the warning messages, you may think the programming is actually good but in reality it is not good.

```

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -p m8a -qq -U simpletest2.hex 
&& echo OK
avrdude warning: Command timed out
avrdude error: write command failed
OK

mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -p m8a -qq -t
avrdude> dump flash
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit
avrdude> mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ cat simpletest2.hex 
:020000040000FA
:20000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00
:20002000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE0
:20008000000000000000000000000000000000000000000000000000000000000000000060
:2000A000000000000000000000000000000000000000000000000000000000000000000040
:00000001FF

```
If I remove the leading 0xFF and then it will be really okay.
```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ cat simpletest3.hex 
:020000040000FA
:20008000000000000000000000000000000000000000000000000000000000000000000060
:2000A000000000000000000000000000000000000000000000000000000000000000000040
:00000001FF
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -p m8a -qq -U simpletest3.hex 
&& echo OK
OK
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -p m8a -qq -t
avrdude> dump flash
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
0090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> quit

```

mcuee, 2022-11-06:
The run log may be useful as well.

```
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -p m8a -U simpletest3.hex 
-vvvv >runlogok.txt 2>&1
mcuee@UbuntuSwift3:~/build/avr/avrdude_bin$ ./avrdude -c stk500v2 -P /dev/ttyUSB0 -p m8a -U simpletest2.hex 
-vvvv >runlog2nok.txt 2>&1
```

[runlogok.txt](https://github.com/avrdudes/avrdude/files/9945133/runlogok.txt)
[runlog2nok.txt](https://github.com/avrdudes/avrdude/files/9945134/runlog2nok.txt)

mcuee, 2022-11-06:
Good old release for Windows.

```
PS C:\work\avr\avrdude_test\avrdude_binary_archive\avrdude-5.11-Patch7610-win32> .\avrdude -p m8 -c stk500v2 -P COM5 -B1 -U .\simpletest2.hex
avrdude.exe: stk500v2_set_sck_period(): p = 1.0 us too small, using 1.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9307
avrdude.exe: NOTE: FLASH memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\simpletest2.hex"
avrdude.exe: input file .\simpletest2.hex auto detected as Intel Hex
avrdude.exe: writing flash (192 bytes):

Writing | ################################################## | 100% 0.03s

avrdude.exe: 192 bytes of flash written
avrdude.exe: verifying flash memory against .\simpletest2.hex:
avrdude.exe: load data flash data from input file .\simpletest2.hex:
avrdude.exe: input file .\simpletest2.hex auto detected as Intel Hex
avrdude.exe: input file .\simpletest2.hex contains 192 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude.exe: verifying ...
avrdude.exe: 192 bytes of flash verified

avrdude.exe: safemode: Fuses OK

avrdude.exe done.  Thank you.

```

[30 more replies]

PR #1273 (2023-01-14): Only send write flash cmd to STK500v2 programmers if page not empty
... except if it's for a bootloader: they might need to know about 0xff pages if they use an SPM page erase command and therefore memory does not look like NOR-memory.

This is a slight optimisation for all parts (as writing 0xff pages on NOR-memory flash is a NOP), but more importantly, prevents a firmware error to surface for those parts that need to carry out value polling to determine when writing the page has finished on the device.

Fixes #425 

Needs testing with affected parts (ATmega8A) and a flash .hex file with an e [... cut]
closed 2023-01-14 by stefanrueger