avrdudes/avrdude #429: [bug #48767] ser_drain() for TCP on Windows doesn't work

avrdude-admin, 2021-12-10:
fibroin <ahalbach@gmx.de>
Fri 12 Aug 2016 08:10:44 PM UTC

I have an Arduino Micro with Optiboot-Bootloader and I can upload sketches with AVRDude over TCP on Linux.

I tried to use AVRDude 6.3 on Windows but AVRDude does not correctly drain the read buffer and so AVRDude gets confused.

In ser_win32.c there is a special version of ser_recv() for TCP-connections, but ser_drain() uses the version for serial connections. But it doesn't work.

I get the output as attached.
The read buffer isn't drained so avrdude recives the incoming data from the initial getsync later.

[file #38191: avrdude fehler.txt](https://savannah.nongnu.org/bugs/download.php?file_id=38191)

This issue was migrated from https://savannah.nongnu.org/bugs/?48767

avrdude-admin, 2021-12-10:
André Pedro <apedro>
Sat 24 Sep 2016 02:16:41 AM UTC

There is an avrdude issue that is a possible bug on socket send/receive methods for win32.

I have run avrdude on verbose mode and I realize that after the first pack of bytes that have been sent "0x30 0x20" the avrdude does not wait for an answer (the bytes came after due to some OS delay). Basically, the income stream gets filled with 0x14 0x10 0x14 0x10 that will be used for the next commands (that of course cause the programming to fail). Do you have an idea how simply solve it ? I'm using the last version 3.6.

avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_drain(): read error: The parameter is incorrect.

avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: ser_drain(): read error: The parameter is incorrect.

avrdude.exe: Send: 0 [30]   [20]
avrdude.exe: Recv: . [14]
avrdude.exe: Recv: . [10]

avrdude-admin, 2021-12-10:
Christopher Cooper <ccooper21>
Thu 30 Mar 2017 01:34:23 AM UTC

See patch [#9304](https://savannah.nongnu.orghttps://savannah.nongnu.org/patch/index.php?9304) for a solution.  I've also committed the patch to [my GitHub repository](https://savannah.nongnu.orghttps://github.com/ccooper21/avrdude).

avrdude-admin, 2021-12-10:
Christopher Cooper <ccooper21>
Thu 13 Apr 2017 11:37:36 PM UTC

**UPDATED LINKS:** See [patch #9304](https://savannah.nongnu.orghttps://savannah.nongnu.org/patch/?9304) for a solution.  I've also committed the patch to [my GitHub repository](https://savannah.nongnu.orghttps://github.com/ccooper21/avrdude/tree/ws-ser_drain).

avrdude-admin, 2021-12-10:
Adriano Costanzo <vashcode>
Sat 22 Apr 2017 10:38:46 PM UTC

Hi ccooper21, I tried your patch but it's not working.

The avrdude is blocked, and this is the log:

avrdude: Version 6.3, compiled on Apr 22 2017 at 23:54:11
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "C:\Users\vasht\Desktop\arduino-1.8.2\hardware\tools\avr/etc/avrdude.conf"

Using Port                    : net:192.168.1.109:328
Using Programmer              : arduino
An error occurred while uploading the sketch

I use Windows 10; the same avrdude works on Linux but in Windows I have the same problem that it discussed on this issue and if I use your patch you can see above the problem.

avrdude-admin, 2021-12-10:
Adriano Costanzo <vashcode>
Sat 22 Apr 2017 11:03:28 PM UTC

Sorry, the problem was arduino patch. I suggest to you, please don't add arduino patch when you compile the avrdude, I based my project on: [https://github.com/arduino/avrdude-build-script](https://savannah.nongnu.orghttps://github.com/arduino/avrdude-build-script)
and these project add arduino patch, and these not works !!!

Your fix is perfect !!!

thank you so much

avrdude-admin, 2021-12-10:
Christopher Cooper <ccooper21>
Wed 26 Apr 2017 09:03:04 PM UTC

Adriano, thanks for testing my patch!  I'm glad to hear that it is working well for you.

It is odd that you had problems using the "avrdude-build-script" project, as I used the same to build and test my patch.  When I made the patch, I didn't find any other patches that affected the same file so its not like there should be a conflict.  Maybe another patch was added since I did my work that is affecting you.  I'm not sure.

-- Chris

mcuee, 2022-05-30:
As per https://savannah.nongnu.org/patch/?9304, patch 9304 has been merged in r1499.

And indeed the current code has the patch already. 
https://savannah.nongnu.org/patch/?9304

Therefore I think this can be closed.