avrdudes/avrdude #430: [bug #48776] avrdude attempts to free() an invalid address

avrdude-admin, 2021-12-10:
esot_eric <None>
Sun 14 Aug 2016 01:40:35 AM UTC

READING the flash into an ihex file: AVR-Dude crashes with:

- - - Error in `avrdude': free(): invalid next size (normal): 0x09323798 ***Aborted

- Linux x86-32bit- ATmega8515* Tried later with ATtiny861, same problem. (also with eeprom-read)

- avr-dude v6.1* Tried later with v6.3, same problem.

- usb-tiny-isp v1.04 (this is the version that comes from adafruit)* Tried later with FT2232H-based programmer, NO PROBLEM.
*  v1.07 has been tried with same problem at second link, below

The ihex file is written and appears to be correct.

VERY LITTLE info 'round the web about this error-message...

I think, mostly, due to the fact that the message isn't coming from avrdude, but from glibc(?)... most OS's probably don't actually print-out an error-message when free() is given an invalid address.

Here's what I've found:

[http://www.avrfreaks.net/forum/error-avrdude-free-invalid-next-size-normal-0x000000000112c720](https://savannah.nongnu.orghttp://www.avrfreaks.net/forum/error-avrdude-free-invalid-next-size-normal-0x000000000112c720)

[https://bugs.launchpad.net/ubuntu/+source/avrdude/+bug/1400185](https://savannah.nongnu.orghttps://bugs.launchpad.net/ubuntu/+source/avrdude/+bug/1400185)

[http://www.mikrocontroller.net/attachment/301320/avrdude_crash.txt](https://savannah.nongnu.orghttp://www.mikrocontroller.net/attachment/301320/avrdude_crash.txt)

[http://savannah.nongnu.org/bugs/?41292](https://savannah.nongnu.orghttp://savannah.nongnu.org/bugs/?41292) (not a usb-tiny-isp)

Otherwise, search-fu is failing me...

Again, from what I can tell, I think this "bug" is probably more-existant than these results, but I don't know what to search for, because, again, most OS's probably don't even report an error, and the file **is** created, so basically it **looks** like the program's exitting normally.

I've done a tiny bit of debugging (within my skillset) and determined that my encounter with this error appears to occur within avrpart.c:

"avr_free_mem() -> m->buf = 0x9323798"

(maybe it was already freed?)

This issue was migrated from https://savannah.nongnu.org/bugs/?48776

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 13 Sep 2016 08:02:44 PM UTC

Well, my major problem I'm having with this bug report is, I
cannot reproduce it.

I've got an USBtiny here (don't know which firmware level),
I tried to run AVRDUDE both with Valgrind as well as with
jemalloc's option to junk-fill the allocated memory.  No
abnormal behaviour is observable.

If you can, I'd ask you to run AVRDUDE under control of a
debugger, in the hope it might reveal a bit more of the
actual problem.

avrdude-admin, 2021-12-10:
Esot.Eric <esot_eric>
Sat 24 Sep 2016 02:47:09 AM UTC

Cool, thanks for looking into this!

Am unfamiliar with most debugging tools, though I did attempt to use valgrind. In so-doing, it updated libc6, which apparently caused it to replace the "free()" message with "Segmentation Fault".

Oddities surrounding use of 'sudo' (a few times, running sudo, it didn't segfault, now it does. Same effect as not running sudo.).

Again, the read-process completes, the file is written, but the segfault occurs after that.

---------------
Here's the output:
$ make read
rm -f flash.hex
valgrind --leak-check=yes avrdude -c usbtiny -pt861 -U flash:r:flash.hex:i
==2729== Memcheck, a memory error detector
==2729== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==2729== Using Valgrind-3.10.0 and LibVEX; rerun with -h for copyright info
==2729== Command: avrdude -c usbtiny -pt861 -U flash:r:flash.hex:i
==2729==
==2729== Invalid read of size 1
==2729==    at 0x407ADD0: usb_parse_descriptor (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x407B709: usb_parse_configuration (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x407D08B: usb_os_find_devices (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x407A791: usb_find_devices (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x8064026: usbtiny_open (usbtiny.c:224)
==2729==    by 0x804BB87: main (main.c:956)
==2729==  Address 0x479cada is 0 bytes after a block of size 18 alloc'd
==2729==    at 0x40291CC: malloc (vg_replace_malloc.c:296)
==2729==    by 0x407D02F: usb_os_find_devices (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x407A791: usb_find_devices (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x8064026: usbtiny_open (usbtiny.c:224)
==2729==    by 0x804BB87: main (main.c:956)
==2729==

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude: Device signature = 0x1e930d (probably t861)
avrdude: reading flash memory:

Reading | ################################################## | 99% 22.38s==2729== Syscall param ioctl(USBDEVFS_CONTROL).data points to unaddressable byte(s)
==2729==    at 0x4214AC9: ioctl (syscall-template.S:81)
==2729==    by 0x407C88C: usb_control_msg (in /lib/i386-linux-gnu/libusb-0.1.so.4.4.4)
==2729==    by 0x8063DA9: usb_in (usbtiny.c:123)
==2729==    by 0x80647B7: usbtiny_paged_load (usbtiny.c:461)
==2729==    by 0x804EFD5: avr_read (avr.c:405)
==2729==    by 0x80650F1: do_op (update.c:236)
==2729==    by 0x804C4E6: main (main.c:1227)
==2729==  Address 0x4755878 is 0 bytes after a block of size 8,192 alloc'd
==2729==    at 0x40291CC: malloc (vg_replace_malloc.c:296)
==2729==    by 0x8050C44: avr_initmem (avrpart.c:273)
==2729==    by 0x804B897: main (main.c:891)
==2729==
Reading | ################################################## | 100% 22.55s

avrdude: writing output file "flash.hex"

avrdude: safemode: Fuses OK (E:FF, H:DF, L:E1)

avrdude done.  Thank you.

==2729==
==2729== HEAP SUMMARY:
==2729==     in use at exit: 88,734 bytes in 99 blocks
==2729==   total heap usage: 24,085 allocs, 23,986 frees, 3,700,669 bytes allocated
==2729==
==2729== LEAK SUMMARY:
==2729==    definitely lost: 0 bytes in 0 blocks
==2729==    indirectly lost: 0 bytes in 0 blocks
==2729==      possibly lost: 0 bytes in 0 blocks
==2729==    still reachable: 88,734 bytes in 99 blocks
==2729==         suppressed: 0 bytes in 0 blocks
==2729== Reachable blocks (those to which a pointer was found) are not shown.
==2729== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==2729==
==2729== For counts of detected and suppressed errors, rerun with: -v
==2729== ERROR SUMMARY: 3 errors from 2 contexts (suppressed: 0 from 0)
-------------------
here's my makefile:
VALGRIND = valgrind --leak-check=yes
#VALGRIND =
MCU = t861
AVRDUDE_PROGRAMMER = usbtiny
AVRDUDE = avrdude -c $(AVRDUDE_PROGRAMMER) -p$(MCU)

read:
rm -f flash.hex
$(VALGRIND) $(AVRDUDE) -U flash:r:flash.hex:i

# $(AVRDUDE) -U eeprom:r:$(BACKUP_DIR)/eeprom.hex:i

-------------
If you've got specific instructions for what to try, lemme know.

Oh, currently it's connected via USB2.0 hub, if that helps.

avrdude-admin, 2021-12-10:
Kārlis Seņko <karliss>
Sun 28 May 2017 12:09:37 PM UTC

I think I found the problem. In usbtiny_page_load chunk_size was 128 but n_bytes 64. This could result in buffer overflow corrupting memory and causing free to fail. Limiting the chunk size fixed the crash.

avrdude-admin, 2021-12-10:
Charlie Green <bytesquid>
Fri 25 Aug 2017 01:43:54 AM UTC

I just encountered this problem as well in avrdude 6.3-2; here's a stack trace and memory map, in case you want it.

$ avrdude -c usbtiny -p m644 -U eeprom:r:eeprom.bkp.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9609 (probably m644)
avrdude: reading eeprom memory:

Reading | ################################################## | 100% 31.47s

avrdude: writing output file "eeprom.bkp.hex"

avrdude: safemode: Fuses OK (E:FD, H:DC, L:FE)

avrdude done.  Thank you.

*** Error in `avrdude': free(): invalid next size (normal): 0x000055d662d446a0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bcb)[0x7f8d22582bcb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76f96)[0x7f8d22588f96]
/lib/x86_64-linux-gnu/libc.so.6(+0x777de)[0x7f8d225897de]
avrdude(+0xb1d5)[0x55d6625f01d5]
avrdude(+0xcafe)[0x55d6625f1afe]
avrdude(+0xbc3a)[0x55d6625f0c3a]
avrdude(+0xcafe)[0x55d6625f1afe]
avrdude(+0xc0d7)[0x55d6625f10d7]
/lib/x86_64-linux-gnu/libc.so.6(+0x35910)[0x7f8d22547910]
/lib/x86_64-linux-gnu/libc.so.6(+0x3596a)[0x7f8d2254796a]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf8)[0x7f8d225322b8]
avrdude(+0x6fda)[0x55d6625ebfda]
======= Memory map: ========
55d6625e5000-55d662653000 r-xp 00000000 103:07 6820509                   /usr/bin/avrdude
55d662853000-55d662854000 r--p 0006e000 103:07 6820509                   /usr/bin/avrdude
55d662854000-55d662855000 rw-p 0006f000 103:07 6820509                   /usr/bin/avrdude
55d662855000-55d66285c000 rw-p 00000000 00:00 0
55d662d44000-55d66306d000 rw-p 00000000 00:00 0                          [heap]
7f8d1c000000-7f8d1c021000 rw-p 00000000 00:00 0
7f8d1c021000-7f8d20000000 ---p 00000000 00:00 0
7f8d21edd000-7f8d21ef3000 r-xp 00000000 103:07 3801623                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f8d21ef3000-7f8d220f2000 ---p 00016000 103:07 3801623                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f8d220f2000-7f8d220f3000 r--p 00015000 103:07 3801623                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f8d220f3000-7f8d220f4000 rw-p 00016000 103:07 3801623                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f8d220f4000-7f8d220f7000 r-xp 00000000 103:07 3802227                   /lib/x86_64-linux-gnu/libdl-2.24.so
7f8d220f7000-7f8d222f6000 ---p 00003000 103:07 3802227                   /lib/x86_64-linux-gnu/libdl-2.24.so
7f8d222f6000-7f8d222f7000 r--p 00002000 103:07 3802227                   /lib/x86_64-linux-gnu/libdl-2.24.so
7f8d222f7000-7f8d222f8000 rw-p 00003000 103:07 3802227                   /lib/x86_64-linux-gnu/libdl-2.24.so
7f8d222f8000-7f8d22311000 r-xp 00000000 103:07 3801566                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f8d22311000-7f8d22510000 ---p 00019000 103:07 3801566                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f8d22510000-7f8d22511000 r--p 00018000 103:07 3801566                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f8d22511000-7f8d22512000 rw-p 00019000 103:07 3801566                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f8d22512000-7f8d226a7000 r-xp 00000000 103:07 3802194                   /lib/x86_64-linux-gnu/libc-2.24.so
7f8d226a7000-7f8d228a7000 ---p 00195000 103:07 3802194                   /lib/x86_64-linux-gnu/libc-2.24.so
7f8d228a7000-7f8d228ab000 r--p 00195000 103:07 3802194                   /lib/x86_64-linux-gnu/libc-2.24.so
7f8d228ab000-7f8d228ad000 rw-p 00199000 103:07 3802194                   /lib/x86_64-linux-gnu/libc-2.24.so
7f8d228ad000-7f8d228b1000 rw-p 00000000 00:00 0
7f8d228b1000-7f8d228d6000 r-xp 00000000 103:07 3801341                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f8d228d6000-7f8d22ad6000 ---p 00025000 103:07 3801341                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f8d22ad6000-7f8d22ada000 r--p 00025000 103:07 3801341                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f8d22ada000-7f8d22adb000 rw-p 00029000 103:07 3801341                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f8d22adb000-7f8d22afc000 r-xp 00000000 103:07 3801262                   /lib/x86_64-linux-gnu/libncurses.so.5.9
7f8d22afc000-7f8d22cfc000 ---p 00021000 103:07 3801262                   /lib/x86_64-linux-gnu/libncurses.so.5.9
7f8d22cfc000-7f8d22cfd000 r--p 00021000 103:07 3801262                   /lib/x86_64-linux-gnu/libncurses.so.5.9
7f8d22cfd000-7f8d22cfe000 rw-p 00022000 103:07 3801262                   /lib/x86_64-linux-gnu/libncurses.so.5.9
7f8d22cfe000-7f8d22d42000 r-xp 00000000 103:07 3801174                   /lib/x86_64-linux-gnu/libreadline.so.7.0
[... 13 lines cut ...]
7f8d2346c000-7f8d23483000 r-xp 00000000 103:07 6822009                   /usr/lib/x86_64-linux-gnu/libelf-0.168.so
7f8d23483000-7f8d23682000 ---p 00017000 103:07 6822009                   /usr/lib/x86_64-linux-gnu/libelf-0.168.so
7f8d23682000-7f8d23683000 r--p 00016000 103:07 6822009                   /usr/lib/x86_64-linux-gnu/libelf-0.168.so
7f8d23683000-7f8d23684000 rw-p 00017000 103:07 6822009                   /usr/lib/x86_64-linux-gnu/libelf-0.168.so
7f8d23684000-7f8d2368b000 r-xp 00000000 103:07 3801237                   /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
7f8d2368b000-7f8d2388a000 ---p 00007000 103:07 3801237                   /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
7f8d2388a000-7f8d2388b000 r--p 00006000 103:07 3801237                   /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
[... cut ...]
7f8d23cb8000-7f8d23cb9000 r--p 00023000 103:07 3802102                   /lib/x86_64-linux-gnu/ld-2.24.so
7f8d23cb9000-7f8d23cba000 rw-p 00024000 103:07 3802102                   /lib/x86_64-linux-gnu/ld-2.24.so
7f8d23cba000-7f8d23cbb000 rw-p 00000000 00:00 0
7ffdbfcc7000-7ffdbfce8000 rw-p 00000000 00:00 0                          [stack]
7ffdbfd6b000-7ffdbfd6d000 r--p 00000000 00:00 0                          [vvar]
7ffdbfd6d000-7ffdbfd6f000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted
$

avrdude-admin, 2021-12-10:
Kārlis Seņko <karliss>
Sat 09 Sep 2017 05:51:04 PM UTC

Here is a patch I use.

Index: avrdude/usbtiny.c
===================================================================
--- avrdude/usbtiny.c	(revision 1395)
+++ avrdude/usbtiny.c	(working copy)
@@ -455,6 +455,10 @@

for (; addr < maxaddr; addr += chunk) {
chunk = PDATA(pgm)->chunk_size;         // start with the maximum chunk size possible
+
+    if (chunk > maxaddr - addr) {
+        chunk = maxaddr - addr;
+    }

// Send the chunk of data to the USBtiny with the function we want
// to perform
@@ -510,6 +514,10 @@
for (; addr < maxaddr; addr += chunk) {
// start with the max chunk size
chunk = PDATA(pgm)->chunk_size;
+
+    if (chunk > maxaddr - addr) {
+        chunk = maxaddr - addr;
+    }

// we can only write a page at a time anyways
if (m->paged && chunk > page_size)

avrdude-admin, 2021-12-10:
The Lightning Stalker <lightningstalker>
Thu 25 Jan 2018 03:58:53 PM UTC

@karliss That patch does not fix the error on my system. At least it writes the file out before it crashes though.

avrdude-admin, 2021-12-10:
Kārlis Seņko <karliss>
Thu 25 Jan 2018 08:40:52 PM UTC

@lightningstalker Can you send the command line you used? The problem I found depends on part specific sizes and the fact I had to fix it in more than one place means I could have missed some of them or it needs to be fixed in some common code path.

avrdude-admin, 2021-12-10:
The Lightning Stalker <lightningstalker>
Fri 26 Jan 2018 10:07:10 AM UTC

Sure thing

$ avrdude -v -pm8 -cusbtiny -Uflash:r:stock.hex:i

Let me know if there's anything else.

avrdude-admin, 2021-12-10:
The Lightning Stalker <lightningstalker>
Thu 01 Feb 2018 10:34:34 PM UTC

After rebooting, the problem seems to have disappeared completely.

avrdude-admin, 2021-12-10:
Dima <dimecho>
Sat 11 Apr 2020 04:49:30 PM UTC

I have experienced the same bug and the provided patch fully fixed the issue! Thank you Kārlis Seņko.

**Will this patch be considered as part of next production?** I think it should.

My setup is USBtiny (firmware "v-usb" v1.04) running on attiny85.

$ avrdude -pt13 -cusbtiny -Uflash:w:main.hex:i

mariusgreuel, 2022-01-12:
Fixed.

closed 2022-01-12 by mariusgreuel

MCUdude, 2022-02-11:
@mariusgreuel is there a specific commit you can refer to, just as a reference?

I'm not having any issues with my USBtinyISP programmer, I just found a reference to this bug in the [ATTinyCore README](https://github.com/SpenceKonde/ATTinyCore#quick-gotcha-list---having-trouble-read-these) and was curious to see which commit actually solved the issue

dl8dtl, 2022-02-11:
```
commit 2c4f9d23b470c5ed3d18862bc8a9c95bad439125
Author: Joerg Wunsch <j@uriah.heep.sax.de>
Date:   Fri Sep 18 21:16:13 2020 +0000

    patch #9732: usbtiny_paged_load overflows buffer e.g. when reading EEPROM
    * usbtiny.c (usbtiny_paged_load, usbtiny_paged_write): ensure chunk
    does not overflow memory area
    
    Submitted by Joel Ray Holveck
    
    
    
    
    git-svn-id: svn://svn.savannah.nongnu.org/avrdude/trunk/avrdude@1444 81a1dc3b-b13d-400b-aceb-764788c761c2
```
So it's been fixed before the Github import. Supposedly, version 6.4 ought to be already fixed.