avrdudes/avrdude #432: [bug #49013] Timeout when using from linux for an arduino uno

avrdude-admin, 2021-12-10:
Calin <None>
Tue 06 Sep 2016 11:52:40 AM UTC

Hi

Really new to all of this so don't trow stones please.

I am using gort [http://gort.io](https://savannah.nongnu.orghttp://gort.io) to help me with development, behind the scene it uses arvdude and I get the following error:

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

My board is an arduino uno and I am running linux (OpenSuse thumbelweed)

I also tried to run arvdude manually:

avrdude -p m328 -c arduino -P /dev/ttyUSB6

I get the same error:

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

Any pointers on how to debug this will be greatly appreciated.

This issue was migrated from https://savannah.nongnu.org/bugs/?49013

avrdude-admin, 2021-12-10:
Alexandru Calinoiu Nicolae <calin>
Tue 06 Sep 2016 11:54:17 AM UTC

I forgot to mention that on the same system I can use arduino ido to upload to the board just fine.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 06 Sep 2016 09:17:16 PM UTC

Well, the first question here is: which AVRDUDE version are
you using, and what is the Arduino tool using?

I vaguely remember Arduino shipping some AVRDUDE that contained
a local hack before, in order to correctly trigger their reset
sequence.  I think all this has then been integrated back into
the mainline of AVRDUDE, so Arduino ought to be able to work
using a stock AVRDUDE build now.

avrdude-admin, 2021-12-10:
Alexandru Calinoiu Nicolae <calin>
Wed 07 Sep 2016 06:24:48 AM UTC

Thanks for the reply,

My version is avrdude version 6.3

Not sure what version of arvdude arduion ide is using, not even sure they use one, is there a way to find out.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 07 Sep 2016 12:19:54 PM UTC

So it's at least a recent version.

**[bug #46005](https://savannah.nongnu.org/bugs/?46005)** mentions something that looks rather similar to
your description.  I don't have an Arduino Uno myself, so
I'd have to rely to someone presenting a good fix for these
Arduino reset issues I could then incorporate.

mariusgreuel, 2022-01-12:
Current version works with Arduino Uno.

closed 2022-01-12 by mariusgreuel