avrdudes/avrdude #434: [bug #49376] Resource busy error on El Capitan

avrdude-admin, 2021-12-10:
Serjio <redddis@bk.ru>
Mon 17 Oct 2016 08:33:00 PM UTC

Tested on two laptops with El Capitan with the same error. Works smoothly on Yosemite.

avrdude -c stk500v2 -P /dev/cu.usbmodemavrdope1 -p  attiny85
avrdude: ser_open(): can't open device "/dev/cu.usbmodemavrdope1": Resource busy

This issue was migrated from https://savannah.nongnu.org/bugs/?49376

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 17 Oct 2016 09:48:39 PM UTC

Maybe the operating system now has some service (or whatever)
that is immediately grabbing a TTY device once it appears?

Alas, I'm afraid that's nothing which can be solved in
AVRDUDE, rather a kind of OS issue.  At the very least, it
requires someone with more insight into OSX.

avrdude-admin, 2021-12-10:
serjio <serjio>
Mon 17 Oct 2016 10:05:05 PM UTC

Indeed, as far as I know from Arduino-related forums (when investigating a similar issue - cdc comms are also buggy in El Capitan -  I bumped into complains that their environment/projects stopped working after upgrade to El Capitan) - it turns out that starting 10.11 Apple somehow changed the way OSX works with usb (enumerates usb?). Besides, they removed tty devices.
Do you mean to say that avrdude needs tty to work properly?
Anyway, I though there were some users of avrdude running El Capitan - wanted to learn if they face such problem and their solutions...

MCUdude, 2022-01-09:
Issue can probably be closed, since this is an OS specific issue, and has nothing to do with Avrdude really.

dl8dtl, 2022-01-09:
Just tested it on MacOS Monterey (12.0.1), works fine.
```j@airy src % ./avrdude -c stk500v2 -P /dev/cu.usbserial-1410 -p m16

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9403 (probably m16)

avrdude: safemode: Fuses OK (E:FF, H:19, L:E1)

avrdude done.  Thank you.```

closed 2022-01-09 by dl8dtl