avrdudes/avrdude #435: [bug #49387] ATXmega32E5 Flash section address wrong

avrdude-admin, 2021-12-10:
Georges Palauqui <g.palauqui@gptechinno.com>
Wed 19 Oct 2016 08:14:01 PM UTC
Device type: x32e5

Current avrdude.conf.in gives for ATxmega32e5 :
memory "apptable"
size		= 0x1000;
offset		= 0x00807000;
page_size	= 0x80;
readsize	= 0x100;
;
memory "boot"
size		= 0x1000;
offset		= 0x00808000;
page_size	= 0x80;
readsize	= 0x100;
;
But according to datasheet ([http://www.atmel.com/Images/Atmel-8153-8-and-16-bit-AVR-Microcontroller-XMEGA-E-ATxmega8E5-ATxmega16E5-ATxmega32E5_Datasheet.pdf](https://savannah.nongnu.orghttp://www.atmel.com/Images/Atmel-8153-8-and-16-bit-AVR-Microcontroller-XMEGA-E-ATxmega8E5-ATxmega16E5-ATxmega32E5_Datasheet.pdf) page 12 figure 8.1) it should be :
memory "apptable"
size		= 0x800;
offset		= 0x00803800;
page_size	= 0x80;
readsize	= 0x80;
;

memory "boot"
size		= 0x800;
offset		= 0x00804000;
page_size	= 0x80;
readsize	= 0x80;
;
Both Size and Offset are wrong.

This issue was migrated from https://savannah.nongnu.org/bugs/?49387

mcuee, 2022-05-29:
It seems to me there are quite a bit of xmega related issues. Not able to help much as I do not have an xmega board. But I am not so sure if all of them are still relevant or not.

Ref: xmega relared issues
* https://github.com/avrdudes/avrdude/issues/164
* https://github.com/avrdudes/avrdude/issues/190
* https://github.com/avrdudes/avrdude/issues/200
* https://github.com/avrdudes/avrdude/issues/243
* https://github.com/avrdudes/avrdude/issues/289
* https://github.com/avrdudes/avrdude/issues/311
* https://github.com/avrdudes/avrdude/issues/314
* https://github.com/avrdudes/avrdude/issues/365
* https://github.com/avrdudes/avrdude/issues/435

MCUdude, 2022-06-19:
For reference, I've attached the `ATxmega32E5.atdf` file, which is used by Microchip Studio when working with the 32E5. I can only assume that this file has the correct content. All these files can be downloaded from [Microchip Packs Repository](https://packs.download.microchip.com).

[ATxmega32E5.atdf.zip](https://github.com/avrdudes/avrdude/files/8934983/ATxmega32E5.atdf.zip)

From ATxmega32E5.atdf:

```xml
<memory-segment start="0x07000" size="0x1000" type="flash" rw="RW" exec="1" name="APPTABLE_SECTION" pagesize="128"/>

<property name="APPTABLE_SECTION_OFFSET" value="0x00807000"/>

<memory-segment start="0x08000" size="0x1000" type="flash" rw="RW" exec="1" name="BOOT_SECTION" pagesize="128"/>

<property name="BOOT_SECTION_OFFSET" value="0x00808000"/>
```

I'm closing this issue because OP may have read from a datasheet that had the memory addresses incorrectly defined. If this was a real problem, the latest ATxmega32E5.adtf would reflect OP's suggested changes.

closed 2022-06-19 by MCUdude