avrdudes/avrdude #436: [bug #49450] Does not write integer to /sys/class/gpio/export

avrdude-admin, 2021-12-10:
Maurik <holtrop@physics.unh.edu>
Wed 26 Oct 2016 04:59:27 PM UTC
Programmer hardware: Beagle Bone
Device type: linuxgpio

In the code linuxgpio.c I believe the format string for the GPIO port is incorrectly specified as %ud, instead of %u. This causes a "d" to be appended to the port and some implementations of /sys/class/gpio/export do not allow for this.

Changing the %ud to %u allows me to run avrdude on the Beagle Bone Black.

This issue was migrated from https://savannah.nongnu.org/bugs/?49450

mcuee, 2022-05-24:
I believe this can be closed as it has been fixed, no more `%ud` in the current codes.
Ref: https://github.com/avrdudes/avrdude/blob/main/src/linuxgpio.c

mcuee, 2022-05-27:
@MCUdude Maybe you can close this one.

MCUdude, 2022-05-27:
Thanks for helping out with some "house keeping" @mcuee!

closed 2022-05-27 by MCUdude

mcuee, 2022-05-29:
> Thanks for helping out with some "house keeping" @mcuee!

Thanks for closing the old issues. I tend to think it is good to reduce the issues to a more managable number like <100. :-)

MCUdude, 2022-05-29:
> I tend to think it is good to reduce the issues to a more manageable number like <100. :-)

Less than 100 issues is definitely a goal. I'll help out closing issues as long as we're confident they've either been resolved, are not relevant anymore, or where OP never responded.