avrdudes/avrdude #438: [bug #49740] Bad EEPROM write on Linux 64

avrdude-admin, 2021-12-10:
Rada Berar <None>
Wed 30 Nov 2016 08:18:06 AM UTC
Device type: USBtiny

I am using USBTiny programmer with AtTiny2313. When programming whether from Eclipse or Therminal on XUbuntu 16.04, 64-bit, I get error:
....
avrdude: verifying ...
avrdude: 2042 bytes of flash verified
avrdude: reading input file "CDC.eep"
avrdude: input file CDC.eep auto detected as Intel Hex
avrdude: writing eeprom (128 bytes):

Writing | ################################################## | 100% 21.47s

avrdude: 128 bytes of eeprom written
avrdude: verifying eeprom memory against CDC.eep:
avrdude: load data eeprom data from input file CDC.eep:
avrdude: input file CDC.eep auto detected as Intel Hex
avrdude: input file CDC.eep contains 128 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 1.94s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
0x00 != 0x43
avrdude: verification error; content mismatch

When I try to read fuses or device ID, Eclipse repports it can not understand avrdude output.

By using the chip, I verified that the EEPROM was written all 0x01 except for last three bytes which were fine.

I tried the same code and the same operations on Windows7 64-bit and everything works fine. The EEPROM gets written OK, Eclipse reads avrdude output OK.

I tried the latest avrdude6.3 and the result is the same.

This issue was migrated from https://savannah.nongnu.org/bugs/?49740

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Wed 30 Nov 2016 09:36:16 AM UTC

> I tried the same code and the same operations on Windows7
> 64-bit and everything works fine.

Same AVRDUDE version, too?

avrdude-admin, 2021-12-11:
Rada Berar <ujagaga>
Wed 30 Nov 2016 06:26:09 PM UTC

Probably not. I installed WinAvr a couple of days ago but it seems to be old. I do not know how to check version. I will try to find the version tomorrow.

avrdude-admin, 2021-12-11:
Rada Berar <ujagaga>
Fri 02 Dec 2016 10:45:08 AM UTC

Windows AVRDUDE version is 5.10

avrdude-admin, 2021-12-11:
Rada Berar <ujagaga>
Sat 03 Dec 2016 09:34:44 AM UTC

I uninstalled the latest AVRDUDE and installed AVRDUDE version 5.11. This one seems to work well.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sat 03 Dec 2016 11:01:53 AM UTC

That's what I excpected.

So it's an issue with recent AVRDUDE versions.

MCUdude, 2022-01-10:
> That's what I excpected.
So it's an issue with recent AVRDUDE versions.

I'm not able to reproduce this. Have you already patched this @dl8dtl?

```
$ cat ee3.hex
:20000000555555555555555555555555555555555555555555555555555555555555555540
:20002000555555555555555555555555555555555555555555555555555555555555555520
:20004000555555555555555555555555555555555555555555555555555555555555555500
:200060005555555555555555555555555555555555555555555555555555555555555555e0
:00000001FF

$ cat ee4.hex
:200000000000000000000000000000000000000000000000000000000000000000000000e0
:200020000000000000000000000000000000000000000000000000000000000000000000c0
:200040000000000000000000000000000000000000000000000000000000000000000000a0
:20006000000000000000000000000000000000000000000000000000000000000000000080
:00000001FF

$ ./avrdude -C avrdude.conf -pattiny2313 -c usbtiny -Ueeprom:w:ee3.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e910a (probably t2313)
avrdude: reading input file "ee3.hex"
avrdude: writing eeprom (128 bytes):

Writing | ################################################## | 100% 0.80s

avrdude: 128 bytes of eeprom written
avrdude: verifying eeprom memory against ee3.hex:
avrdude: load data eeprom data from input file ee3.hex:
avrdude: input file ee3.hex contains 128 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.14s

avrdude: verifying ...
avrdude: 128 bytes of eeprom verified

avrdude: safemode: Fuses OK (E:FF, H:9F, L:62)

avrdude done.  Thank you.

$ ./avrdude -C avrdude.conf -pattiny2313 -c usbtiny -Ueeprom:w:ee4.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e910a (probably t2313)
avrdude: reading input file "ee4.hex"
avrdude: writing eeprom (128 bytes):

Writing | ################################################## | 100% 0.79s

avrdude: 128 bytes of eeprom written
avrdude: verifying eeprom memory against ee4.hex:
avrdude: load data eeprom data from input file ee4.hex:
avrdude: input file ee4.hex contains 128 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.14s

avrdude: verifying ...
avrdude: 128 bytes of eeprom verified

avrdude: safemode: Fuses OK (E:FF, H:9F, L:62)

avrdude done.  Thank you.
```

dl8dtl, 2022-03-29:
> Have you already patched this @dl8dtl?

Not sure, will re-check myself. If it works now, it's probably safe to assume it's fixed.

dl8dtl, 2022-03-30:
Well, I didn't have a '2313 around but a '4313. Technically, they ought to behave similarly though.

Just tried writing the top-level `INSTALL` file into EEPROM:
```
$ ./build_freebsd/src/avrdude -c usbtiny -p attiny4313 -U ee:w:INSTALL:r

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e920d (probably t4313)
avrdude: reading input file "INSTALL"
avrdude: writing eeprom (68 bytes):

Writing | ################################################## | 100% 0.44s

avrdude: 68 bytes of eeprom written
avrdude: verifying eeprom memory against INSTALL:

Reading | ################################################## | 100% 0.08s

avrdude: 68 bytes of eeprom verified

avrdude done.  Thank you.

$ ./build_freebsd/src/avrdude -c usbtiny -p attiny4313 -t

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e920d (probably t4313)
avrdude> d ee
>>> d ee 
0000  49 6e 73 74 61 6c 6c 61  74 69 6f 6e 20 49 6e 73  |Installation Ins|
0010  74 72 75 63 74 69 6f 6e  73 0a 2a 2a 2a 2a 2a 2a  |tructions ******|
0020  2a 2a 2a 2a 2a 2a 2a 2a  2a 2a 2a 2a 2a 2a 2a 2a  |****************|
0030  2a 2a 2a 0a 0a 53 65 65  20 52 45 41 44 4d 45 2e  |***  See README.|
0040  6d 64 2e 0a ff ff ff ff  ff ff ff ff ff ff ff ff  |md. ............|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q
>>> q 

avrdude done.  Thank you.
```

closed 2022-03-30 by dl8dtl