avrdudes/avrdude #71: [bug #17487] AVR RC oscillator calibration routine not supported (feature request)

avrdude-admin, 2021-12-10:
John Voltz <ninevoltz@metalink.net>
Mon 21 Aug 2006 12:05:01 PM UTC

This is a feature request rather than a bug.

AVRDUDE currently does not support RC oscillator calibration as outlined in the AVR053 appnote.

[file #10564: RC calibration.pdf](https://savannah.nongnu.org/bugs/download.php?file_id=10564)
[file #10757: avrdudeupdate.tar.gz](https://savannah.nongnu.org/bugs/download.php?file_id=10757)
[file #12867: arbitrary_rw.patch](https://savannah.nongnu.org/bugs/download.php?file_id=12867)
[file #12868: mistake.patch](https://savannah.nongnu.org/bugs/download.php?file_id=12868)
[file #12869: fixed_arbitrary_rw.patch](https://savannah.nongnu.org/bugs/download.php?file_id=12869)

This issue was migrated from https://savannah.nongnu.org/bugs/?17487

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 18 Sep 2006 09:51:29 PM UTC

I promised to handle that before releasing 5.2.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 09 Oct 2006 02:38:41 PM UTC

The part of the patch implementing the -O option has been
committed.

I still feel we should seek for a more general method to
transfer the result to an arbitrary location.

avrdude-admin, 2021-12-10:
John Voltz <ninevoltz9>
Sat 26 May 2007 06:46:24 PM UTC

I just submitted a patch to allow avrdude to read and write at arbitrary memory locations. You simply specify a block length (in bytes) followed by the byte offset in the hex file then finally the byte offset in flash or eeprom.

Here's a command line example:

avrdude -cstk500v2 -B4 -pm16 -P/dev/ttyS0 -v -e -Uflash:w:main.hex:i,10,0,6

this will write 10 bytes from byte offset 0 in main.hex to byte offset 6 in flash

values can be in decimal or hex, and all are optional (will default to zero). Values must be multiples of two since the AVR memory is arranged in 16-bit words.

avrdude-admin, 2021-12-10:
John Voltz <ninevoltz9>
Sat 26 May 2007 07:08:28 PM UTC

I forgot to mention how my last patch is useful. It can be used for RC oscillator calibration by first reading the AVRs EEPROM to a temporary file, then you can write it back to the AVR in an arbitrary location in the flash or EEPROM.

It can also be used to store configuration constants in flash, which is my latest interest since I have been having a lot of trouble lately with EEPROM corruption, and yes, I am using the brown-out detector in my application! :oP

avrdude-admin, 2021-12-10:
John Voltz <ninevoltz9>
Sat 26 May 2007 07:49:33 PM UTC

Not long after submitting my patch I discovered a nasty mistake in my code. Here's **[patch #3](https://savannah.nongnu.org/patch/?3)**

(**[file #12868](https://savannah.nongnu.org/support/download.php?file_id=12868)**)

avrdude-admin, 2021-12-10:
Charles Goyard <cgo>
Fri 28 Aug 2009 07:51:09 PM UTC

Hi,
the documentation states JTAG ICE mkII is supported for calibration, but it is not.
According to the AVR067, the programmer has to issue a  CMND_ISP_PACKET containing the ISP CMD_OSCCAL.
A quick fix is to change the manual page. A nice thing would be to add the support.
Regards,
Charles

mcuee, 2022-11-06:
Since `-O` has been implemented, the enhancement porting has been done. So the only remaining issue is with JTAGICE mkii calibration, I will remove the `enhancement` label now.

mcuee, 2023-06-03:
Given that this issue has been there for so long and JTAG ICE mkII has been obsolete for quite some time, I am going to cloe this one. We can always re-open if PR is raised.

closed 2023-06-03 by mcuee

mcuee, 2023-10-18:
> John Voltz Sat 26 May 2007 07:49:33 PM UTC
> 
> Not long after submitting my patch I discovered a nasty mistake in my code. Here's **[patch #3](https://savannah.nongnu.org/patch/?3)**
> 
> (**[file #12868](https://savannah.nongnu.org/support/download.php?file_id=12868)**)

The contents of the patch is attached here for easier reference.
```
Only in avrdude-5.4: ac_cfg.h
Only in avrdude-5.4-orig: arbitrary_rw.patch
Only in avrdude-5.4: avrdude.conf
Only in avrdude-5.4: avrdude.conf.tmp
Only in avrdude-5.4: avrdude.kdevelop
Only in avrdude-5.4: avrdude.kdevelop.pcs
Only in avrdude-5.4: avrdude.kdevses
Only in avrdude-5.4: config.log
Only in avrdude-5.4: config.status
Only in avrdude-5.4: debug
Only in avrdude-5.4: .deps
Only in avrdude-5.4-orig/doc: avrdude.info
Only in avrdude-5.4/doc: Makefile
Only in avrdude-5.4-orig/doc: stamp-vti
Only in avrdude-5.4-orig/doc: version.texi
Only in avrdude-5.4: Doxyfile
Only in avrdude-5.4: main.hex
Only in avrdude-5.4: Makefile
Only in avrdude-5.4: stamp-h1
diff -rup avrdude-5.4-orig/update.c avrdude-5.4/update.c
--- avrdude-5.4-orig/update.c	2007-05-26 15:49:50.000000000 -0400
+++ avrdude-5.4/update.c	2007-05-26 15:47:32.000000000 -0400
@@ -25,7 +25,6 @@
 #include <errno.h>
 #include <string.h>
 #include <time.h>
-
 #include "avrdude.h"
 #include "avr.h"
 #include "config.h"
@@ -36,7 +35,7 @@
 UPDATE * parse_op(char * s)
 {
   char buf[1024];
-  char * p, * cp, c, ** end;
+  char * p, * cp, c, * end;
   UPDATE * upd;
   int i;
   size_t fnlen;
@@ -166,7 +165,7 @@ UPDATE * parse_op(char * s)
       upd->moffset = 0;
     } else {
       p++;
-      upd->length = (int) strtol(p, end, 0);
+      upd->length = strtol(p, &end, 0);
       if ((upd->length % 2) == 1) {
         fprintf(stderr, "\n%s: length specifier must be a multiple of two, you specified %d\n",
                 progname, upd->length);
@@ -187,7 +186,7 @@ UPDATE * parse_op(char * s)
       upd->moffset = 0;
     } else {
       p++;
-      upd->foffset = (int) strtol(p, end, 0);
+      upd->foffset = strtol(p, &end, 0);
       if ((upd->foffset % 2) == 1) {
         fprintf(stderr, "\n%s: file offset specifier must be a multiple of two, you specified %d\n",
                 progname, upd->foffset);
@@ -208,7 +207,7 @@ UPDATE * parse_op(char * s)
       upd->moffset = 0;
     } else {
       p++;
-      upd->moffset = (int) strtol(p, end, 0);
+      upd->moffset = strtol(p, &end, 0);
       if ((upd->moffset % 2) == 1) {
         fprintf(stderr, "\n%s: memory offset specifier must be a multiple of two, you specified %d\n",
                 progname, upd->moffset);
Only in avrdude-5.4/windows: .deps
Only in avrdude-5.4/windows: Makefile

```