avrdudes/avrdude #97: [bug #19882] Many "error in USB receive" messages received before successful programming.

avrdude-admin, 2021-12-10:
mellis <None>
Wed 16 May 2007 03:10:57 PM UTC

I'm having problems using avrdude with an AVRISPmkII to upload a .hex file to an ATmega168 (on an Arduino board with 16 MHz external oscillator).  I'm using the latest versions of libusb and avrdude from CVS on an Intel Mac running Mac OS X 10.4.9 (Tiger).  The command line is: avrdude -p m168 -c avrispmkII -P usb -U flash:w:ATmegaBOOT_168.hex

I receive multiple "avrdude: stk500v2_recv_mk2: error in USB receive" messages every 15 seconds or so over the course of a few minutes.  Then, the chip is programmed successfully, and at a seemingly "normal" speed.

If I add a -vvvv to my command line, I get the following output:

avrdude: Version 5.3cvs, compiled on May 16 2007 at 09:42:00
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)

System wide configuration file is "/usr/local/etc/avrdude.conf"
User configuration file is "/Users/dmellis/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port            : usb
Using Programmer      : avrispmkII
avrdude: usbdev_open(): Found AVRISP mkII, serno: 0000A0019837
avrdude: Sent: . [01]
avrdude: usbdev_recv_frame(): usb_bulk_read(): usb_bulk_read: An error occured during read (see messages above)
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: Sent: . [01]
avrdude: usbdev_recv_frame(): usb_bulk_read(): usb_bulk_read: An error occured during read (see messages above)
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: Sent: . [01]
avrdude: usbdev_recv_frame(): usb_bulk_read(): usb_bulk_read: An error occured during read (see messages above)
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: Sent: . [01]
avrdude: usbdev_recv_frame(): usb_bulk_read(): usb_bulk_read: An error occured during read (see messages above)
avrdude: stk500v2_recv_mk2: error in USB receive
...
avrdude: Sent: . [01]
avrdude: usbdev_recv_frame(): usb_bulk_read(): usb_bulk_read: An error occured during read (see messages above)
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_getsync(): timeout communicating with programmer

followed by the standard programming output (information about the part, etc).  I'm attaching the full output.

Any idea what's wrong?  I'm happy to spend some time investigating, but I don't know where to start.

[file #12782: avrdude-trace.txt](https://savannah.nongnu.org/bugs/download.php?file_id=12782)

This issue was migrated from https://savannah.nongnu.org/bugs/?19882

avrdude-admin, 2021-12-10:
David A. Mellis <mellis>
Thu 14 Jun 2007 05:37:41 PM UTC

It seems that my AVRISP mkII is not responding to the sync command (CMD_SIGN_ON aka the byte 1), even with the latest firmware.  The sync is retried 33 times, each time with a 10 second timeout, leading to the long delay I noticed.

I noticed the following lines in the stk500v2_getsync() function in stk500v2.c (lines 483-484):

if (pgmtype == PGMTYPE_JTAGICE_MKII)
return 0;

Is that to avoid a similar problem with the JTAGICE mkII?  Would it make sense to do the same for the AVRISP mkII, or am I the only one having these problems?  If the latter, anyone have any ideas what might be wrong?

avrdude-admin, 2021-12-10:
David A. Mellis <mellis>
Wed 25 Jul 2007 03:33:15 PM UTC

I have the same problem under Windows with the avrdude and libusb from the latest version of WinAVR (20070525).  I get 31 errors:
avrdude.exe: stk500v2_recv_mk2: error in USB receive
and then the flash is written successfully.  Has anyone used an AVRISP mkII with avrdude without getting these errors?  Can avrdude be patched to return from stk500v2_getsync() for the avrispmkii as well as the jtagice_mkii?

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 29 Oct 2007 09:59:34 PM UTC

I just tried uploading something with an AVRISP mkII, and
cannot see this.  Can you verify the problem still exists?

(Quite possible it has been fixed some time ago.)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 29 Oct 2007 10:54:25 PM UTC

See **[patch #5655](https://savannah.nongnu.org/patch/?5655)** which apparently addresses the same issue.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 26 Aug 2011 12:19:25 PM UTC

Is this still an issue at all?

MCUdude, 2022-01-10:
I'm closing this issue because there are currently no known issues related to the latest macOS version version of Avrdude and the AVRISPmkII programmer.

closed 2022-01-10 by MCUdude