avrdudes/avrdude #440: [bug #49898] Config file error for many parts

avrdude-admin, 2021-12-11:
Owen <None>
Wed 21 Dec 2016 03:11:01 AM UTC
Programmer hardware: USBASP
Device type: m328

Verify error setting efuse to 0x05.

It seems that the config file contains:
memory "efuse"
size = 1;
min_write_delay = 4500;
max_write_delay = 4500;
read = "0 1 0 1 0 0 0 0 0 0 0 0 1 0 0 0",
"x x x x x x x x o o o o o o o o";

write = "1 0 1 0 1 1 0 0 1 0 1 0 0 1 0 0",
"x x x x x x x x x x x x x i i i";
;

Which I think causes it to write 0x05, and read back 8 bits incl unimplemented bits as report as 0xfd which of course does not verify.

This looks like a global change that has messed up lots of the config file, do a diff with v6.1 and you will not what has happened.

Meanwhile, 6.1 looks the best version out there.

Owen

This issue was migrated from https://savannah.nongnu.org/bugs/?49898

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Wed 21 Dec 2016 09:27:50 AM UTC

This seems to be a duplicate of **[bug #22248](https://savannah.nongnu.org/bugs/?22248)** to me.

avrdude-admin, 2021-12-11:
Owen Duffy <owenduffy>
Wed 21 Dec 2016 01:04:46 PM UTC

I read most of the long discussion.

For as long as I can remember, the Atmel tools have required the user to specify ONLY the implemented bits for the example case I gave, three bits. They do not state in the datasheet that the user must specify unimplemented bits at 1, but they do hint that they will read back as one (because being unimplemented, the should always be 'unprogrammed'.

In Atmel's tools if I select BOD=2.7 from the feature selection, it calculates and shows efuse=0x05, and if programmed, that WILL verify.

That is all rational.

It seems that recently, AVRDUDE's direction is to mask out ONLY three bits from user supplied efuse parm (for this part), but include 8 bits read back from the part for verification. Why verify bits that the user did not specify.

AVRDUDE has changed to strike out on its own, it requires a different efuse specification than the Atmel tools display... not for programming the fuse correctly, but purely for verification.

Most tools follow Atmel's lead, eg [http://www.engbedded.com/fusecalc/](https://savannah.nongnu.orghttp://www.engbedded.com/fusecalc/) .

I think you have made a mistake, the solution for users who want consistency with Atmel and compatibility with all the existing scripts, makefiles and project documentation seems to be v6.1.

MCUdude, 2022-03-25:
Similar issue to #461. I'll quote the answer I gave to #461, which is a similar issue

> This is currently a non-issue. Unused bits should be set to 1, but Avrdude does it for you anyways, and give you a warning.
Here's the output when I try to write 0x05 as efuse:

```
$ ./avrdude -C avrdude.conf -c usbasp -p atmega328p -Uefuse:w:0x05:m

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "0x05"
avrdude: writing efuse (1 bytes):

Writing |                                                    | 0% 0.00s ***failed;  
Writing | ################################################## | 100% 0.03s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0x05:

Reading | ################################################## | 100% 0.00s

avrdude: WARNING: ignoring mismatch in unused bits of "efuse"
         (0xfd != 0x05). To prevent this warning set unused bits
         to 1 when writing (double check with your datasheet first).
avrdude: 1 bytes of efuse verified

avrdude done.  Thank you.
```

closed 2022-03-25 by MCUdude