avrdudes/avrdude #441: [bug #50087] option parse error: after -U flash:w:program_to_upload.hex, subsequent -V ignored

avrdude-admin, 2021-12-11:
Britton Kerin <britton.kerin@gmail.com>
Thu 19 Jan 2017 09:08:55 AM UTC
Programmer hardware: arduino
Device type: atmega328P

Moving the -V in front of -U causes the read-back verification to be skipped as expected.

I guess it's probably all options and not just -V.

I wildly guess that it's somehow related to the not-present optional 4th sub-argument to the -U option (the format argument).

This issue was migrated from https://savannah.nongnu.org/bugs/?50087

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Thu 19 Jan 2017 10:38:01 AM UTC

The problem is that the -U option parser already evaluates
the "verify" flag (see the "HERE!" comment):

```
case 'U':
upd = parse_op(optarg);
if (upd == NULL) {
avrdude_message(MSG_INFO, "%s: error parsing update operation '%s'\n",
progname, optarg);
exit(1);
}
ladd(updates, upd);

if (/* HERE! */ verify && upd->op == DEVICE_WRITE) {
upd = dup_update(upd);
upd->op = DEVICE_VERIFY;
ladd(updates, upd);
}
break;

```
Thus, if the -V option has not been present at this point,
it's being ignored ("verify" defaults to 1).

Unfortunately, there's no easy fix for that.

mcuee, 2022-06-08:
@dl8dtl In this case, is it possible just to come out with a document fix about the sequence of `-U` and `-V`.

dl8dtl, 2022-06-08:
> @dl8dtl In this case, is it possible just to come out with a document fix about the sequence of `-U` and `-V`?

Makes sense …

stefanrueger, 2022-08-04:
Fixed by PR #1053

closed 2022-08-04 by stefanrueger