avrdudes/avrdude #448: [bug #50480] Add support of at*****A devices

avrdude-admin, 2021-12-11:
sigmaN <None>
Tue 07 Mar 2017 02:22:08 PM UTC
Programmer hardware: all
Device type: atmega8A

There is no support for A devices like atmega8A attiny13A...etc

Yes I know about -F option, but why we can't just add signatures for *A devices and be happy without -F??

This issue was migrated from https://savannah.nongnu.org/bugs/?50480

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Tue 07 Mar 2017 02:27:15 PM UTC

Please, make a survey which devices actually need a different
ID at all.  I think there are a number of A devices which just
use the same ID as their non-A counterparts.

Of course, there's nothing wrong with also adding those as
aliases as well.  With the version 6.x config file format,
adding aliases is pretty simple.

avrdude-admin, 2021-12-11:
Vasily Trotskyi <sigman>
Tue 07 Mar 2017 02:43:39 PM UTC

Yes, you are right  IDs are equal. Just checked datasheets...
Hm. So we only need -p parts with a and it will be done. I mean it will be good to have out of the box -p t13a or -p m8a without manual aliasing.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Tue 07 Mar 2017 03:00:41 PM UTC

I think there are actually a few devices where Atmel indeed
changed the IDs for the A devices.  They weren't very consistent
in that, alas.

That's why I've been asking for someone to take a small survey,
and add the result to this bug report (or post it to the
mailinglist).  The survey could be based on the datasheets,
or alternatively, on the Atmel Studio XML files.

avrdude-admin, 2021-12-11:
Vasily Trotskyi <sigman>
Tue 07 Mar 2017 08:46:05 PM UTC

I combined files from two Atmel packs
/Program Files (x86)/Atmel/Studio/7.0/packs/atmel/ATmega_DFP/1.1.130
and /Program Files (x86)/Atmel/Studio/7.0/packs/atmel/ATtiny_DFP/1.1.102

Then run script to see deference between A and not A signatures
#!/bin/bash
#Files with A in the end of name
AFILES=`ls ./atdf/ | grep .*A.atdf`
for afile in $AFILES
do
#remove A from end of file name and check if file exists
notafile=${afile//A.atdf/.atdf}
if [ -f ./atdf/$notafile ]; then
sign=`cat ./atdf/$notafile | grep SIGNATURE[012]`
asign=`cat ./atdf/$afile | grep SIGNATURE[012]`
#compare signatures of A and not A version
if [ "$sign" != "$asign" ]; then
echo "$afile"
echo "$asign"
echo "$notafile"
echo "$sign"
fi
fi
done

output:
ATmega324PA.atdf
<property name="SIGNATURE0" value="0x1e"/>
<property name="SIGNATURE1" value="0x95"/>
<property name="SIGNATURE2" value="0x11"/>
ATmega324P.atdf
<property name="SIGNATURE0" value="0x1e"/>
<property name="SIGNATURE1" value="0x95"/>
<property name="SIGNATURE2" value="0x08"/>

so only one MCU signatures is not equal for A and non-A versions.

avrdude-admin, 2021-12-11:
Joerg Wunsch <joerg_wunsch>
Sun 07 Nov 2021 04:46:51 PM UTC

OK, both ATmega324P and ATmega324PA are already in avrdude.conf.in.

I just added ATmega8A as an alias for ATmega8.

Which further A devices are still missing now?

MCUdude, 2022-02-11:
Fixed in #832