avrdudes/avrdude #454: [bug #51117] Problems with extended address (>128K) and buffer size

avrdude-admin, 2021-12-11:
Stephan Schreiber <None>
Thu 25 May 2017 11:16:43 PM UTC
Programmer hardware: AVROSP (Butterfly)
Device type: ATXMega256A3(U)

Using a black-box (client-provided) which is connected via Ft232R USB adapter, most likely ancient boot loader, following AVROSP protocol (and working correctly with avrosp.exe).
My job here is to port a Windows Firmware Updater (using avrosp.exe) to Mac OS X.

ATXMega256A3 (might be -U, can't find out without disassembling the unit which I am clearly not supposed to do).

Due to NDA I cannot disclose any further details, but I can provide a partial serial log if you're interested.
I've analyzed serial communications and that's how I finally found out what is going wrong there.

For some reason, buffer size is reported as 0x0080 (128 bytes) and the update process fails, as 512 bytes (page size correctly given in avrdude.conf). are expected.

The even more serious problem, however, is that extended addressing is not being used by avrdude in this case.
"A" 0xff00" becomes "A" 0x0000 instead of "H" 0x010000", thus overwriting flash memory from the beginning instead of appending the code properly.

I have applied this (rough) patch, which makes it work; just wanted to let you know.
I probably can't use avrdude anyway, because of the restrictions GPL would impose.

Anyway, here's the patch:

--- butterfly.c.orig	2014-07-16 22:14:58.000000000 +0200
+++ butterfly.c	2017-05-23 22:49:20.000000000 +0200
@@ -321,6 +321,10 @@
PDATA(pgm)->buffersize = (unsigned int)(unsigned char)c<<8;
butterfly_recv(pgm, &c, 1);
PDATA(pgm)->buffersize += (unsigned int)(unsigned char)c;
+
+  // overriding it for now
+  PDATA(pgm)->buffersize = 512;
+
avrdude_message(MSG_INFO, "Programmer supports buffered memory access with buffersize=%i bytes.\n",
PDATA(pgm)->buffersize);

@@ -424,6 +428,7 @@

static void butterfly_set_addr(PROGRAMMER * pgm, unsigned long addr)
{
+if( addr < 0x10000 ) {
char cmd[3];

cmd[0] = 'A';
@@ -432,6 +437,19 @@

butterfly_send(pgm, cmd, sizeof(cmd));
butterfly_vfy_cmd_sent(pgm, "set addr");
+
+  } else {
+
+  char cmd[4];
+
+  cmd[0] = 'H';
+  cmd[1] = (addr >> 16) & 0xff;
+  cmd[2] = (addr >> 8) & 0xff;
+  cmd[3] = addr & 0xff;
+
+  butterfly_send(pgm, cmd, sizeof(cmd));
+  butterfly_vfy_cmd_sent(pgm, "set extaddr");
+  }
}

Best regards,
Stephan

This issue was migrated from https://savannah.nongnu.org/bugs/?51117

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Thu 15 Jun 2017 09:07:03 PM UTC

Any chance you have or could create a patch for version 6.1? I tried to applyu this patch but get a malformed patch at line 13. I assume due to differing butterfly.c files betwen version 6.1 and 6.3.

avrdude-admin, 2021-12-11:
Stephan Schreiber <tozzi>
Thu 15 Jun 2017 10:24:46 PM UTC

I'd love to help, but I'm actually still quite busy reinventing the wheel and porting the whole thing to Swift, using ORSSerial Library.
Can't you switch to 6.3? OTOH, this patch would be quite easy to apply manually.
Just look for the corresponding section in butterfly.c (butterfly_set_addr).

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Fri 16 Jun 2017 01:31:54 AM UTC

I totally understand. I made the changes manually. Reason I am not using 6.3 is because I ran into issues using GPIO on my raspberry pi with 6.3. Could never get it work but 6.1 would work fine. Thanks for the patch, I will report back if it is successful in 6.1 (running into other issues with my raspberry pi preventing me from testing it at the moment).

avrdude-admin, 2021-12-11:
Stephan Schreiber <tozzi>
Fri 16 Jun 2017 01:37:18 AM UTC

Thanks!
I hope the maintainer of this project returns and takes care of our issues.
Do let me know if the patch worked for you! Fingers crossed.

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Fri 16 Jun 2017 03:28:40 AM UTC

So I got avrdude reinstalled on my raspberry pi and working the way it was before...which was the verification was failing on a flash where file is 152kb.

I made changes to the butterfly.c file but am not sure I am taking the proper steps afterwards. The steps I did take were to:

make
sudo make install

My thought process was that this would recompile avrdude with my changes to butterfly.c. Am I wrong on this? Apologies as this is a new realm for me and I am learning as I go.

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Fri 16 Jun 2017 03:34:35 AM UTC

And as I am looking at all of this, please excuse my ignorance, would changes to the butterfly.c file have any affect on my circumstance where I am attempting to use the "linuxgpio" programmer?

mcuee, 2022-06-11:
This is probably the same as #360.
* #360

MCUdude, 2022-08-05:
Is there a place I can get a working avr109/911 bootloader so I can test the provided patch which a 128kiB and a 256kiB AVR?

For some reason, related issues are all xmega-related, but since this is about the 109/911 protocol, I doubt the actual chip matters much, except for their page size and total flash memory size.

mcuee, 2022-08-05:
> Is there a place I can get a working avr109/911 bootloader so I can test the provided patch which a 128kiB and a 256kiB AVR?
>
> For some reason, related issues are all xmega-related, but since this is [...]

@MCUdude 
You can refer to the comments here.
* https://www.microchip.com/en-us/application-notes/an1644

And here.
https://github.com/avrdudes/avrdude/discussions/1008#discussioncomment-3007290

@dl8dtl is saying that this is a protocol extension for AVR109 bootloader.

You can download the official AVR109 examples here and see if you can extend it for 128KB/256KB flash parts.
https://www.microchip.com/en-us/application-notes/an1644

mcuee, 2022-08-06:
I have changed the label to `enhancement` since it is very much similar to #360 (avr109/avr911 bootloader for xmega).

mcuee, 2022-08-06:
@MCUdude 
I was thinking xboot was specific to xmega AVRs but then I find out that it also supports normal mega AVRs. So you may want to give it a try. I am not so sure if the current avr109 implementation in avrdude supports xboot well enough or not.
https://github.com/alexforencich/xboot

There is an enhancement request for xboot here.
* https://github.com/avrdudes/avrdude/issues/200 (for xmega)

There is another avr109 compatible bootloader for ATmega1284P here. You may also want to take a look.
https://github.com/ppeluffo/avr109

mcuee, 2022-08-07:
@MCUdude 
As per the following avrfreaks thread, you can get avr109 to work with ATmega2561 or similar parts.
https://www.avrfreaks.net/forum/avr-prog-and-new-mega-family-devices

If avrdude does not work, you can try AVROSP2.
https://xdevs.com/doc/xDevs.com/M16/AVROSP2/

mcuee, 2022-08-07:
This is what I get with the current git with my quick mod of the avr109 for ATmega2560 (tested on Arduino Mega CH340 clone).

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avr109 -P COM6 -b 115200 -p m2560  -v

avrdude.exe: Version 7.0-20220804 (5f5002e)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : avr109
             Overriding Baud Rate          : 115200
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : butterfly
             Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.5; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:
    Device code: 0x2c

avrdude.exe: devcode selected: 0x2c
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c avr109 -P COM6 -b 115200 -p m2560 
 -D -U .\hex\Blink.ino.mega.hex

Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.5; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:
    Device code: 0x2c

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: input file .\hex\Blink.ino.mega.hex auto detected as Intel Hex
avrdude.exe: reading input file .\hex\Blink.ino.mega.hex for flash
avrdude.exe: writing 1536 bytes flash ...

Writing |                                                    
| 0% 0.00savrdude.exe: error: programmer did not respond to command: set extaddr
avrdude.exe: error: programmer did not respond to command: write block
Writing | ########                                           | 16% 0.03s ***failed;
...
avrdude.exe: Error: butterfly programmer uses avr_write_page() but does not
provide a cmd() method.
 *** page 255 (addresses 0x0500 - 0x05ff) failed to write

Writing | ################################################## | 100% 0.79s

avrdude.exe: 1536 bytes of flash written
avrdude.exe: verifying flash memory against .\hex\Blink.ino.mega.hex

Reading |                                                   
 | 0% 0.00savrdude.exe: error: programmer did not respond to command: set extaddr
Reading | ########                                          
 | 16% 5.00savrdude.exe: error: programmer did not respond to command: set extaddr
...

```

mcuee, 2022-08-07:
Applying the patch does not help. I think I need to modify the avr109 further to get it working with ATmega2560.

```
PS C:\work\avr\avrdude_test\avrdude_main> git diff
diff --git a/src/butterfly.c b/src/butterfly.c
index 1d5fafd..a571b72 100644
--- a/src/butterfly.c
+++ b/src/butterfly.c
@@ -321,6 +321,10 @@ static int butterfly_initialize(PROGRAMMER * pgm, AVRPART * p)
   PDATA(pgm)->buffersize = (unsigned int)(unsigned char)c<<8;
   butterfly_recv(pgm, &c, 1);
   PDATA(pgm)->buffersize += (unsigned int)(unsigned char)c;
+
+  // overriding it for now
+  PDATA(pgm)->buffersize = 512;
+
   avrdude_message(MSG_INFO, "Programmer supports buffered memory access with buffersize=%i bytes.\n",
                   PDATA(pgm)->buffersize);

@@ -425,6 +429,7 @@ static void butterfly_display(PROGRAMMER * pgm, const char * p)

 static void butterfly_set_addr(PROGRAMMER * pgm, unsigned long addr)
 {
+  if( addr < 0x10000 ) {
   char cmd[3];

   cmd[0] = 'A';
@@ -433,6 +438,19 @@ static void butterfly_set_addr(PROGRAMMER * pgm, unsigned long addr)

   butterfly_send(pgm, cmd, sizeof(cmd));
   butterfly_vfy_cmd_sent(pgm, "set addr");
+
+  } else {
+
+  char cmd[4];
+
+  cmd[0] = 'H';
+  cmd[1] = (addr >> 16) & 0xff;
+  cmd[2] = (addr >> 8) & 0xff;
+  cmd[3] = addr & 0xff;
+
+  butterfly_send(pgm, cmd, sizeof(cmd));
+  butterfly_vfy_cmd_sent(pgm, "set extaddr");
+  }
 }

```

mcuee, 2022-08-07:
WIth xboot it seems to perform a bit better but still failed with or without the patch.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue454.exe -c avr109 -P COM6 -b 115200 -p m2560  -D -v

avrdude_issue454.exe: Version 7.0-20220804 (5f5002e)
                      Copyright (c) Brian Dean, http://www.bdmicro.com/
                      Copyright (c) Joerg Wunsch

                      System wide configuration file is "C:/work/avr/avrdude_test/avrdude_bin/avrdude.conf"

                      Using Port                    : COM6
                      Using Programmer              : avr109
                      Overriding Baud Rate          : 115200
                      AVR Part                      : ATmega2560
                      Chip Erase delay              : 9000 us
                      PAGEL                         : PD7
                      BS2                           : PA0
                      RESET disposition             : dedicated
                      RETRY pulse                   : SCK
                      Serial program mode           : yes
                      Parallel program mode         : yes
                      Timeout                       : 200
                      StabDelay                     : 100
                      CmdexeDelay                   : 25
                      SyncLoops                     : 32
                      PollIndex                     : 3
                      PollValue                     : 0x53
                      Memory Detail                 :

                                                        Block Poll               Page                       Polled
                        Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                        ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                        eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                        flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                        lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                        calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                        signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

                      Programmer Type : butterfly
                      Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude_issue454.exe: devcode selected: 0x7b
avrdude_issue454.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude_issue454.exe: Device signature = 0x1e9801 (probably m2560)

avrdude_issue454.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue454.exe -c avr109 -P COM6 -b 115200 -p m2560  -D -U .\hex\Blink.ino.mega.hex

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude_issue454.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude_issue454.exe: Device signature = 0x1e9801 (probably m2560)
avrdude_issue454.exe: input file .\hex\Blink.ino.mega.hex auto detected as Intel Hex
avrdude_issue454.exe: reading input file .\hex\Blink.ino.mega.hex for flash
avrdude_issue454.exe: writing 1536 bytes flash ...

Writing | ################################################## | 100% 0.19s

avrdude_issue454.exe: 1536 bytes of flash written
avrdude_issue454.exe: verifying flash memory against .\hex\Blink.ino.mega.hex

Reading | ################################################## | 100% 0.19s

avrdude_issue454.exe: verification error, first mismatch at byte 0x0000
                      0xff != 0x0c
avrdude_issue454.exe: verification error; content mismatch

avrdude_issue454.exe done.  Thank you.

```

mcuee, 2022-08-07:
Actually for xboot flash writing is good even though the verification does not work. And I cn see the LED blinking.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -qqp m2560  
-U lfuse:r:-:h -U hfuse:r:-:h -U efuse:r:-:h
0xff
0xd8
0xfd

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c usbasp -qqp m2560 
 -U .\xboot_m2560.hex && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue454 -c avr109 -P COM6 -b 115200
 -qqp m2560  -D -U flash:w:.\hex\Blink.ino.mega.hex:i && echo OK
Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude_issue454.exe: verification error, first mismatch at byte 0x0000
                      0xff != 0x0c
avrdude_issue454.exe: verification error; content mismatch
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_issue454 -c usbasp -qqp m2560 
 -U flash:v:.\hex\Blink.ino.mega.hex:i && echo OK
OK

```

mcuee, 2022-08-07:
I believe the above xboot bootloader is good for ATmega2560 and there is an issue with avrdude.

AVROSP2 seems to work fine with flash writing. I am using the following GUI version which seems to have more devices support.
https://xdevs.com/doc/xDevs.com/M16/AVROSP2/

Run log from the above GUI.
```
Defaulting to Bootloader mode ...
Found XBOOT++
Entering programming mode...
Signature = 0x1E 0x98 0x01
Leaving programming mode...
Checking programmer type ...
Unrecognized ID found on Com Port 6
Defaulting to Bootloader mode ...
Found XBOOT++
Entering programming mode...
Erasing chip contents...
Reading HEX input file for flash operations...
Programming Flash contents... 0x0000 TO 0x05FF
Using block mode...
Block Size reported =  256
Reading Flash contents...0x0000 TO 0x05FF
Using block mode...
Comparing Flash data...
Equal!
Leaving programming mode...
Checking programmer type ...
Unrecognized ID found on Com Port 6
Defaulting to Bootloader mode ...
Found XBOOT++
Entering programming mode...
Reading HEX input file for flash operations...
Reading Flash contents...0x0000 TO 0x05FF
Using block mode...
Comparing Flash data...
Equal!
Leaving programming mode...
Checking programmer type ...
Unrecognized ID found on Com Port 6
Defaulting to Bootloader mode ...
Found XBOOT++
Entering programming mode...
Reading Flash contents... 0x0000 TO 0x3FFFF
Using block mode...
Writing hex file to: C:\work\avr\avrdude_test\avrdude_bin\hex\avrosp2_readback - Complete!
Leaving programming mode...
```

mcuee, 2022-08-07:
@stefanrueger 
Now I think this issue should be quite similar to the recent issues that you helped to fix for the ATmega2560 (avrftdi, FT245R, PICKit 2 and USBtinyISP). Hopefully you can take a look when you got the time. Thanks.

mcuee, 2022-08-07:
@MCUdude 
I am not so sure if ATxmega256a3 may get more issues than ATmega2560 when using xboot. Hopefully you can get access to the  ATxmega256a3 (or similar) board soon with the help from Microchip.

If you can get AVR109 (not xboot) to work with ATmega2560, please share here as well. Thanks.

MCUdude, 2022-08-07:
@mcuee sorry for the dumb question, but what's really the difference between AVR109, AVR910, AVR911 and Xboot? and is "STK600 xboot" the same as "Xdevs xboot"?

I'm also quite surprised that the "proper" AVR109/910 source code is hard to find. Microchip has hidden it quite well, and there are multiple versions/forks floating around. We should collect the pieces and host them somewhere so we have a known good reference we can use when we troubleshoot AVR109/910/911 issues.

mcuee, 2022-08-07:
AVR109 -- bootlader
https://www.microchip.com/en-us/application-notes/an1644 (you can download the AN and FW source codes here)

AVR910 -- simple pogrammer (mostly outdated other than to program ebn files)
http://ww1.microchip.com/downloads/en/appnotes/atmel-0943-in-system-programming_applicationnote_avr910.pdf

AVR911 -- The AVR Open-source Programmer (AVROSP) application for AVR109 and AVR910
https://www.microchip.com/en-us/application-notes/an2568 (you can download AN and PC application here)

ATxmega and AVR109 AN:
http://ww1.microchip.com/downloads/en/Appnotes/Atmel-8242-XMEGA-Boot-Loader-Quick-Start-Guide_ApplicationNote_AVR1605.pdf

xboot -- extension to AVR109, mostly for xmega AVRs but apparently also for mega AVRs.
https://github.com/alexforencich/xboot

I am not so sure about the difference between `STK600 xboot` and `Xdevs xboot` as I do not knwo of them.

mcuee, 2022-08-07:
> I'm also quite surprised that the "proper" AVR109/910 source code is hard to find. Microchip has hidden it quite well, and there are multiple versions/forks floating around. We should collect the [...]

The official Microchip AVR109 bootloader firmware source codes can be downloaded here.
https://www.microchip.com/en-us/application-notes/an1644

Indeed I have issues to get AVR109 to work with ATmega2560 using the above official codes. Even for  ATmega8A and ATmega16A which should work out of the box, I am having issues now with minimum system (8MHz crystal and a reset button and power supply input) . I only managed to get it work with ATmega32A and ATmega328P (Arduino Uno Clone).

I will start a discussion to collect info about AVR109 along with its extensions like xboot.

MCUdude, 2022-08-07:
@mcuee thanks for the clarification! I think we should focus on improving AVR109 and xboot support, and ignore AVR910 for now. AVR911/AVROSP looks like it's just a PC application. I'll see if I can compile some AVR109 and xboot bootloaders for various targets and help out with some testing.

MCUdude, 2022-08-07:
I've tested a little. The AVR109 build process was really painful, but I finally got something.

AVR109 compiled for ATmega1284P (works):

<details>

```
$ ./avrdude -cavr109 -p atmega1284p -b 115200 -P /dev/cu.usbserial-1420 -Uflash:w:chaucer112k.hex:i

Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.5; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer112k.hex"
avrdude: writing flash (110910 bytes):

Writing | ################################################## | 100% 14.83s

avrdude: 110910 bytes of flash written
avrdude: verifying flash memory against chaucer112k.hex:

Reading | ################################################## | 100% 11.87s

avrdude: 110910 bytes of flash verified

avrdude done.  Thank you.
```

</details>

AVR109 compiled for ATmega2560 (does not work):
<details>

```
$ ./avrdude -cavr109 -p atmega2560 -b 115200 -P /dev/cu.SLAB_USBtoUART -Uflash:w:chaucer256k.hex:i

Connecting to programmer: .
Found programmer: Id = "AVRBOOT"; type = S
    Software Version = 1.5; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer256k.hex"
avrdude: writing flash (176758 bytes):

Writing |                                                    | 0% 0.00savrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: write block
 ***failed;  
[... 287 lines cut ...]
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set extaddr
avrdude: butterfly_recv(): programmer is not responding
Reading | #                                                  | 1% 25.08savrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
Reading | #                                                  | 2% 25.26savrdude: error: programmer did not respond to command: set extaddr
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: error: programmer did not respond to command: set extaddr
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set extaddr
^C

```

</details>

MCUdude, 2022-08-07:
Xboot seems to work well:

AVR109 (XBOOT) ATmega1284P (works):

<details>

```
$ ./avrdude -cavr109 -p atmega1284p -b 115200 -P /dev/cu.usbserial-1410 -Uflash:w:chaucer112k.hex:i

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer112k.hex"
avrdude: writing flash (110910 bytes):

Writing | ################################################## | 100% 16.80s

avrdude: 110910 bytes of flash written
avrdude: verifying flash memory against chaucer112k.hex:

Reading | ################################################## | 100% 11.60s

avrdude: 110910 bytes of flash verified

avrdude done.  Thank you.
```

</details>

AVR109 (XBOOT) ATmega2560 (works):

<details>

```
$ ./avrdude -cavr109 -p atmega2560 -b 115200 -P /dev/cu.SLAB_USBtoUART -Uflash:w:chaucer256k.hex:i

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "chaucer256k.hex"
avrdude: writing flash (176758 bytes):

Writing | ################################################## | 100% 23.00s

avrdude: 176758 bytes of flash written
avrdude: verifying flash memory against chaucer256k.hex:

Reading | ################################################## | 100% 17.53s

avrdude: 176758 bytes of flash verified

avrdude done.  Thank you.
```

</details>

MCUdude, 2022-08-07:
> I believe the above xboot bootloader is good for ATmega2560 and there is an issue with avrdude.

I'm not so sure... I also had issues with AVR109/xboot when trying to either write or verify the flash. Try removing the -D flag and use -e instead, as I've done in my tests. The bootloader can't erase itself, so it's not "dangerous" in any way. It would be interesting to see how a 256kiB xmega performs. I just need to get a hold of one first.

mcuee, 2022-08-08:
> I'm not so sure... I also had issues with AVR109/xboot when trying to either write or verify the flash. Try removing the -D flag and use -e instead, as I've done in my tests. The bootloader can't [...]

@MCUdude 
Yes you are right. Thanks for the `-e` tip and now I can confirm xboot hex file works fine and there is no issue with avrdude in this case.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c avr109 -P COM3 -b 115200
 -qqp m2560 -e -U flash:w:.\hex\Blink.ino.mega.hex:i && echo OK
Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=256 bytes.

Programmer supports the following devices:
    Device code: 0x7b

OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git.exe -c usbasp -qqp m2560 
-U flash:v:.\hex\Blink.ino.mega.hex:i &&
 echo OK
OK

```

mcuee, 2022-08-08:
Just look at the history, avrdude already supports ATmega256x long ago with the following commit on 30-Jan-2007.
https://github.com/avrdudes/avrdude/commit/6fc6e9b184a5fb3cf28ca863f823ed6b84b0638a

This issue is saying that it somehow does not work for ATxmega2560A3. This needs to be confirmed. So I will change the label.

mcuee, 2022-08-08:
Based on the test results here, I have also closed #304 as the test by Hans shows that the current avrdude works with ATxmega128A3U.
* #304

MCUdude, 2022-08-08:
I doubt the _original_ AVR109 bootloader supports devices with more than 128kiB of flash, and the only popular 109 compatible bootloader I know of that does is xboot, which also has been around for a very long time.

@mcuee I don't have a Windows computer available at the moment, but can you test if the [AVR109_atmega2560.zip](https://github.com/avrdudes/avrdude/files/9276583/AVR109_atmega2560.zip) you provided works with the AVROSP application? If yes, that means that the original AVR109 bootloader _does_ support 256kiB targets and that there is a bug in Avrdude.

(And here is the binary I compiled. Short PB7 (Arduino pin 13) to ground to enter bootloader mode. Also, note that the hfuse needs to be set to `0xd4`). [AVR109_atmega2560_PB7.hex.zip](https://github.com/avrdudes/avrdude/files/9278978/AVR109_atmega2560_PB7.hex.zip)

MCUdude, 2022-08-08:
> @mcuee I don't have a Windows computer available at the moment, but can you test if the [AVR109_atmega2560.zip](https://github.com/avrdudes/avrdude/files/9276583/AVR109_atmega2560.zip) you provided [...]

BTW, for some reason, I couldn't get the ATmega16U2 USB to serial adapter on my ATmega2560 to work with the original AVR109 bootloader. Both the RX and TX LEDs flashed twice a second, but this was printed again and again in -vvvv mode. When using a dedicated USB to serial adapter (CP2012 based in my case), it was able to communicate with Avrdude. the ATmega16u2 did work with xboot though, even though the baud rate was the same for both (115200).

```
$ ./avrdude -cavr109 -p atmega2560 -b 115200 -P /dev/cu.usbmodem14101 -Uflash:w:chaucer112k.hex:i -vvvv
executable_abspath = /Users/hans/Downloads/avrdude/src/avrdude
executable_abspath_len = 41
executable_dirpath = /Users/hans/Downloads/avrdude/src
executable_dirpath_len = 33
sys_config = /Users/hans/Downloads/avrdude/src/avrdude.conf
sys_config_found = true

avrdude: Version 7.0-20220508
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is "/Users/hans/Downloads/avrdude/src/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem14101
         Using Programmer              : avr109
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[... 58 lines cut ...]
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
.avrdude: Send: . [1b] 
avrdude: Send: S [53] 
avrdude: Recv: ? [3f] 
^C
```

mcuee, 2022-08-08:
> I doubt the _original_ AVR109 bootloader supports devices with more than 128kiB of flash, and the only popular 109 compatible bootloader I know of that does is xboot, which also has been around for a very long time.

xboot was started in 2010 (first import to Google Code was on 10 May 2010.
https://github.com/alexforencich/xboot/commit/1ec9a9c2ab4bc21b5646c4d83816588d8ff197d6

avrdude seems to support AVR109 for ATmega2560 back on 30-Jan-2007. 
https://github.com/avrdudes/avrdude/commit/6fc6e9b184a5fb3cf28ca863f823ed6b84b0638a

So there must be some AVR109 implementation for ATmega2560 back in 2007. It is just Microchip's AVR109 FW release may need to be tuned to work with ATmega2560. 

> @mcuee I don't have a Windows computer available at the moment, but can you test if the [AVR109_atmega2560.zip](https://github.com/avrdudes/avrdude/files/9276583/AVR109_atmega2560.zip) you provided [...]

The hex file I compiled did not work with AVROSP either. So it is not an issue with avrdude.

> (And here is the binary I compiled. Short PB7 (Arduino pin 13) to ground to enter bootloader mode. Also, note that the hfuse needs to be set to `0xd4`). [AVR109_atmega2560_PB7.hex.zip](https://github.com/avrdudes/avrdude/files/9278978/AVR109_atmega2560_PB7.hex.zip)

I will try your binary as well. Thanks.

mcuee, 2022-08-08:
> BTW, for some reason, I couldn't get the ATmega16U2 USB to serial adapter on my ATmega2560 to work with the original AVR109 bootloader. Both the RX and TX LEDs flashed twice a second, but this was [...]

Interesting to know that. My Arduino Mega2560 is the clone without the ATmega16U2, rather it is using CH340. So I do not have this issue you mentioned. It could be the USB to Serial FW implementation of ATmega16U2 (CDC-ACM) is not as good.

Arduino is using very old version of LUFA.
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2/arduino-usbserial

Their USB DFU implementation is using the same old version of LUFA and I found a bug.
* https://github.com/arduino/ArduinoCore-avr/issues/490
* https://github.com/avrdudes/avrdude/issues/1043
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2/arduino-usbdfu

I am having issues with ATmega8A/ATmega16A (1024B AVR109 bootloader) using an FT232R USB-TTL converter.

mcuee, 2022-08-08:
> And the following avrfreak thread says that avrdude 5.41 works with ATmega2561 in 2007.
> https://www.avrfreaks.net/forum/avr-prog-and-new-mega-family-devices
> So there must be some AVR109 implementation for ATmega2560 back in 2007. It is just Microchip's AVR109 FW release may need to be tuned to work with ATmega2560. 

Apparently the comments by Mike H never materialized. Atmel did not update the AN or the FW codes to mention the `H` command.
https://www.avrfreaks.net/comment/232269#comment-232269

Okay, I received a response back from Atmel. They had actually updated the AVR109 Appnote a couple of weeks ago.

_The new command is: 'H' for the bootloader.
Followed by a 3 byte address (MSB first).
Bootloader should respond with a CR (13).

Mike H._

MCUdude, 2022-08-08:
> Apparently the comments by Mike H never materialized. Atmel did not update the AVR109 AN or the FW codes to mention the H command. Therefore it is normal that if we started from Microchip AVR109 FW codes, it would not work with ATmega2560. xboot followed the above info and got it to work with ATmega2560.

Great find! The question is; does there exist a modified AVR109 bootloader that supports 256kiB targets that _isn't_ xboot?
If yes, we should see if we can get this tested working with Avrdude for 256kiB targets. If not, all we have to do before we can close this issue is to confirm that xboot works with xmegas with more than 128kiB flash.

It's a bit sad that the "official" AVR109 bootloader never got support for 256kiB targets. I wouldn't expect xmega support as that's a lot of work, but support for 256kiB targets shouldn't be too much work for those who already know the source code. Maybe we could annoy Microchip technical support enough that they will set someone to do the job?

MCUdude, 2022-08-08:
>> (And here is the binary I compiled. Short PB7 (Arduino pin 13) to ground to enter bootloader mode. Also, note that the hfuse needs to be set to 0xd4). [AVR109_atmega2560_PB7.hex.zip](https://github.com/avrdudes/avrdude/files/9278978/AVR109_atmega2560_PB7.hex.zip)

> I will try your binary as well. Thanks.

No need! I tried it on a Windows computer, and it didn't work with AVROSP either.

mcuee, 2022-08-08:
> Great find! The question is; does there exist a modified AVR109 bootloader that supports 256kiB targets that isn't xboot?
If yes, we should see if we can get this tested working with Avrdude for 256kiB targets. If not, all we have to do before we can close this issue is to confirm that xboot works with xmegas with more than 128kiB flash.

I think we can close this issue once you confirm that avrdude/xboot combination works for the ATxmega2560A3U or other parts with higher than 128kiB flash memory.

I think there were AVR109 implementation for ATmega2560 before xboot (say in 2007 based on the 200i7 avrfreak forum post I mentioned before. However, I can not find anything now so I think the old implementations no long matter.

mcuee, 2022-08-11:
> BTW, for some reason, I couldn't get the ATmega16U2 USB to serial adapter on my ATmega2560 to work with the original AVR109 bootloader. Both the RX and TX LEDs flashed twice a second, but this was [...]

@MCUdude 
I just tried similar things on my Arduino Clone with ATmega16U2. And it seems to work fine. I am using PD4 and 57600 bps on my board. I was doing the same test with another Arduino Uno with CH340G and the same 57600bps FW works. But then 115200bps FW did not work. 
https://github.com/avrdudes/avrdude/issues/980#issuecomment-1200139029

So maybe you can redude to 57600bps to see if that helps the ATmega16U2. Probably CP2102 works better in this aspect.

MCUdude, 2022-09-24:
I can confirm that the patch attached to the first post resolves this issue and #360.

However, the patch isn't perfect. It hardcodes the page size to 512 bytes, so this has to be figured out.

I was a bit worried that `if(addr < 0x10000)` would affect 128kiB parts (as these works perfectly _without_ this patch), but it doesn't seem to cause any problems for the ATxmega128A3U at least:

```
$ ./avrdude -cavr109 -p atxmega128a3u -b 115200 -P /dev/cu.usbserial-AH00M3HQ -e -Uflash:w:0x55_128kib.hex 

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9742 (probably x128a3u)
avrdude: erasing chip
avrdude: reading input file 0x55_128kib.hex for flash
avrdude: writing 131071 bytes flash ...

Writing | ################################################## | 100% 20.46s

avrdude: 131071 bytes of flash written
avrdude: verifying flash memory against 0x55_128kib.hex

Reading | ################################################## | 100% 19.94s

avrdude: 131071 bytes of flash verified

avrdude done.  Thank you.

```

MCUdude, 2022-09-24:
It seems like the `PDATA(pgm)->buffersize = 512;` override isn't necessary. The bootloader sends the correct size regardless:

Output:

```
Buffer size before override=512 bytes.
Buffer size after override=512 bytes.
```

stefanrueger, 2022-10-05:
Fixed with PR #1101

closed 2022-10-05 by stefanrueger