avrdudes/avrdude #455: [bug #51320] 0x00 written as 0xFF when byte count is less than 0x10

avrdude-admin, 2021-12-11:
Mike Brandon <smbrandonjr@gmail.com>
Mon 26 Jun 2017 08:13:10 PM UTC
Programmer hardware: linuxgpio
Device type: m2650

I have two different .hex files (unfortunately I am unable to share entire .hex files as I do not own them). Both write and verify fine when using AVR Studio and an AVRISP programmer.

When using avrdude with linuxgpio, the write process completes but the verification process returns an error. In both files it occurs on lines of the .hex file where the data byte count is less than 0x10 (this is the only common theme I have been able to identify). When I look at the specific address mentioned by avrdude for the verification error, the original data that SHOULD have been written was 0x00. The data that WAS written was 0xFF.

The attached files only pertain to one of the hex files I have experienced this with.

[file #41054: 0x40ca-read.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41054)
[file #41053: avrdude-write-read.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41053)
[file #41052: 0x40ca-data.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41052)
[file #41079: blink-2560-mod.hex](https://savannah.nongnu.org/bugs/download.php?file_id=41079)
[file #41127: blink-mod-bitbang-read.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41127)
[file #41126: blink-mod-bitbang-write.PNG](https://savannah.nongnu.org/bugs/download.php?file_id=41126)

This issue was migrated from https://savannah.nongnu.org/bugs/?51320

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Fri 30 Jun 2017 12:03:35 AM UTC

Here is an intel hex file that will return a verification error due to "00" getting written as "FF".

(**[file #41079](https://savannah.nongnu.org/support/download.php?file_id=41079)**)

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Wed 05 Jul 2017 05:35:02 PM UTC

In further troubleshooting of my issue, I did examine the verbose output during the writing and flashing of the blink-2560-mod hex file. Interestingly, the bitbang_cmd() output shows that it did write "00" for the byte in question. During the read, "FF" is being read. I do know that if I read the firmware with AVR Studio it will also confirm the byte mismatch so even though avrdude bitbang shows "00", it is either not writing it (and FF is a remnant of the initial chip erase) or it is writing FF.

I have attached two more images showing the verbose output during this scenario.

[image]
[image]

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Tue 11 Jul 2017 06:56:57 PM UTC

In further testing, the issue is not specific to trying to write "00". I will receive a verification error anytime the data byte count is less than 0x10 AND the last data byte != 0xFF.

I have verified this response with numerous .hex files on the Atmega1284 and Atmega 2560 chips.

Atmel Studio will write and verify these same .hex files without error.

avrdude-admin, 2021-12-11:
Mike <smbrandonjr>
Tue 18 Jul 2017 02:54:09 PM UTC

After further troubleshooting, I believe I have discovered more precisely what is causing the verification error and in an effort to not bury it below this bug thread, I want to open a new bug. This bug can be closed.

avrdude-admin, 2021-12-11:
Andrzej Pietrasiewicz <andrzejtp>
Sun 28 Jul 2019 03:37:03 PM UTC

**[comment #2:](https://savannah.nongnu.org#comment2)**

> In further troubleshooting of my issue, I did examine the verbose output during the writing and flashing of the blink-2560-mod hex file. Interestingly, the bitbang_cmd() output shows that it did [...]
[image]

avrdude-admin, 2021-12-11:
Andrzej Pietrasiewicz <andrzejtp>
Sun 28 Jul 2019 03:39:39 PM UTC

Please see [https://www.avrfreaks.net/comment/2739451#comment-2739451](https://savannah.nongnu.orghttps://www.avrfreaks.net/comment/2739451#comment-2739451)

The problem is with hex file records containing an odd number of data bytes.

avrdude-admin, 2021-12-11:
Andrzej Pietrasiewicz <andrzejtp>
Wed 31 Jul 2019 08:24:23 PM UTC

I have a solution - if an ihex record has odd length then mark the missing high byte as allocated and let it contain 0xff:

diff --git a/fileio.c b/fileio.c
index 02c50ab..e3c82a4 100644
--- a/fileio.c
+++ b/fileio.c
@@ -334,6 +334,10 @@ static int ihex2b(char * infile, FILE * inf,
mem->buf[nextaddr+i] = ihex.data[i];
mem->tags[nextaddr+i] = TAG_ALLOCATED;
}
+	if (ihex.reclen & 0x1) {
+          mem->buf[nextaddr+i] = 0xff;
+          mem->tags[nextaddr+i] = TAG_ALLOCATED;
+	}
if (nextaddr+ihex.reclen > maxaddr)
maxaddr = nextaddr+ihex.reclen;
break;

But why is the record length odd in the first place if flash locations are 2-byte locations?

mcuee, 2022-06-03:
Related issue with Mega2560 and linuxgpio.
https://github.com/avrdudes/avrdude/issues/458

mcuee, 2022-07-10:
I will try to test this soon with the above-mentioned hex file for ATmega2560.
```
:1000000006C1000016C1000014C1000012C10000AA
:1000100010C100000EC100000CC100000AC10000A8
:1000200008C1000006C1000004C1000002C10000B8
:1000300000C10000FEC00000FCC00000FAC00000CB
:10004000F8C00000F6C00000F4C00000F2C00000DC
:10005000F0C00000EEC00000ECC00000C2C1000013
:10006000E8C00000E6C00000E4C00000E2C00000FC
:10007000E0C00000DEC00000DCC00000DAC000000C
:10008000D8C00000D6C00000D4C00000D2C000001C
:10009000D0C00000CEC00000CCC00000CAC000002C
:1000A000C8C00000C6C00000C4C00000C2C000003C
:1000B000C0C00000BEC00000BCC00000BAC000004C
:1000C000B8C00000B6C00000B4C00000B2C000005C
:1000D000B0C00000AEC00000ACC00000AAC000006C
:1000E000A8C000004501480137013B013F017F01E5
:1000F0004C01500156015A015E01640168016C0116
:100100007F01720176017A0100002100240027009E
:100110002A002D0030003300010100000401070116
:100120000A0100002200250028002B002E003100CB
:10013000340002010000050108010B010505050559
:10014000070508080808020202020A0A080804044F
:10015000040401010101010101010303030303037D
:100160000303040707070C0C0C0C0C0C0C0C02020C
:10017000020206060606060606060B0B0B0B0B0B09
:100180000B0B010210202008081020401020408096
:10019000020102010804020101020408102040804B
:1001A00080402010080402018004020180402010D9
:1001B0000804020108040201010204081020408022
:1001C000010204081020408000000A0B02090C0DF7
:1001D0000E080703040100000000000000000000FA
:1001E000000000000000000000000000000000000F
:1001F00000000000121110000000000000000000CC
:1002000000000000000000000000000000001124B9
:100210001FBECFEFD1E2DEBFCDBF00E00CBF22E0BA
:10022000A0E0B2E001C01D92A930B207E1F723D1EE
:10023000CEC1E6CECF93C82FE5EDF1E04491EFE8D3
:10024000F1E02491E9E4F1E09491992309F46AC082
:10025000442309F454C050E0FA013197E231F1052A
:1002600008F04DC08827EE58FF4F8F4FA8C18091EE
:1002700080008F7707C0809180008F7D03C08091C0
:100280008000877F809380003AC084B58F7702C05A
:1002900084B58F7D84BD33C08091B0008F7703C05B
:1002A0008091B0008F7D8093B00029C08091900034
:1002B0008F7707C0809190008F7D03C08091900060
:1002C000877F809390001BC08091A0008F7707C02C
:1002D0008091A0008F7D03C08091A000877F8093D4
:1002E000A0000DC0309120013F7707C03091200160
:1002F0003F7D03C030912001377F30932001E92FEB
:10030000F0E0EE0FFF1FEE5DFE4FA591B4918FB7A9
:10031000F894EC91C11103C020952E2301C02E2B1F
:100320002C938FBFCF9108953FB7F8948091050229
:1003300090910602A0910702B091080226B5A89BF1
:1003400005C02F3F19F00196A11DB11D3FBFBA2F67
:10035000A92F982F8827820F911DA11DB11DBC01C7
:10036000CD0142E0660F771F881F991F4A95D1F78C
:1003700008958F929F92AF92BF92CF92DF92EF92A9
:10038000FF926B017C01D0DF4B015C01C114D104F1
:10039000E104F104E9F0C8DFDC01CB018819990917
:1003A000AA09BB09883E9340A105B10578F321E075
:1003B000C21AD108E108F10888EE880E83E0981E81
:1003C000A11CB11CC114D104E104F10421F7DECF5A
:1003D000FF90EF90DF90CF90BF90AF909F908F9065
:1003E00008951F920F920FB60F9211242F933F93EF
:1003F0008F939F93AF93BF938091010290910202DC
:10040000A0910302B09104023091000223E0230F77
:100410002D3720F40196A11DB11D05C026E8230F3C
:100420000296A11DB11D20930002809301029093BA
:100430000202A0930302B0930402809105029091FE
:100440000602A0910702B09108020196A11DB11DFC
:100450008093050290930602A0930702B0930802CE
:10046000BF91AF919F918F913F912F910F900FBEB0
:100470000F901F901895789484B5826084BD84B5E0
:10048000816084BD85B5826085BD85B5816085BD8F
:1004900080916E00816080936E0010928100809147
:1004A00081008260809381008091810081608093CF
:1004B0008100809180008160809380008091B100F4
:1004C00084608093B1008091B00081608093B0001F
:1004D0008091910082608093910080919100816071
:1004E0008093910080919000816080939000809132
:1004F000A10082608093A1008091A100816080931F
:10050000A1008091A00081608093A00080912101D2
:10051000826080932101809121018160809321017B
:100520008091200181608093200180917A00846015
:1005300080937A0080917A00826080937A00809123
:100540007A00816080937A0080917A00806880933D
:100550007A001092C100EFE8F1E02491E9E4F1E0C3
:100560008491882399F090E0880F991FFC01E85F3F
:10057000FE4FA591B491FC01EE5DFE4F85919491E3
:100580008FB7F894EC91E22BEC938FBFC0E0D0E0F2
:1005900081E050DE68EE73E080E090E0EADE80E02B
:1005A00049DE68EE73E080E090E0E3DE60E070E05A
:1005B000CB01DFDE209761F30E940000E9CFEE0F50
:0F05C000FF1F881F8BBF0790F691E02D19940045
:0205D000FFCF5B
:00000001FF

```

mcuee, 2022-07-11:
I can confirm that there is a bug here.
```
pi@raspberrypi:~/Desktop/build/avr/avrdude/build_linux/src $ sudo ./avrdude -C ./avrdude.conf -c linuxgpio -P GPIO -p m2560 -U flash:w:./blink-2560-mod.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blink-2560-mod.hex"
avrdude: writing flash (1490 bytes):

Writing | ################################################## | 100% 0.35s

avrdude: 1490 bytes of flash written
avrdude: verifying flash memory against ./blink-2560-mod.hex:

Reading | ################################################## | 100% 0.64s

avrdude: verification error, first mismatch at byte 0x05ce
         0xff != 0x00
avrdude: verification error; content mismatch

avrdude done.  Thank you.

```

mcuee, 2022-07-11:
Not so sure if this has anything similar to the issue mentioned in #825.

mcuee, 2022-07-11:
I can confirm that the patch fixed the issue.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude_issue455 $ git diff
diff --git a/src/fileio.c b/src/fileio.c
index 7c021ce..f5e7e32 100644
--- a/src/fileio.c
+++ b/src/fileio.c
@@ -336,6 +336,12 @@ static int ihex2b(char * infile, FILE * inf,
           mem->buf[nextaddr+i] = ihex.data[i];
           mem->tags[nextaddr+i] = TAG_ALLOCATED;
         }
+
+        if(ihex.reclen & 0x1) {
+          mem->buf[nextaddr+i] = 0xff;
+          mem->tags[nextaddr+i] = TAG_ALLOCATED;
+        }
+
         if (nextaddr+ihex.reclen > maxaddr)
           maxaddr = nextaddr+ihex.reclen;
         break;

pi@raspberrypi:~/Desktop/build/avr/avrdude_issue455/build_linux/src $ sudo ./avrdude -C ./avrdude.conf 
-c linuxgpio -P GPIO -p m2560 -U flash:w:./blink-2560-mod.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blink-2560-mod.hex"
avrdude: writing flash (1490 bytes):

Writing | ################################################## | 100% 0.33s

avrdude: 1490 bytes of flash written
avrdude: verifying flash memory against ./blink-2560-mod.hex:

Reading | ################################################## | 100% 0.64s

avrdude: 1490 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-11:
> Not so sure if this has anything similar to the issue mentioned in #825 or #936.
> 
> * [Fix TPI memory writes for odd number of bytes #825](https://github.com/avrdudes/avrdude/pull/825)
> * [Fix Issue #918: Enable avrdude to send full input file incl trailing 0xff #936](https://github.com/avrdudes/avrdude/pull/936)

@stefanrueger 
Looks like #936 does not help in this case. Still please take a look to see if there are similarities here.

```

pi@raspberrypi:~/Desktop/build/avr/avrdude_sr/build_linux/src $ sudo ./avrdude -C ./avrdude.conf 
-c linuxgpio -P GPIO -p m2560 -U flash:w:./blink-2560-mod.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "./blink-2560-mod.hex"
avrdude: writing flash (1490 bytes):

Writing | ################################################## | 100% 0.34s

avrdude: 1490 bytes of flash written
avrdude: verifying flash memory against ./blink-2560-mod.hex:

Reading | ################################################## | 100% 0.65s

avrdude: verification error, first mismatch at byte 0x05ce
         0xff != 0x00
avrdude: verification error; content mismatch

avrdude done.  Thank you.

```

mcuee, 2022-07-11:
I can confirm that linuxspi has the same issue and the same patch works to fix the issue.

```
pi@raspberrypi:~/Desktop/build/avr/avrdude/build_linux/src $ ./avrdude -C ./avrdude.conf  -p m2560 -c linuxspi -P /dev/spidev0.0:/dev/gpiochip0 -U flash:w:blink-2560-mod.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "blink-2560-mod.hex"
avrdude: writing flash (1490 bytes):

Writing | ################################################## | 100% 0.35s

avrdude: 1490 bytes of flash written
avrdude: verifying flash memory against blink-2560-mod.hex:

Reading | ################################################## | 100% 0.90s

avrdude: verification error, first mismatch at byte 0x05ce
         0xff != 0x00
avrdude: verification error; content mismatch

avrdude done.  Thank you.

pi@raspberrypi:~/Desktop/build/avr/avrdude_issue455/build_linux/src $ ./avrdude -C ./avrdude.conf  -p m2560 -c linuxspi -P /dev/spidev0.0:/dev/gpiochip0 -U flash:w:blink-2560-mod.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "blink-2560-mod.hex"
avrdude: writing flash (1490 bytes):

Writing | ################################################## | 100% 0.36s

avrdude: 1490 bytes of flash written
avrdude: verifying flash memory against blink-2560-mod.hex:

Reading | ################################################## | 100% 0.95s

avrdude: 1490 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-07-11:
BTW, there is no issue with an STK500v2 clone.
```
PS C:\work\avr\avrdude_test\avrdude-7.0_bin64> .\avrdude_issue1004v3.exe -c stk500v2 -P COM6 
-qqp m2560 -U flash:w:.\blink-2560-mod.hex:i && echo OK
OK
```

stefanrueger, 2022-07-12:
@mcuee I had a look and I am *puzzled* by the patch. A hex file can piece together an application in whichever way it likes, eg, a 10-byte contiguous input file might well be described in a hex file as one record of 7 bytes followed by another record of 3 bytes. It should not matter *how* the hex file looks like, what matters is how the resulting file after reading looks like. On that grounds alone I would reject the patch. 

So, if the patch works in the observed case, it must have been something else that helped...

mcuee, 2022-07-31:
> So, if the patch works in the observed case, it must have been something else that helped...

@stefanrueger 
Just wondering if you have some pointer to further debug the issue. Thanks.

mcuee, 2023-01-05:
I can confirm that PR #1265 fixed this issue.

Analysis by @stefanrueger 
https://github.com/avrdudes/avrdude/issues/1263#issuecomment-1371159492

closed 2023-01-05 by stefanrueger